libusb/libusb #1567: Claim interface in parallel on multiple devices with same vendor id and product id on MacOS returns LIBUSB_ERROR_ACCESS

kaciula, 2024-09-18:
I have several usb devices with the same vendor id and product id connected to the computer. I can open, claim interface and interact with each device in a linear fashion, one by one and everything works fine.

However, if I am trying to do parallel work on more than one device at a time, it does not work. I can open a thread for each device, open the device successfully but only the first interface claim succeeds while the others return LIBUSB_ERROR_ACCESS.

Is this a known limitation? I don't see why I cannot claim an interface on a different device (with same vendor id and product id) at the same time. I would appreciate any insights.

mcuee, 2024-09-18:
Does the device have unique Serial Number?

kaciula, 2024-09-18:
@mcuee iSerialNumber field is null for all devices. Why does it matter in this case? I have distinct device handles.

kaciula, 2024-09-18:
I've now tested using WebUSB and I can open multiple devices and claim interfaces in parallel.

kaciula, 2024-09-18:
It turns out there was a bug on my part. The same first device was erroneusly being open in separate threads. Sorry about this.

closed 2024-09-18 by kaciula