libusb/libusb #1568: nut_libusb_get_interrupt: Connection timed out

masterwishx, 2024-09-30:
Using NUT plugin for Unraid , libusb-1.0.26 (API: 0x1000108)

i have a lot of `nut_libusb_get_interrupt: Connection timed out` in debug log and 
maybe also affect on https://github.com/networkupstools/nut/issues/2643

opened issue in https://github.com/networkupstools/nut/issues/2644
not sure where its related here or to nut

Mostly looks like this :  
```
send_to_all: SETINFO driver.state "quiet"
Sep 27 09:37:30 DaRKNaS usbhid-ups[24218]: [D6:24218] send_to_all: write 29 bytes to socket 16 succeeded (ret=29): SETINFO driver.state "quiet"
Sep 27 09:37:31 DaRKNaS usbhid-ups[24218]: [D5:24218] send_to_all: SETINFO driver.state "updateinfo"
Sep 27 09:37:31 DaRKNaS usbhid-ups[24218]: [D6:24218] send_to_all: write 34 bytes to socket 16 succeeded (ret=34): SETINFO driver.state "updateinfo"
Sep 27 09:37:31 DaRKNaS usbhid-ups[24218]: [D1:24218] upsdrv_updateinfo...
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D2:24218] nut_libusb_get_interrupt: Connection timed out
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D1:24218] Got 0 HID objects...
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D1:24218] Quick update...
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D4:24218] 
```

[image]

from https://github.com/networkupstools/nut/blob/1560afe330d09479370f3ed25e3289392c7aebc5/drivers/libusb1.c#L1077 : 

[image]

tormodvolden, 2024-09-30:
I am not sure you need to open an issue for now, since you already have a bug filed for it in nut and it is not clear if there is at all an issue with libusb.

masterwishx, 2024-10-01:
OK, so will close for now ?

closed 2024-11-22 by mcuee