libusb/libusb #1569: libusb_get_device_list() will not show a device's new address if it is connected, removed, then reconnected.

JonathanLeeIFX, 2024-09-30:
This was originally opened on the [node-usb](https://github.com/node-usb/node-usb) project here: node-usb/node-usb#803

My issue is that `node-usb/getDeviceList()` does not appear to update the list of connected devices when a target device is removed and then reconnected.
**thegecko** confirmed the issue and added that this appears to be Windows-specific. It does not occur on macos.
The source of the issue appears to be `libusb_get_device_list()`.

### Steps to Reproduce
I tested this using "usbview.exe" to check the Device Address and a Typescript program that tries to find a Silicon Labs C8051 device with VID=0x10C4 and PID=0xEA61 every 3 seconds.

1. Connect the device and check the Device Address in "usbview.exe".
2. Launch the Typescript program. Observe that it can detect the device and reports the same Device Address.
3. Disconnect the device. Wait 2 seconds.
4. Reconnect the device. Observe that "usbview.exe" shows a different Device Address for the C8051, but the Typescript program continues to report the old Device Address.

### Additional Information
- Operating System: Windows 10
- Runtime: Node v16.17.0
- Node USB Version: usb@2.13.0

### Typescript to Reproduce:
```
import * as usb from 'usb';
async function main() {
	let i = 0;
	do {
		await delay(3000);	
		const devices = usb.findByIds(<VENDOR_ID>, <PRODUCT_ID>);		
		console.log(`device run ${i}: `, devices);
		i++;
	} while (true);
}
function delay(ms: number) {
    return new Promise( resolve => setTimeout(resolve, ms) );
}
main();
```

tormodvolden, 2024-09-30:
The Windows backend doesn't have hot-plugging support, so you'll have to run libusb_get_device_list() repeatedly. I don't know what the node wrapper is doing. Can you reproduce this with an equivalent C program, or at least trace the libusb calls of the node wrapper?

JonathanLeeIFX, 2024-10-01:
I'm attempting to create an equivalent C program, but I have gotten stuck as I am unfamiliar with C.
Please find my current work here: [libusbTest](https://github.com/JonathanLeeIFX/libusbTest)

I am able to call `libusb_get_version()` which implies the DLL should be linked correctly.
However, uncommenting `libusb_init_context()` and `libusb_exit()` on lines 19 and 20 causes issues. It will compile correctly, but then nothing will print out.
I would expect to at least see my "hello" and "done", but it seems to exit with no error code.

If you have an idea about how to proceed, I would appreciate the help.

tormodvolden, 2024-10-01:
I don't have any problem compiling (`gcc main.cpp -lusb-1.0`) and running your program (on Linux) with or without lines 19 and 20 uncommented. And I agree that libusb_get_version() working shows that the linking works. But I don't know how you have compiled libusb or if there are some possible incompatibilities. When I tested it was of course not with your library binaries.

Maybe you can try starting out with the libusb sources and build the examples there?

JonathanLeeIFX, 2024-10-02:
After many trials, I have gotten the code to work. I had to change to `libusb_init()`. Please see my above repo.
As it turns out, `libusb` works as expected and the device address is updating properly. I have included an example output in the README.

I will inform **thegecko** and let him know that `libusb` is working properly.
Thank you for your time and assistance.

closed 2024-10-02 by JonathanLeeIFX

tormodvolden, 2024-10-02:
> I had to change to `libusb_init()`

That is unexpected, because libusb_init(NULL) should be equivalent to libusb_init_context(NULL, NULL, 0)

JonathanLeeIFX, 2024-10-03:
That was my expectation as well. I can only assume I messed something up during setup or compilation/linking files.