libusb/libusb #1571: [linux] Missing netlink message  lead to hotplug not triggering

fabiensanglard, 2024-10-09:
I am encountering an issue with libusb hotplug on linux.

Context: We are not using `udev` because we don't know if our users will have it installed. Instead we compile libusb to use `netlink` for hotplug detection.
The machine I am using is a ThinkStation P620 (documentation [here](https://download.lenovo.com/pccbbs/thinkcentre_pdf/p620_ug_en.pdf)).

As long as users use the front USB ports, they encounter no issue. Hotplug is working as expected. However if users use the back USB 3.X ports,
there is a problem with hotplugging (weirdly, the USB 2.0 on the back are fine). Note that several users have reported the same issue when using
these ports.

The end-user problem is that if the unplug a device and replug it in within 10s, the device is not detected by libusb.  

I have tracked down the problem to netlink events coming late (or not at all). When a user unplug a device from the problematic port,
here is the netlink messages libusb gets:

```
[2763.143728] [002e0835] libusb: debug [dump_message] unbind@/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1/2-1:1.0
[2763.143735] [002e0835] libusb: debug [dump_message] ACTION=unbind
[2763.143741] [002e0835] libusb: debug [dump_message] DEVPATH=/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1/2-1:1.0
[2763.143747] [002e0835] libusb: debug [dump_message] SUBSYSTEM=usb
[2763.143753] [002e0835] libusb: debug [dump_message] DEVTYPE=usb_interface
[2763.143757] [002e0835] libusb: debug [dump_message] PRODUCT=18d1/4ee7/510
[2763.143762] [002e0835] libusb: debug [dump_message] TYPE=0/0/0
[2763.143769] [002e0835] libusb: debug [dump_message] INTERFACE=255/66/1
[2763.143774] [002e0835] libusb: debug [dump_message] SEQNUM=8528
.
. 9 seconds later
.
[2772.636012] [002e0835] libusb: debug [dump_message] remove@/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1/2-1:1.0
[2772.636016] [002e0835] libusb: debug [dump_message] ACTION=remove
[2772.636023] [002e0835] libusb: debug [dump_message] DEVPATH=/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1/2-1:1.0
[2772.636029] [002e0835] libusb: debug [dump_message] SUBSYSTEM=usb
[2772.636034] [002e0835] libusb: debug [dump_message] DEVTYPE=usb_interface
[2772.636072] [002e0835] libusb: debug [dump_message] PRODUCT=18d1/4ee7/510
[2772.636077] [002e0835] libusb: debug [dump_message] TYPE=0/0/0
[2772.636083] [002e0835] libusb: debug [dump_message] INTERFACE=255/66/1
[2772.636089] [002e0835] libusb: debug [dump_message] MODALIAS=usb:v18D1p4EE7d0510dc00dsc00dp00icFFisc42ip01in00
[2772.636096] [002e0835] libusb: debug [dump_message] SEQNUM=8529
...
[2772.636341] [002e0835] libusb: debug [dump_message] unbind@/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1
[2772.636386] [002e0835] libusb: debug [dump_message] ACTION=unbind
[2772.636391] [002e0835] libusb: debug [dump_message] DEVPATH=/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1
[2772.636403] [002e0835] libusb: debug [dump_message] SUBSYSTEM=usb
[2772.636410] [002e0835] libusb: debug [dump_message] MAJOR=189
[2772.636420] [002e0835] libusb: debug [dump_message] MINOR=179
[2772.636426] [002e0835] libusb: debug [dump_message] DEVNAME=bus/usb/002/052
[2772.636435] [002e0835] libusb: debug [dump_message] DEVTYPE=usb_device
[2772.636440] [002e0835] libusb: debug [dump_message] PRODUCT=18d1/4ee7/510
...
[2772.636558] [002e0835] libusb: debug [dump_message] remove@/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1
[2772.636563] [002e0835] libusb: debug [dump_message] ACTION=remove
[2772.636567] [002e0835] libusb: debug [dump_message] DEVPATH=/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1
[2772.636572] [002e0835] libusb: debug [dump_message] SUBSYSTEM=usb
[2772.636578] [002e0835] libusb: debug [dump_message] MAJOR=189
[2772.636583] [002e0835] libusb: debug [dump_message] MINOR=179
[2772.636588] [002e0835] libusb: debug [dump_message] DEVNAME=bus/usb/002/052
[2772.636594] [002e0835] libusb: debug [dump_message] DEVTYPE=usb_device
[2772.636599] [002e0835] libusb: debug [dump_message] PRODUCT=18d1/4ee7/510
[2772.636605] [002e0835] libusb: debug [dump_message] TYPE=0/0/0
[2772.636610] [002e0835] libusb: debug [dump_message] BUSNUM=002
[2772.636615] [002e0835] libusb: debug [dump_message] DEVNUM=052
[2772.636621] [002e0835] libusb: debug [dump_message] SEQNUM=8531
```

Sidenote: `unbind` is ignore. libusb only uses `remove` messages.

Notice how the first "remove" netlink messages took 9s to arrive. This reproduces 100%.

If the user replug the device before the `remote` event arrives, no netlink event are received beside the first `unbind` (which is ignored). See the trace.

```
[1987.184003] [002f1bab] libusb: debug [usbi_wait_for_events] poll() returned 1
[1987.184035] [002f1ba9] libusb: debug [linux_netlink_parse] ****** GOT NETLINK
[1987.184047] [002f1ba9] libusb: debug [dump_message] _start_
[1987.184075] [002f1ba9] libusb: debug [dump_message] unbind@/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1/2-1:1.0
[1987.184081] [002f1ba9] libusb: debug [dump_message] ACTION=unbind
[1987.184088] [002f1ba9] libusb: debug [dump_message] DEVPATH=/devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-1/2-1:1.0
[1987.184093] [002f1ba9] libusb: debug [dump_message] SUBSYSTEM=usb
[1987.184098] [002f1ba9] libusb: debug [dump_message] DEVTYPE=usb_interface
[1987.184103] [002f1ba9] libusb: debug [dump_message] PRODUCT=18d1/4ee7/510
[1987.184109] [002f1ba9] libusb: debug [dump_message] TYPE=0/0/0
[1987.184113] [002f1ba9] libusb: debug [dump_message] INTERFACE=255/66/1
[1987.184119] [002f1ba9] libusb: debug [dump_message] SEQNUM=8678
[1987.184124] [002f1ba9] libusb: debug [dump_message] _end_
[1987.184130] [002f1ba9] libusb: debug [linux_netlink_parse] unknown device action unbind
[1989.902003] [002f1bab] libusb: debug [reap_for_handle] urb type=3 status=-108 transferred=0
[1989.902023] [002f1bab] libusb: debug [handle_bulk_completion] handling completion status -108 of bulk urb 1/1
[1989.902028] [002f1bab] libusb: debug [handle_bulk_completion] device removed
[1989.902037] [002f1bab] libusb: debug [usbi_handle_transfer_completion] transfer 0x560d46bfb4c0 has callback 0x560d13ffd7b0
[1989.902089] [002f1bab] libusb: debug [libusb_free_transfer] transfer 0x560d46bfb4c0
[1989.902103] [002f1bab] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[1989.902110] [002f1bab] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[1989.902204] [002f1ba7] libusb: debug [libusb_free_transfer] transfer 0x560d46bfb080
[1989.902210] [002f1ba7] libusb: debug [libusb_release_interface] interface 0
[1989.902221] [002f1ba7] libusb: error [release_interface] release interface failed, errno=22
[1989.902242] [002f1ba7] libusb: debug [libusb_close]  
[1989.902267] [002f1bab] libusb: debug [usbi_wait_for_events] poll() returned 1
[1989.902275] [002f1bab] libusb: debug [handle_event_trigger] event triggered
[1989.902281] [002f1bab] libusb: debug [handle_event_trigger] someone is closing a device
[1989.902291] [002f1bab] libusb: debug [libusb_try_lock_events] someone else is closing a device
[1989.902297] [002f1bab] libusb: debug [libusb_event_handler_active] someone else is closing a device

```

Notice how `usbi_handle_transfer_completion` completed (with `LIBUSB_TRANSFER_NO_DEVICE`). This made our app stop using the device. But since netlink never gets the `add` event, even thought the device is physically connected, the hotplug callback never triggers. This device can no longer be accessed.

1. Have we ever encountered this kind of behavior where netlink events are slow to arrive?
2. Since we get the `LIBUSB_TRANSFER_NO_DEVICE` upon transfer, we could remove the device from the known-devices in libusb core. But since the netlink message
for the device being added will never be issued, it does not solve the problem. A solution would be to have a hotplug system based on scanning `/dev/bus/usb`. This
is what our [old-stack](https://cs.android.com/android/platform/superproject/main/+/main:packages/modules/adb/client/usb_linux.cpp;l=1;drc=7799b4ba18dd0fc0298ccac65014ca533dacbddd) uses and it is resilient to the problem described above.
3. Could it be that the workstation back ports are just "misconfigured" via lpm or ltp?

mcuee, 2024-10-10:
udev should be available in most of the Linux machines.

tormodvolden, 2024-10-10:
Can you find out if it is netlink that is slow or if there is possibly an issue in libusb? There are some netlink debug suggestions e.g. at https://jvns.ca/blog/2017/09/03/debugging-netlink-requests/

fabiensanglard, 2024-10-10:
With further investigation, it does not look like it is a problem specific to `netlink`. Even udev events are delayed (and if the device is replugged fast enough, nothing else comes up).

```
# <--- Plug device 
$ udevadm monitor
# <-- Unplug device
KERNEL[760401.164215] unbind   /devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-2/2-2:1.0 (usb)
# <-- Nothing for 10s
KERNEL[760410.484517] remove   /devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-2/2-2:1.0 (usb)
KERNEL[760410.484859] unbind   /devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-2 (usb)
KERNEL[760410.484892] remove   /devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-2 (usb)
UDEV  [760410.489278] unbind   /devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-2/2-2:1.0 (usb)
UDEV  [760410.493611] remove   /devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-2/2-2:1.0 (usb)
UDEV  [760415.801758] unbind   /devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-2 (usb)
UDEV  [760415.808128] remove   /devices/pci0000:20/0000:20:08.1/0000:23:00.3/usb2/2-2 (usb)
```

Meanwhile, the read bulk transfer completes with an error. I think to go deeper I need to find out where the urb completion callback happens in the kernel. The [documentation](https://docs.kernel.org/driver-api/usb/URB.html) does much about how the completion handler runs, or even which thread.

fabiensanglard, 2024-10-19:
At this point, it seems it is an error with the firmware/drivers on the machine.

One last question though. `udev` gets its events from `netlink` yet, it is not guaranteed to be available on a Linux machine. Why does `libusb` even have a `udev` backend for hotplug events? Why not always use `netlink`?

mcuee, 2024-10-19:
netlink backend is a fall-back and second class citizen when it comes to hotplug under Linux. Initially there was no netlink fall-back but later it was added.

fabiensanglard, 2024-10-21:
Upon chatting with @hjelmn (who wrote the udev part), he explained that although `udev` receives its event from `netlink`, it is in charge of creating the entries in `/dev/bus/usb` and `sysfs`.

So udev is indeed better because listening to netlink could lead to a race condition where `udev` has no created the nodes in `/dev/bus/usb` yet or has no set the file permission to allow non-root users to access them.

closed 2024-10-22 by fabiensanglard