libusb/libusb #1572: Assertion failed: fd != NULL in poll_windows.c line 145

bognar-dev, 2024-10-10:
I am encountering an assertion failure when running my application that uses libusb. The error message is as follows:

```Assertion failed: fd != NULL, file C:\BuildAgent\work\9c51a5e2c1a6d1aa\src\libusb\libusb-1.0.22\libusb-1.0.22\libusb\os\poll_windows.c, line 145```

[image]

Steps to Reproduce:  
Initialize libusb context.
Open a USB device using libusb_open_device_with_vid_pid.
Perform operations with the device.
Expected Behavior: The application should run without assertion failures.  Actual Behavior: The application crashes with the assertion failure mentioned above.  

Code Snippet:

```c++
void CLibUsbThread::run() {
	emit debug("<CLibUsbThread> Thread started");
	if (!m_UsbHandle) {
		m_status = LIBUSB_TRANSFER_NO_DEVICE;
		emit debug("<CLibUsbThread> No USB device handle");
		return;
	}

	unsigned char buffer[USB_BULK_PACKET_SIZE];
	struct libusb_transfer* transfer = libusb_alloc_transfer(0);
	if (!transfer) {
		m_status = LIBUSB_TRANSFER_CANCELLED;
		emit debug("<CLibUsbThread> Failed to allocate transfer, status: " + libusbTransferStatusToString(transfer->status));
		return;
	}

	libusb_fill_bulk_transfer(transfer, m_UsbHandle, m_EPin, buffer, USB_BULK_PACKET_SIZE, callback_wrapper, this, USB_BULK_READ_TIMEOUT);
	emit debug("<CLibUsbThread> Bulk transfer filled, submitting transfer");

	int result = libusb_submit_transfer(transfer);
	if (result < 0) {
		emit debug("<CLibUsbThread> Failed to submit transfer, result: " + libusbErrorToString(result));
		libusb_free_transfer(transfer);
		return;
	}

	while (!m_StopThread) {
		int completed = 0;
		libusb_handle_events_completed(nullptr, &completed);
		if (m_status == LIBUSB_TRANSFER_TIMED_OUT) {
			emit debug("<CLibUsbThread> Transfer timeout");
		}
		else if (m_status == LIBUSB_TRANSFER_CANCELLED) {
			emit debug("<CLibUsbThread> Transfer cancelled");
			m_StopThread = true;
		}
		QCoreApplication::processEvents(QEventLoop::AllEvents, 20);
	}

	emit debug("<CLibUsbThread> Stopping and freeing transfer");
	libusb_cancel_transfer(transfer);
	libusb_free_transfer(transfer);
	emit debug("<CLibUsbThread> Thread run loop exited");
}

```

OS: Windows
libusb version: 1.0.22
Compiler: VS2019 V16 8.1 CMake 3.29

bognar-dev, 2024-10-10:
> What operations do you mean with
> 
> > Perform operations with the device
> 
> ?

Unplug and Plug a device, while you have more than one device connected.

Sorry if I was unclear

mcuee, 2024-10-10:
```
OS: Windows
libusb version: 1.0.22
Compiler: VS2019 V16 8.1 CMake 3.29
```

Please upgrade to latest 1.0.27 release version. Super old version like 1.0.22 will not be supported.

bognar-dev, 2024-10-10:
I Updated my version and now I get:
libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0 libusb: debug [handle_event_trigger] event triggered

Process finished with exit code -1,073,741,819.

tormodvolden, 2024-10-10:
So you get a debug warning now. Does it crash or report an assert any longer? Can you please provide a debug log?

mcuee, 2024-10-10:
@bognar-dev 
Please also take note libusb Windows does not support hotplug now. That may make the application a bit complicated if you want to deal with hotplug case.

Please give this PR a try as well.
https://github.com/libusb/libusb/pull/1406

Or maybe you can look at this PR to see if the idea helps or not.
https://github.com/libusb/libusb/pull/1498

mcuee, 2025-03-23:
Close first because of inactivity. Please re-open if there is new info pointing out to a potential libusb problem.

closed 2025-03-23 by mcuee