libusb/libusb #1573: libusb.open() on Ubuntu under WSL2 hangs when accessing USB devices without permission multiple times

jacob-lerenius, 2024-10-10:
How to reproduce:
On a Windows system running  WSL2, forward the USB device to WSL2 using usbipd attach --wsl --busid x-y, where x and y is the busID
call libusb.open() on a device without permission several times in Linux. The first times it correctly returns with error -3 and no handle is set, but the thread hangs after some time doing this. 

Proc stack:
[<0>] pipe_write+0x378/0x650
[<0>] new_sync_write+0x197/0x1b0
[<0>] vfs_write+0x1d8/0x270
[<0>] ksys_write+0xab/0xe0
[<0>] do_syscall_64+0x38/0xc0
[<0>] entry_SYSCALL_64_after_hwframe+0x62/0xcc

gdb:
#0  __GI___libc_write (nbytes=78, buf=0x7f4faa1ce360, fd=2) at ../sysdeps/unix/sysv/linux/write.c:26
#1  __GI___libc_write (fd=2, buf=0x7f4faa1ce360, nbytes=78) at ../sysdeps/unix/sysv/linux/write.c:24
#2  0x00007f4faaa8beed in _IO_new_file_write (f=0x7f4faac1c6a0 <_IO_2_1_stderr_>, data=0x7f4faa1ce360, n=78) at ./libio/fileops.c:1180
#3  0x00007f4faaa8c7d0 in new_do_write (to_do=<optimized out>, data=0x7f4faa1ce360 "libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes\n", fp=0x7f4faac1c6a0 <_IO_2_1_stderr_>) at ./libio/libioP.h:947
#4  _IO_new_file_xsputn (n=78, data=<optimized out>, f=<optimized out>) at ./libio/fileops.c:1254
#5  _IO_new_file_xsputn (f=0x7f4faac1c6a0 <_IO_2_1_stderr_>, data=<optimized out>, n=78) at ./libio/fileops.c:1196
#6  0x00007f4faaa80a9d in __GI__IO_fputs (str=0x7f4faa1ce360 "libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes\n", fp=0x7f4faac1c6a0 <_IO_2_1_stderr_>) at ./libio/libioP.h:947
#7  0x00007f4faaf626b2 in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
#8  0x00007f4faaf627aa in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
#9  0x00007f4faaf66384 in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
#10 0x00007f4faaf681f9 in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
#11 0x00007f4faaf67dc5 in libusb_open () from /lib/x86_64-linux-gnu/libusb-1.0.so.0

mcuee, 2024-10-10:
WSL does not support USB natively. How do you get USB supoort?

jacob-lerenius, 2024-10-10:
libusb works well with WSL using usbipd when permission is set up correctly but this is not always the case and in that situation, it is not good when the method call never returns and hangs the thread.

tormodvolden, 2024-10-10:
Is the above trace from when it is hanging? So it is actually stuck on writing the error message?

tormodvolden, 2024-10-10:
I don't expect any changes, but it would be nice if you could use 1.0.27 or git master.

tormodvolden, 2024-10-10:
In that case it looks like a WSL issue. libusb is blocked while writing the error message on stderr.

jacob-lerenius, 2024-10-10:
> I don't expect any changes, but it would be nice if you could use 1.0.27 or git master.

I use the latest version in the repo under my WSL Ubuntu distro, do you have info about how do I update it to 1.0.27 or git master?

tormodvolden, 2024-10-10:
You can download sources, build and install inside WSL like you would on a normal Linux installation.

jacob-lerenius, 2024-10-10:
> In that case it looks like a WSL issue. libusb is blocked while writing the error message on stderr.

Ok, not sure what you want to do from here then. At least the issue is reported. I will try to build from source and replace the default driver but as you say, probably no difference.

jacob-lerenius, 2024-10-10:
Issue found relating to how Java created the process and stderr pipe was never read in that case in Linux. Sorry to take up your time but hopefully may help others facing similar issues :)

closed 2024-10-10 by jacob-lerenius

tormodvolden, 2024-10-10:
I hope they will then at least mention they are wrapping it in Java...