libusb/libusb #1576: libusb on Android

sebinho, 2024-10-16:
Dear All,

I know this topic has been discussed before but I am bringing it back to life because I am trying to implement something on an Android device but I cannot get it to work.
To give a bit of background, I did build an app using Kivy Framework (the app is written in Python). I am able to talk to USB devices if I access the JAVA classes directly from Python using usb4a (via pyjnius).

But now I would like to access USB devices using a native C++ project. This project is using libusb to talk to USB devices. I tried to port the code from [here](https://github.com/ibaiGorordo/libusbNativeAndroidTest/tree/main). But I get the following errors in my logcat:

```
10-16 19:34:12.742  6950  6950 I SDLThread: type=1400 audit(0.0:2428): avc: denied { read } for name="usb" dev="tmpfs" ino=334 scontext=u:r:untrusted_app:s0:c103,c256,c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=1 app=org.test.myapp
10-16 19:34:12.742  6950  6950 I SDLThread: type=1400 audit(0.0:2429): avc: denied { open } for path="/dev/bus/usb" dev="tmpfs" ino=334 scontext=u:r:untrusted_app:s0:c103,c256,c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=1 app=org.test.myapp
10-16 19:34:12.742  6950  6950 I SDLThread: type=1400 audit(0.0:2431): avc: denied { open } for path="/sys/bus/usb/devices" dev="sysfs" ino=24050 scontext=u:r:untrusted_app:s0:c103,c256,c512,c768 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1 app=org.test.myapp
```

It seems to be some kind of permission issue from SELinux, but not sure what I can do....

Can someone help? 
Is there any progress regarding libusb on Android?

Many thanks in advance

T-Dynamos, 2024-10-21:
Perhaps you can try: https://github.com/libusb/libusb/pull/1164#issuecomment-2423942561.

wuwbobo2021, 2024-10-25:
I have encountered another problem: `libusb_wrap_sys_device()` failed with ioctl errno 9 returned by `ioctl(fd, IOCTL_USBFS_CONNECTINFO, &ci);` in `op_wrap_sys_device()`, `linux_usbfs.c` in `libusb` 1.0.27.

<https://github.com/kevinmehall/nusb/issues/86>

wuwbobo2021, 2024-11-11:
Well, I failed with `rusb` (libusb wrapper in Rust) but made `nusb` working on Android: <https://crates.io/crates/android-usbser>. Really strange.

wyc-go-for-it, 2025-03-22:
Android 12 libusb_init return -1

mcuee, 2025-03-22:
> Android 12 libusb_init return -1

Please try PR #1164.