libusb/libusb #1579: [Windows 11] Issue USB device detection on 1.0.27 - Works fine on 1.0.26

JackoboLeChocobo, 2024-10-23:
Hi all,

I met a detection issue on libusb 1.0.27 but the code works on libusb 1.0.26.

Here is the code used : 

```
int Detect_Device(void)
{
    /* Initialise libusb. */
    SDL_Log("Init LibUSB... \n");
    res = libusb_init(0);
    if (res != 0)
    {
        SDL_Log("Error initialising libusb.\n");
        return 1;
    }

    SDL_Log("LibUSB Init Sucessfully ! \n");

    SDL_Log("Detecting MD Dumper... \n");

    /* Get the first device with the matching Vendor ID and Product ID. If
     * intending to allow multiple demo boards to be connected at once, you
     * will need to use libusb_get_device_list() instead. Refer to the libusb
     * documentation for details. */

    handle = libusb_open_device_with_vid_pid(0, 0x0483, 0x5740);

    if (!handle)
    {
        SDL_Log("Unable to open device.\n");
        return 1;
    }

    /* Claim interface #0. */

    if(libusb_kernel_driver_active(handle, 0) == 1)
    {
        SDL_Log("Kernel Driver Active");
        if(libusb_detach_kernel_driver(handle, 0) == 0)
            SDL_Log("Kernel Driver Detached!");
        else
        {
            SDL_Log("Couldn't detach kernel driver!\n");
            libusb_close(handle);
        }
    }
    res = libusb_claim_interface(handle, 0);
    //SDL_Log("Test 1 : %d\n",res);
    if (res != 0)
    {
        if(libusb_kernel_driver_active(handle, 1) == 1)
        {
            SDL_Log("Kernel Driver Active");
            if(libusb_detach_kernel_driver(handle, 1) == 0)
                SDL_Log("Kernel Driver Detached!");
            else
            {
                SDL_Log("Couldn't detach kernel driver!\n");
                libusb_close(handle);
            }
        }
        res = libusb_claim_interface(handle, 1);
        //SDL_Log("Test 2 : %d\n",res);
        if (res != 0)
        {
            SDL_Log("Error claiming interface.\n");
            return 1;
        }
    }
```

libusb_claim_interface() on 1.0.27 returns always -5 as value.

Do you have an idea to help me to understand the issue ?

JackoboLeChocobo, 2024-10-28:
After applied suggested modifications according to documentation, nothing again ... :/

Here is the new code used now : 

```
   /* Initialise libusb. */
    SDL_Log("Init LibUSB... \n");
    
    res = libusb_init(&context);
    if (res != 0)
    {
        SDL_Log("Error initialising libusb.\n");
        return 1;
    }

    SDL_Log("LibUSB Init Sucessfully ! \n");

    SDL_Log("Detecting MD Dumper... \n");
	
	count = libusb_get_device_list(context, &devs);
	if (count <= 0)
    {
        SDL_Log("Error getting device list\n");
        return 1;
    }

	for (size_t idx = 0; idx < count; ++idx)
		{
		libusb_device *device = devs[idx];
		struct libusb_device_descriptor desc = {0};

		res = libusb_get_device_descriptor(device, &desc);
		assert(res == 0);

		if(desc.idVendor=VENDOR_ID && desc.idProduct==PRODUCT_ID) device_found=idx;
		}
	
	if(device_found!=-1)
		{
		SDL_Log("MD Dumper Device Found !\n");
		libusb_device *device = devs[device_found];
		struct libusb_device_descriptor desc = {0};
		res = libusb_get_device_descriptor(device, &desc);

		SDL_Log("LibUSB Device ID = %d\n",device_found);
		SDL_Log("LibUSB Device Vendor = %04x:%04x\n",desc.idVendor,desc.idProduct);
		
		libusb_open(device, &handle);

		if (!handle)
			{
			SDL_Log("Unable to open MD Dumper Device.\n");
			return 1;
			}
		else
			{
			SDL_Log("MD Dumper Device opened !\n");
			}
		}
	else
		{
		SDL_Log("MD Dumper Device Not Found !\n");
		return 1;
		}
	
	int if_num_max=2;
	for (if_num = 0; if_num < if_num_max; if_num++) {
        if (libusb_kernel_driver_active(handle, if_num))
			{
            res = libusb_detach_kernel_driver(handle, if_num);
			}
        res = libusb_claim_interface(handle, if_num);
        if (res < 0) 
			{
            SDL_Log("Error claiming interface %d: %s\n", if_num, libusb_error_name(res));
			if(if_num==if_num_max-1)
				{
				SDL_Log("Exiting...");
				return 1;
				}
			}
		else
			{
            SDL_Log("Interface %d claimed\n", if_num);
			if_num=2;
			}
    }
```

JackoboLeChocobo, 2024-10-28:
[image]

Image of what we saw on Windows 11 ( I added A,B,C,D texts to check if apps freezes or not )

mcuee, 2024-10-29:
Please post the full debug log.

JackoboLeChocobo, 2024-10-30:
When i can do it i will post the info.

Here is the USBView from Windows Dev Tool of the device : 

```
[Port17]  :  MD Dumper STM32F1

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            2
Companion Hub Symbolic Link Name: USB#ROOT_HUB30#5&1e461e56&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "MD Dumper STM32F1"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x01
Open Pipes:                           2

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0200
bDeviceClass:                      0xFF  -> This is a Vendor Specific Device
bDeviceSubClass:                   0x00
bDeviceProtocol:                   0x00
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x0483 = STMicroelectronics
idProduct:                       0x5740
bcdDevice:                       0x0200
iManufacturer:                     0x01
     English (United States)  "Sege4ever 2023"
iProduct:                          0x02
     English (United States)  "MD Dumper STM32F1"
iSerialNumber:                     0x03
     English (United States)  "36FFD4053059583229570943"
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x01

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x82  -> Direction: IN - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x01

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x02
wTotalLength:                    0x0020  -> Validated
bNumInterfaces:                    0x01
bConfigurationValue:               0x01
iConfiguration:                    0x00
bmAttributes:                      0x80  -> Bus Powered
MaxPower:                          0x32 = 100 mA

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x01
bAlternateSetting:                 0x00
bNumEndpoints:                     0x02
bInterfaceClass:                   0xFF  -> Interface Class Unknown to USBView
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x01

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x82  -> Direction: IN - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x01
```

JackoboLeChocobo, 2024-11-01:
Bug found : Tested on 1.0.27 release source code
get_interface_descriptor_by_number() doesn't work actually. Using 1.0.26 code works like a charm

```
/*  ISSUE
 * 
 * 	if_desc = get_interface_descriptor_by_number(dev_handle, conf_desc, iface, altsetting);
	if (if_desc == NULL) {
		r = LIBUSB_ERROR_NOT_FOUND;
		goto end;
	}
*/

	if (iface >= conf_desc->bNumInterfaces) {
		usbi_err(HANDLE_CTX(dev_handle), "interface %d out of range for device", iface);
		return LIBUSB_ERROR_NOT_FOUND;
	}
	if_desc = &conf_desc->interface[iface].altsetting[altsetting];
```

tormodvolden, 2024-11-02:
From your USBView dump it seems the device has one interface, with bInterfaceNumber = 1.

Do you get errors if you use your code with if_num = 1 ?

tormodvolden, 2024-11-02:
v1.0.26 had a bug on Windows and macos where it was not using the interface number correctly, and it was fixed in #1191.

JackoboLeChocobo, 2024-11-03:
1.0.27 returned error when I tried to claim interface. If i apply the modification like indicated before, I can use the device on 1.0.27.
1.0.26 worked without issue when using the device.

Here is the screen of the software working with the 1.0.27 modified version used
[image]

tormodvolden, 2024-11-03:
That was not my question. Do you get errors if you use your code with if_num = 1 ?

Your USB device is unusual since it uses bInterfaceNumber = 1 for the first interface, whereas usually 0 would be used.

I think 1.0.27 will work fine if you follow the API. And that 1.0.26 worked for you even if you used if_num = 0, which is wrong since the bInterfaceNumber is 1, because 1.0.26 on Windows wrongly used the number as an index instead of matching the bInterfaceNumber.

mcuee, 2024-11-03:
I believe the device does not meet USB specifications.

https://www.usb.org/document-library/usb-20-specification
USB 2.0 Spec: Table 9-12.
**bInterfaceNumber**
Number of this interface. **Zero-based value** identifying the index in the array of concurrent interfaces supported by this
configuration.

mcuee, 2024-11-03:
@JackoboLeChocobo 
Please report this issue to the FW developer o the device. Thanks.

JackoboLeChocobo, 2024-11-03:
@mcuee as Contributor of his software, Developer of the device ( card, FW and Software too ) has the link of this ticket ( I added it on my pull request for the patch ). I added the information on his Discord to see if according to him the issue is on his FW. I will keep the solution in all case and adaptés it for all possible issues for people.

I add again that the card actually on Linux with the 1.0.27 version without the modification ( because it concerns Windows only ) works without issue with the current FW on the device.

tormodvolden, 2024-11-08:
It is interesting that it works on Linux, because the Windows change in 1.0.27 should have made it work the same as on Linux. Can you please paste or attach the output from `lsusb -v -d 0483:5740` ?

closed 2025-09-20 by mcuee