libusb/libusb #1580: Some assembly problems

sevstels, 2024-11-05:
1>core.c
1>src\LibUSB\libusb\core.c(652,26): error : cannot initialize a variable of type 'struct discovered_devs *' with an rvalue of type 'void *'
1>        struct discovered_devs *ret =
1>                                ^
1>src\LibUSB\libusb\core.c(693,17): error : assigning to 'struct discovered_devs *' from incompatible type 'void *'
1>        new_discdevs = realloc(discdevs,
1>                       ^~~~~~~~~~~~~~~~~
1>src\LibUSB\libusb\core.c(714,24): error : cannot initialize a variable of type 'struct libusb_device *' with an rvalue of type 'void *'
1>        struct libusb_device *dev = calloc(1, PTR_ALIGN(sizeof(*dev)) + priv_size);
1>                              ^     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>src\LibUSB\libusb\core.c(866,8): error : assigning to 'struct libusb_device **' from incompatible type 'void *'
1>        ret = calloc((size_t)len + 1, sizeof(struct libusb_device *));
1>              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>src\LibUSB\libusb\core.c(1364,16): error : assigning to 'struct libusb_device_handle *' from incompatible type 'void *'
1>        _dev_handle = calloc(1, PTR_ALIGN(sizeof(*_dev_handle)) + priv_size);
1>                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>src\LibUSB\libusb\core.c(1418,16): error : assigning to 'struct libusb_device_handle *' from incompatible type 'void *'
1>        _dev_handle = calloc(1, PTR_ALIGN(sizeof(*_dev_handle)) + priv_size);
1>                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>src\LibUSB\libusb\core.c(2045,10): error : cannot initialize return object of type 'unsigned char *' with an rvalue of type 'void *'
1>                return usbi_backend.dev_mem_alloc(dev_handle, length);
1>                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>src\LibUSB\libusb\core.c(2301,15): warning : passing an object that undergoes default argument promotion to 'va_start' has undefined behavior [-Wvarargs]
1>        va_start(ap, option);
1>                     ^
1>src\LibUSB\libusb\core.c(2292,21):  note: parameter of type 'enum libusb_option' is declared here
1>        enum libusb_option option, ...)
1>                           ^
1>src\LibUSB\libusb\core.c(2452,9): error : assigning to 'struct libusb_context *' from incompatible type 'void *'
1>        _ctx = calloc(1, PTR_ALIGN(sizeof(*_ctx)) + priv_size);
1>               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>src\LibUSB\libusb\core.c(2474,26): error : cannot initialize a variable of type 'enum libusb_option' with an rvalue of type 'int'
1>        for (enum libusb_option option = 0 ; option < LIBUSB_OPTION_MAX ; option++) {

mcuee, 2024-11-05:
What is the compiler used here?

sevstels, 2024-11-05:
1>linux_usbfs.c
1>src\LibUSB\lib\linux_usbfs.c(2551,8): error : case value evaluates to -18, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -EXDEV:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2550,8): error : case value evaluates to -63, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -ENOSR:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2549,8): error : case value evaluates to -70, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -ECOMM:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2548,8): error : case value evaluates to -84, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -EILSEQ:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2547,8): error : case value evaluates to -71, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -EPROTO:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2546,8): error : case value evaluates to -62, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -ETIME:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2542,8): error : case value evaluates to -75, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -EOVERFLOW:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2538,8): error : case value evaluates to -32, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -EPIPE:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2534,8): error : case value evaluates to -108, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -ESHUTDOWN:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2533,8): error : case value evaluates to -19, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -ENODEV:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2531,8): error : case value evaluates to -104, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -ECONNRESET:
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2530,8): error : case value evaluates to -2, which cannot be narrowed to type 'unsigned int' [-Wc++11-narrowing]
1>                case -ENOENT: /* cancelled */
1>                     ^
1>src\LibUSB\lib\linux_usbfs.c(2670,9): error : no matching function for call to 'usbi_handle_transfer_completion'
1>        return usbi_handle_transfer_completion(itransfer, status);
1>               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>src\LibUSB\lib/libusbi.h(819,5):  note: candidate function not viable: no known conversion from 'int' to 'enum libusb_transfer_status' for 2nd argument
1>int usbi_handle_transfer_completion(struct usbi_transfer *itransfer,
1>    ^

mcuee, 2024-11-05:
> I'm trying to build an Android application in Visual Studio 2019. Clang compiler. This is only a small part of the critical errors.... and then it gets worse. The code uses a lot of messy styling. Which probably causes problems with the library's stability.
> 
> Microsoft Visual Studio Professional 2019 Version 16.11.41 Visual C++ for Cross Platform Mobile Development (Android) 16.0.35213.202

This is not supported. You are on your own in this case.

sevstels, 2024-11-05:
What does that have to do with being supported or not? I'm just telling you where the mistakes are. You can see for example in the last listing unsigned int is trying to compare with negative numbers!

mcuee, 2024-11-06:
@tormodvolden 

Please take a look at this. Thanks.

sevstels, 2024-11-06:
Thanks, I'll look it up a bit later. I compiled your version with very significant edits. But haven't tested how it works yet. I will post the result within a week.

tormodvolden, 2024-11-06:
These are warnings you will get if you compile it with a C++ compiler. Sean already has suggested some patches for this.

It does look weird that urb_desc->status which is unsigned integer is compared to negative constants, but this is a Linux oddity. See https://www.kernel.org/doc/html/latest/driver-api/usb/usb.html#asynchronous-i-o-support where usbdevfs_iso_packet_desc.status is unsigned int, whereas it goes on to say "usb_iso_packet_descriptor’s status field will return -EXDEV". Or https://www.kernel.org/doc/html/latest/driver-api/usb/error-codes.html#error-codes-returned-by-in-urb-status-or-in-iso-frame-desc-n-status-for-iso where the same error codes are used for usb.status, which is int.

In practice it doesn't matter and it doesn't affect the library's stability.

sevstels, 2026-04-20:
Thank dear. But it's been two years already. I finished everything a long time ago. After rewriting a significant portion of the code by hand. :)