libusb/libusb #1583: libusb returns two different values for device descriptor when use two different functions

dragbahs, 2024-11-13:
I want to use libusb library to communicate with an HID device. I could implement many of its functions and it was ok. but when I want to get the device descriptor with ```libusb_get_device_descriptor()``` and ```libusb_get_descriptor()``` functions, it shows me different values for some fields like **Control Max Packet Size**.

I've coded as following for two functions:

```
#include <stdio.h>
#include "libusb.h"

int main(void)
{
  libusb_device_handle* dev_handle;
  int init = libusb_init(NULL);
  dev_handle = libusb_open_device_with_vid_pid(NULL, 0x04d8, 0x01ff);
  libusb_claim_interface(dev_handle, 0);

  unsigned char buf[18];
  
  //first function
  libusb_get_descriptor(dev_handle, LIBUSB_DT_DEVICE, 0, buf, sizeof(buf));

  for (int i = 0; i < sizeof(buf); i++)
  {
    printf("%d ", buf[i]);
  }

// output
//18 1 0 2 0 0 0 64 216 4 255 1 0 1 1 2 0 1
  
  printf("\n");
  
  //second function
  struct libusb_device_descriptor desc;
  libusb_get_device_descriptor(libusb_get_device(dev_handle), &desc);
  printf("%d ", desc.bMaxPacketSize0);

  libusb_release_interface(dev_handle, 0);
  libusb_close(dev_handle);
  libusb_exit(NULL);
  return 0;
}
```

If you run the above code, the returned value for **CTRL_Max_Packet_Size** from the first function is equal to ```64``` but for the second function is ```8``` that obviously the second one is correct.  another issue is for the ```bcdDevice``` field. the first function returns ```0 1``` values and the second function returns ```1 0``` values. in this case the second function is correct and the first one is wrong. ( the evidence is that the two functions have returned the vid(0x04d8) value as ```216 4``` not ```4 216``` )

I'm using Windows 10, 64bit edition and I have compiled my code with both visual studio and mingw64. both return same incorrect values.

tormodvolden, 2024-11-13:
Hi, thanks for your report. The reason they may be different is that the first returns cached values from the library, whereas the second actually requests the descriptor from the device and passes it on as-is. In the case of HID devices, the cached values has up to now been partially "guessed" during enumeration. However, we changed this in commit 4528752 (see PR #1378) but this hasn't been released yet. Can you please try libusb from latest git, and see if that works correctly? The older code in v1.0.27 will hardcode 64 and 0x100 for the values you mention. I am not sure about the inverse byte order, this can be a different issue that we'd need to fix.

dragbahs, 2024-11-13:
Thanks for your reply.  I found that my default driver for HID device was ```hidUsb``` and according to the documentation It should be WinUsb. Unfortunately I did not have notice to this subject and the problem occurred. I changed the driver to winUsb using the ```Zadig``` software and now the issue is solved.

tormodvolden, 2024-11-13:
OK, but I am still curious if latest git master would work with the hidUsb driver.

dragbahs, 2024-11-13:
I will check in the next few days and report.

dragbahs, 2024-11-14:
Hi, I downloaded the latest version of ```libusb``` from your Git repository and selected two HID devices to test the communication. One device is an HID mouse, and the other is an HID programmer from Microchip Technology Inc. Unfortunately, the previous issue still persists.

[image]

According to the above image, the ```bMaxPacketSize``` value is always incorrect. However, for some devices, the ```bcdDevice``` value may be incorrect, while for others, the ```bcdUSB``` or other values may be incorrect.

I've checked the two devices with ```USBTrace``` software, and for both of them, the values returned by the second function are completely correct. However, for the first function, the returned values are not reliable. The image below shows the information provided by ```USBTrace``` software for the first device (an HID mouse).

[image]

All values completely match the values returned by the second function.
I think the problem lies in the control transfer communication, because the first function uses the ```libusb_control_transfer``` function to retrieve information from the device, while the second function uses the device struct to get data

mcuee, 2024-11-14:
@dragbahs 

We actually suggest hidapi project to be the preferred library for HID device. Can you give hidapi a try?
https://github.com/libusb/libusb/wiki/FAQ#user-content-Does_libusb_support_USB_HID_devices

**Warning: use of the HID backend is highly discouraged.** libusb project recommends the user to switch to [hidapi](https://github.com/libusb/hidapi).

mcuee, 2024-11-27:
@dragbahs 

Just wondering which  HID programmer from Microchip Technology Inc you are using for this test. I have a few such programmers and I can test to see if I can reproduce the issue or not. Thanks.

dragbahs, 2024-12-09:
I'm so sorry if I'm late. The HID programmer that I'm using is the ```PICKIT3``` programmer from Microchip Technology Inc. You can check all the errors mentioned above for the ```hidUSB``` default driver.

tormodvolden, 2024-12-10:
> According to the above image, the `bMaxPacketSize` value is always incorrect. However, for some devices, the `bcdDevice` value may be incorrect, while for others, the `bcdUSB` or other values may be incorrect.

It looks exactly like what you'd expect from using an older libusb than version nano 11932. Please attach a full debug log so that we can be sure what library you are using.

dragbahs, 2024-12-14:
Excuse me, sir, I’m not sure what the ```nano 11932``` version refers to. I’ve downloaded the latest version of your library from your GitHub repository, just like everyone else. Please simply write the sample C code that I provided in my first message and use the PICKIT3 programmer for a sample device. I’ve attached an image showing the version I’ve downloaded. I'm sorry because I can't provide more information.
[image]

dragbahs, 2024-12-14:
It shows that the version is the latest.

tormodvolden, 2024-12-15:
Don't worry about the nano version for now, please just provide a full debug log, by setting the environment variable LIBUSB_DEBUG=4 and capture the (stderr) text output from your program.

mcuee, 2024-12-17:
USBASP programmer:
https://github.com/dioannidis/usbasp/blob/master/firmware/usb_descriptors.h

```
$ ./libusb_issue_1583.exe
18 1 1 2 0 0 0 8 192 22 220 5 16 1 1 2 3 1
8 0110 16c0:05dc
```

mcuee, 2024-12-17:
Dell Mouse.

```
   ---------------------- Device Descriptor ----------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x201 (USB Version 2.0.1)
bDeviceClass             : 0x00 (defined by the interface descriptors)
bDeviceSubClass          : 0x00
bDeviceProtocol          : 0x00
bMaxPacketSize0          : 0x40 (64 bytes)
idVendor                 : 0x413C (Dell Inc.)
idProduct                : 0xB06E
bcdDevice                : 0x0101
iManufacturer            : 0x00 (No String Descriptor)
iProduct                 : 0x01 (String Descriptor 1)
iSerialNumber            : 0x00 (No String Descriptor)
bNumConfigurations       : 0x01 (1 Configuration)
Data (HexDump)           : 12 01 01 02 00 00 00 40 3C 41 6E B0 01 01 00 01   .......@<An.....
                           00 01                                             ..

$ ./libusb_issue_1583.exe
18 1 0 2 0 0 0 64 60 65 111 176 0 1 0 1 0 1
64 0101 413c:b06f
```

mcuee, 2024-12-17:
Logitech M100r USB wired mouse.

```

    ---------------------- Device Descriptor ----------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x200 (USB Version 2.0) -> but device is Low-Speed only
bDeviceClass             : 0x00 (defined by the interface descriptors)
bDeviceSubClass          : 0x00
bDeviceProtocol          : 0x00
bMaxPacketSize0          : 0x08 (8 bytes)
idVendor                 : 0x046D (Logitech Inc.)
idProduct                : 0xC077
bcdDevice                : 0x7200
iManufacturer            : 0x01 (String Descriptor 1)
 Language 0x0409         : "Logitech"
iProduct                 : 0x02 (String Descriptor 2)
 Language 0x0409         : "USB Optical Mouse"
iSerialNumber            : 0x00 (No String Descriptor)
bNumConfigurations       : 0x01 (1 Configuration)
Data (HexDump)           : 12 01 00 02 00 00 00 08 6D 04 77 C0 00 72 01 02   ........m.w..r..
                           00 01                                             ..

$ ./libusb_issue_1583.exe
18 1 0 2 0 0 0 64 109 4 119 192 0 1 1 2 0 1
8 7200 046d:c077
```

dragbahs, 2024-12-26:
"Hi Sir, I’m sorry if I’m late. In your tests, the issue has occurred on some devices. My question is: is the issue I reported correct? Is it resolvable, or should I change the default driver?

dragbahs, 2024-12-26:
And unfortunately I don't have a ```PICKIT4``` programmer.

dragbahs, 2024-12-26:
[image]

tormodvolden, 2025-01-07:
> Since I can reproduce the issue, here is the debug log, using libusb-1.0.27 release.

Can you please also confirm that the issue has been fixed in latest git?

dragbahs, 2025-01-10:
Hi, I have tested the updated version, and unfortunately, the problem still exists. I enabled the ```LIBUSB_LOG_LEVEL_DEBUG`` option and the result is as follows:

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.009923] [00000768] libusb: debug [libusb_get_device_list]
[ 0.009818] [00000e50] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.025048] [00000768] libusb: debug [winusb_get_device_list] ENUM pass HUB {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 0.025629] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_05E3&PID_0610\6&1DF8B2A&0&2
[ 0.025978] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [57]
[ 0.026331] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_8087&PID_0024\5&2F329127&0&1
[ 0.026561] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [58]
[ 0.026846] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\ROOT_HUB20\4&29F8B418&0
[ 0.027068] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [59]
[ 0.027393] [00000768] libusb: debug [winusb_get_device_list] assigning Root Hub 'USB\ROOT_HUB20\4&29F8B418&0' bus number 1
[ 0.027575] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_8087&PID_0024\5&4A4062C&0&1
[ 0.027804] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [5A]
[ 0.028022] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\ROOT_HUB30\4&B98BC84&0&0
[ 0.028152] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [5B]
[ 0.028382] [00000768] libusb: debug [winusb_get_device_list] assigning Root Hub 'USB\ROOT_HUB30\4&B98BC84&0&0' bus number 2
[ 0.028474] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\ROOT_HUB20\4&16AE9335&0
[ 0.028597] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [5C]
[ 0.028816] [00000768] libusb: debug [winusb_get_device_list] assigning Root Hub 'USB\ROOT_HUB20\4&16AE9335&0' bus number 3
[ 0.028893] [00000768] libusb: debug [winusb_get_device_list] ENUM pass DEV {A5DCBF10-6530-11D2-901F-00C04FB951ED}
[ 0.029039] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_0BDA&PID_F179\00E04C000001
[ 0.029353] [00000768] libusb: debug [get_api_type] driver(s): RtlWlanu
[ 0.029691] [00000768] libusb: debug [get_api_type] upper filter driver(s): vwifibus
[ 0.029828] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [16]
[ 0.030090] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_04D8&PID_900A\BUR195068601
[ 0.030449] [00000768] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.030663] [00000768] libusb: debug [get_api_type] matched driver name against HID API
[ 0.030768] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [3E]
[ 0.030934] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_04F2&PID_B34F\0X0001
[ 0.031374] [00000768] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.031546] [00000768] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.031654] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [3F]
[ 0.031780] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_24AE&PID_2013\7&30B3D4CC&1&2
[ 0.032103] [00000768] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032262] [00000768] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.032358] [00000768] libusb: debug [winusb_get_device_list] allocating new device for session [40]
[ 0.032579] [00000768] libusb: debug [winusb_get_device_list] ENUM pass HCD {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
[ 0.032753] [00000768] libusb: debug [winusb_get_device_list] ENUM processing PCI\VEN_8086&DEV_1E2D&SUBSYS_183E103C&REV_04\3&11583659&0&D0
[ 0.032931] [00000768] libusb: debug [winusb_get_device_list] ENUM processing PCI\VEN_8086&DEV_1E31&SUBSYS_183E103C&REV_04\3&11583659&0&A0
[ 0.033083] [00000768] libusb: debug [winusb_get_device_list] ENUM processing PCI\VEN_8086&DEV_1E26&SUBSYS_183E103C&REV_04\3&11583659&0&E8
[ 0.033422] [00000768] libusb: debug [winusb_get_device_list] ENUM pass GEN
[ 0.034310] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_05E3&PID_0610\6&1DF8B2A&0&2
[ 0.034541] [00000768] libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_05E3&PID_0610\6&1DF8B2A&0&2'
[ 0.034686] [00000768] libusb: debug [winusb_get_device_list] found existing device for session [57]
[ 0.034931] [00000768] libusb: debug [init_device] found 1 configurations (current config: 1) for device 'USB\VID_05E3&PID_0610\6&1DF8B2A&0&2'
[ 0.035950] [00000768] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.036086] [00000768] libusb: debug [init_device] (bus: 3, addr: 5, depth: 2, port: 2): 'USB\VID_05E3&PID_0610\6&1DF8B2A&0&2'
[ 0.036213] [00000768] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_0BDA&PID_F179\00E04C000001
[ 0.036433] [00000768] libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_F179\00E04C000001'
[... cut ...]
[ 0.216183] [00000768] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.216512] [00000768] libusb: debug [libusb_unref_device] destroy device 3.0
[ 0.216908] [00000768] libusb: debug [libusb_exit] destroying default context
[ 0.217328] [00000e50] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.220251] [00000768] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000A0
[ 0.220666] [00000768] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000009C
```

dragbahs, 2025-01-10:
18 1 0 2 0 0 0 **_64_** 216 4 10 144 0 1 1 2 3 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
8

dragbahs, 2025-01-10:
I have used the ```PICKIT3``` programmer from Microchip Technology Inc.

[image]

tormodvolden, 2025-01-10:
Thanks, but a disadvantage of the LIBUSB_LOG_LEVEL_DEBUG setting is that it doesn't give debug info from the library initialization. Using the environment variable LIBUSB_DEBUG (with value 4) would give this.

I think anyway there is a misunderstanding about "latest version". There is the latest release, which is 1.0.27 and doesn't have this fixed, and there is the development version maintained in git, popularly called "latest git" which I believe has this fixed. But we are not providing binaries / snapshot builds on a regular basis, so people have to build latest git on their own.

I can offer this snapshot build (version nano 11941, current master de38189e) but without any guarantees that it will run with your code (it is compiled with MinGW).
[libusb_nano_11941_win64.zip](https://github.com/user-attachments/files/18382766/libusb_nano_11941_win64.zip)

dragbahs, 2025-01-11:
Thanks, I used your ```.dll``` file and now the issue is fixed. but what's the difference between the latest ```1.0.27``` version and the ```nano``` version? Why don't you include the solution in the ```1.0.27``` version? Is there a special reason for that?

tormodvolden, 2025-01-11:
Great that it works. We fixed this after 1.0.27 was released. So the fix will appear in the 1.0.28 release, but we haven't made that release yet.

It is kind of interesting timing that we got this bug report now. The issue has been there forever since the HID backend was added decades ago, but nobody has reported it. We discovered it during development testing last year-ish, but it has had low priority so it wasn't immediately fixed up.

tormodvolden, 2025-01-11:
I will close this ticket since it has been fixed in commit 4528752. (We only track development status, and not release status.)

closed 2025-01-11 by tormodvolden

dragbahs, 2025-01-11:
Great, Looking forward for the ```1.0.28 release```.