libusb/libusb #1586: hotplug: opening usb device just after hotplug notification fails

bearsh, 2024-11-25:
Hi

I did some test with the hotplug mechanism in hidapi using the libusb backend. hidapi, in this case, uses the native hotplug api from libusb plus a queue and a separate thread to call into the application.

According this the libusb doc (https://libusb.sourceforge.io/api-1.0/libusb_hotplug.html, https://github.com/libusb/libusb/blob/de38189e8014fa393f4d8c1d9d3fdf5e2a95899d/libusb/hotplug.c#L99) it should be possible to open the usb device in the callback handler. For some reason, this is not going to work for me even the actual notification and opening is decoupled by a internal queue and done in another thread. Opening the device returns `LIBUSB_ERROR_ACCESS`. Putting in a short delay or a retry loop makes it works after a couple of milliseconds.

See also the discussion in https://github.com/libusb/hidapi/pull/674#pullrequestreview-2454043420

mcuee, 2024-11-25:
What is the device and operating system used?

bearsh, 2024-11-26:
> What is the device and operating system used?

- HID Device
- Ubuntu 24.02

mcuee, 2024-11-26:
My guess is that may have something to do with kernel hid driver detaching, which may take a little bit of time.

Putting in a short delay or a retry loop seems to be a reasonable work-around.

mcuee, 2024-11-26:
If you can, try it under FreeBSD to see if there is still an issue or not. FreeBSD seems to bind a ugen driver along with uhid driver for generic HID device (not keyboard/mouse/etc though).

bearsh, 2024-11-26:
> My guess is that may have something to do with kernel hid driver detaching, which may take a little bit of time.

yeah, that was my guess too

> Putting in a short delay or a retry loop seems to be a reasonable work-around.

that would be something for hidapi then

> What is the HID device?

it's custom sensor device with a custom protocol based on hid

> If you can, try it under FreeBSD to see if there is still an issue or not. FreeBSD seems to bind a ugen driver along with uhid driver for generic HID device (not keyboard/mouse/etc though).

that would take some time as I currently don't a system running freebsd...

mcuee, 2024-11-27:
@tormodvolden 

Please share your view as well. Thanks.

mcuee, 2025-03-23:
Close first because of inactivity. Please re-open if there is new info pointing out to a potential libusb problem.

closed 2025-03-23 by mcuee