libusb/libusb #1588: [Linux] Memory leak on `libusb_close`

fabiensanglard, 2024-12-05:
Asan is reporting a memory leak when `[libusb_close](https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#ga779bc4f1316bdb0ac383bddbd538620e)` is called when a `libusb_bulk_transfer` is pending.

```
=================================================================
==263678==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 56 byte(s) in 1 object(s) allocated from:
    #0 0x55c9a0bce7f9 in calloc out/stage2/runtimes/runtimes-x86_64-unknown-linux-gnu-bins/out/llvm-project/compiler-rt/lib/asan/asan_malloc_linux.cpp:75:3
    #1 0x55c9a14de3ac in submit_bulk_transfer external/libusb/libusb/os/linux_usbfs.c:1988:9
    #2 0x55c9a14ce35c in libusb_submit_transfer external/libusb/libusb/io.c:1555:6
    #3 0x55c9a14d3a06 in do_sync_bulk_transfer external/libusb/libusb/sync.c:194:6
    #4 0x55c9a14d37a3 in libusb_bulk_transfer external/libusb/libusb/sync.c:286:9
```

The line leaked is [here](https://github.com/libusb/libusb/blob/de38189e8014fa393f4d8c1d9d3fdf5e2a95899d/libusb/os/linux_usbfs.c#L1988).

```
usbi_dbg(TRANSFER_CTX(transfer), "need %d urbs for new transfer with length %d", num_urbs, transfer->length);
	urbs = calloc(num_urbs, sizeof(*urbs));
	if (!urbs)
```

# Analysis:

From a summary inspection, it looks like the way this should have been freed is when `op_handle_events` calls `reap_for_handle` which calls [handle_bulk_completion](https://github.com/libusb/libusb/blob/de38189e8014fa393f4d8c1d9d3fdf5e2a95899d/libusb/os/linux_usbfs.c#L2483C1-L2484C20):

```
completed:
	free(tpriv->urbs);
	tpriv->urbs = NULL;
```

However this seems to not happen because when a device is closed, `op_handle_events` will not find the opened device anymore.

fabiensanglard, 2024-12-24:
I am not quite sure how to fix this. Perhaps instead of deleting the device, we should move it to another list so we can keep reaping from it until all URBs have finished?

mcuee, 2025-06-03:
Not so sure if the following is relevant or not -- the better way i probably to use async transfer.

From libusb Wiki FAQ: the following is talking about `libusb_exit` but it may be relevant for `libusb_close` as well.
https://github.com/libusb/libusb/wiki/FAQ#what-are-the-differences-between-synchronous-and-asynchronous-device-io

**What are the differences between Synchronous and asynchronous device I/O?**
Please read libusb-1.0 API documentation here. https://libusb.sourceforge.io/api-1.0/libusb_io.html

Please take note that ALL transfers MUST be freed before libusb_exit() is called.

Please take note that ALL libusb objects MUST be released before libusb_exit() is called.

Please also take note that there is no way to cancel Synchronous transfer after the request has been submitted.

mcuee, 2025-06-03:
@tormodvolden and @sonatique 

Just wondering if you have some comments here as well. Thanks.

sonatique, 2025-06-04:
@mcuee : I am not super familiar with linux backend, so it's just my 2 cents, .but here are a few considerations:

1) The documentation indeed clearly states what user has to do before calling libusb_exit. But I don't think we could "decide" that it shall apply to libusb_close from now on because we found a bug ;-)

2) Using "async transfer" is not a solution since anyway sync transfers are implemented using async transfers and the issue is in async transfers

3) from what I see, @fabiensanglard idea is probably going into the right direction. I don't think trying to free urb in op_close (i.e. in the backend implementation of do_close since it is linux usb_fs specific) can work because it implies waiting to the transfer to complete somehow.

4) This makes me think: what if we just added some code to interrupt all transfers in op_close before doing anything else? And also modify the implementation of op_handle_events to also reap transfers that does not belong to any open device?