libusb/libusb #1589: [Windows 10] libusb_open fail and return LIBUSB_ERROR_NO_DEVICE .

qzl-core1, 2024-12-06:
When I use **libusb_open**, it returns LIBUSB_ERROR_NO_DEVICE, but when I use **libusb_open_device_with_vid_pid**, the device can be opened normally and the endpoint can be opened for communication.
My USB device is a custom winUSB device
**libusb** version:1.0.25

Here is the code used :
```
  int main() {
    static struct libusb_device *usb_dev = NULL;
    static struct libusb_device_handle* dev_handle = NULL;
    libusb_context *ctx = NULL;
    libusb_init(&ctx);
    libusb_set_option(ctx, LIBUSB_OPTION_USE_USBDK);
    struct libusb_device **device_list;
    int num_devices = libusb_get_device_list(NULL, &device_list);
    int dm_cnt = 0;
    for (int i = 0; i < num_devices; i++)
    {
        struct libusb_device *dev = device_list[i];
        struct libusb_device_descriptor dev_desc;
        libusb_get_device_descriptor(dev, &dev_desc);
        int id_vendor =  dev_desc.idVendor;
        int id_product = dev_desc.idProduct;
        if(id_vendor == DM_USBD_VID && id_product == DM_USBD_PID)
        {
            usb_dev = dev;
            dm_cnt++;
        }
    }
    libusb_free_device_list(device_list,num_devices);

    int e;
    if(usb_dev == NULL)
    {
        libusb_exit(ctx);
        return -1;
    }

    e = libusb_open(usb_dev , &dev_handle);
    if(e < 0)
    {
        printf("err code = %d , libusb_open failed:[%s]\n", e ,libusb_strerror(e));
        libusb_close(dev_handle);
        libusb_exit(ctx);
        while(1);
        return -1;
    }
    else
    {
        printf("open successful\n");
    }
    return 0;
}
```

mcuee, 2024-12-09:
Please provide more info as per the following guide.
https://github.com/libusb/libusb/wiki/Troubleshooting

Maybe you may want to try the latest release version 1.0.27.

qzl-core1, 2024-12-11:
> Please provide more info as per the following guide. https://github.com/libusb/libusb/wiki/Troubleshooting
> 
> Maybe you may want to try the latest release version 1.0.27.

I'm very sorry.
the issue with my code has now been resolved. Upon got all devices, I called `libusb_free_device_list`, which resulted in the freeing of my target device.

closed 2024-12-17 by mcuee