libusb/libusb #1592: Using libusb with USBTMC Instruments Failed

crazyRabbitboomboomboom, 2024-12-23:
Hello, when I use libusb to communicate with an instrument, I find that I can successfully detect the device and retrieve its VID and PID. However, when I try to send commands to the instrument, I always get the error LIBUSB_ERROR_IO (error code: -1). Why is this happening? Does libusb not support communication with instruments that use the USBTMC protocol?

mcuee, 2025-01-04:
You have to know the USB communication protocol in order to send the correct comand. You may have to provide more details.
https://github.com/libusb/libusb/wiki/Troubleshooting

tormodvolden, 2025-02-23:
Did you check if sigrok can support your device? https://sigrok.org/wiki/USBTMC

mcuee, 2025-03-02:
Close this issue for now as it is not a libusb issue.

closed 2025-03-02 by mcuee