libusb/libusb #1593: MacOS Segmentation fault

Aterox-os, 2025-01-03:
Hello,
I am trying to use the librealsense library for my intel depth camera. The library has libusb as a dependency. Both libraries work as intended under windows but I am using an M2 MacBook. Here a segmentation fault occurs. The the problematics piece of code is from libusb. It's the `src/libusb/context-libusb.cpp` file. Here is the [Librealsense project](https://github.com/IntelRealSense/librealsense)
```
#include "context-libusb.h"
#include "../types.h"

namespace librealsense
{
    namespace platform
    {       
        usb_context::usb_context() : _ctx(NULL), _list(NULL), _count(0)
        {
            auto sts = libusb_init(&_ctx);
            if(sts != LIBUSB_SUCCESS)
            {
                LOG_ERROR("libusb_init failed");
            }
            _count = libusb_get_device_list(_ctx, &_list);
        }
        
        usb_context::~usb_context()
        {
            libusb_free_device_list(_list, true);
            assert(_handler_requests == 0); // we need the last libusb_close to trigger an event to stop the event thread
            if (_event_handler.joinable())
                _event_handler.join();
            libusb_exit(_ctx);
        }
        
        libusb_context* usb_context::get()
        {
            return _ctx;
        } 
    
        void usb_context::start_event_handler()
        {
            std::lock_guard<std::mutex> lk(_mutex);
            if (!_handler_requests) {
                // see "Applications which do not use hotplug support" in libusb's io.c
                if (_event_handler.joinable()) {
                    _event_handler.join();
                    _kill_handler_thread = 0;
                }
                _event_handler = std::thread([this]() {
                    while (!_kill_handler_thread)
                        libusb_handle_events_completed(_ctx, &_kill_handler_thread);
                });
            }
            _handler_requests++;
        }

        void usb_context::stop_event_handler()
        {
            std::lock_guard<std::mutex> lk(_mutex);
            _handler_requests--;
            if (!_handler_requests)
                // the last libusb_close will trigger and event and the handler thread will notice this is set
                _kill_handler_thread = 1;
        }

        libusb_device* usb_context::get_device(uint8_t index)
        {
            return index < _count ? _list[index] : NULL;
        }
        
        size_t usb_context::device_count()
        {
            return _count;
        }
    }
}

```

This is the problematic part where EXC_BAD_ACCESS (code=1, address=0x28) occurs:
```
        usb_context::usb_context() : _ctx(NULL), _list(NULL), _count(0)
        {
            auto sts = libusb_init(&_ctx);
            if(sts != LIBUSB_SUCCESS)
            {
                LOG_ERROR("libusb_init failed");
            }
            _count = libusb_get_device_list(_ctx, &_list);
        }
```
It's a problem with the line `_count = libusb_get_device_list(_ctx, &_list);`. 
I hope that some has a solution for the problem.

seanm, 2025-01-03:
What version of libusb?  Have you tried master?

`address=0x28` suggests a null pointer offset by 28.  Have you tried address sanitizer?

Aterox-os, 2025-01-03:
@seanm I am using version 1.0.27. I haven't tried an address sanitizer. I will do it tomorrow.

seanm, 2025-01-04:
https://developer.apple.com/documentation/xcode/diagnosing-memory-thread-and-crash-issues-early

https://clang.llvm.org/docs/AddressSanitizer.html

tormodvolden, 2025-02-23:
@Aterox-os Did you try building libusb from git master?

mcuee, 2025-06-06:
@Aterox-os 

Please help to update. You can use libusb-1.0.29 release. Thanks.

 Old issues longer than three months without updates may be closed.

mcuee, 2025-07-09:
> I can confirm this happens on 1.0.29

Please help to post your test code and try with Address Sanitizer. Thanks.
https://developer.apple.com/documentation/xcode/diagnosing-memory-thread-and-crash-issues-early

jdarpinian, 2025-07-09:
It's a null dereference, doesn't really need ASAN

seanm, 2025-07-09:
ASan also gives information like the backtrace of the crash and backtrace of where the memory was allocated, which are things that are not yet described here.

What would help even more of course is a self-contained minimized repro case.

(Remember we are unpaid volunteers. I might spend my time trying your small example, but I'm not going download some other library I've never used, figure out how to build it, etc. etc.  The easier for you make it for others to help, they more likely they'll help.)

seanm, 2025-07-09:
Did you build libusb as debug?  frame 0 has no filename/line number, but frame 20 for example does...

seanm, 2025-07-09:
There's only one call to `usbi_mutex_lock` in `libusb_get_device_list` though, so I guess it's that. Could you do an `assert(ctx)` just before the `usbi_mutex_lock(&ctx->usb_devs_lock);`?

jdarpinian, 2025-07-09:
I've done some more debugging. The context is null, unsurprisingly. As you can see in the code referenced in the first comment, librealsense calls libusb_init just before this, and while it checks the error code it still continues to call libusb_get_device_list. libusb_init returns -99.

jdarpinian, 2025-07-09:
Hard for me to say if this is a bug in librealsense or libusb, I'm going to guess the former.

seanm, 2025-07-09:
-99 is `LIBUSB_ERROR_OTHER`, next I would step in lldb to find where that's getting set.

mcuee, 2025-07-11:
> I've done some more debugging. The context is null, unsurprisingly. As you can see in the code referenced in the first comment, librealsense calls libusb_init just before this, and while it checks the error code it still continues to call libusb_get_device_list. libusb_init returns -99.

Good finding. Looks like a bug of librealsense to me already.

closed 2026-02-28 by mcuee