libusb/libusb #1594: Access Denied on ClaimInterface

edfapay, 2025-01-05:
I have successfully Init, List devices, Open device. but when i am moving to claimInterface Its throws Access Denied, Below is the code

```
package me.mariodev.socketio_server_java_demo;

import io.socket.emitter.Emitter;
import io.socket.socketio.server.SocketIoNamespace;
import io.socket.socketio.server.SocketIoServer;
import io.socket.socketio.server.SocketIoSocket;
import org.usb4java.*;

import java.io.BufferedReader;
import java.io.IOException;
import java.io.InputStreamReader;

public class Main {
    static Context context = new Context();
    static int result;
    static DeviceHandle handle;

    // Enter data using BufferReader
    static BufferedReader reader = new BufferedReader(new InputStreamReader(System.in));

    public static void main (String[] args) throws IOException {
        usb();
    }

    public static void usb() throws IOException {

        reader.readLine();
        init();

        reader.readLine();
        listDevices();

        reader.readLine();
        open((short) 0x04e8, (short) 0x6860);

        reader.readLine();
        attachDetach();

        reader.readLine();
        claimInterface();
    }

//        Initialization
    public static void init(){
        result = LibUsb.init(context);
        if (result != LibUsb.SUCCESS)
            throw new LibUsbException("Unable to initialize libusb.", result);

        System.out.println("1: Init Success");
    }

    public static void open(short vid, short pid){
        handle = LibUsb.openDeviceWithVidPid(context, vid, pid);
        if (result != LibUsb.SUCCESS)
            throw new LibUsbException("Unable to open device channel.", result);

        System.out.println("2: Open Success");
    }

    public static void attachDetach(){
        // Check if kernel driver is attached to the interface
        int attached = LibUsb.kernelDriverActive(handle, 1);
        if (attached < 0) {
            throw new LibUsbException(
                    "Unable to check kernel driver active", result);
        }

        // Detach kernel driver from interface 0 and 1. This can fail if
        // kernel is not attached to the device or operating system
        // doesn't support this operation. These cases are ignored here.
        result = LibUsb.detachKernelDriver(handle, 1);
        if (result != LibUsb.SUCCESS &&
                result != LibUsb.ERROR_NOT_SUPPORTED &&
                result != LibUsb.ERROR_NOT_FOUND) {
            throw new LibUsbException("Unable to detach kernel driver",
                    result);
        }
    }

    public static void claimInterface(){
        result = LibUsb.claimInterface(handle, 1);
        if (result != LibUsb.SUCCESS)
            throw new LibUsbException("Unable to claim interface", result);

        try {
            // Use interface here
        } finally {
            result = LibUsb.releaseInterface(handle, 1);
            if (result != LibUsb.SUCCESS) throw new LibUsbException("Unable to release interface", result);
        }

    }

    public static void listDevices(){

        // Read the USB device list
        DeviceList devices = new DeviceList();
        result = LibUsb.getDeviceList(context, devices);
        if (result < 0) throw new LibUsbException("Unable to get device list", result);

        try
        {
            // Iterate over all devices and scan for the right one
            for (Device device: devices)
            {
                DeviceDescriptor descriptor = new DeviceDescriptor();
                result = LibUsb.getDeviceDescriptor(device, descriptor);
                if (result != LibUsb.SUCCESS)
                    throw new LibUsbException("Unable to read device descriptor", result);

                System.out.println("-----------------------");
                System.out.println("VendorID:   " + descriptor.idVendor());
                System.out.println("ProductID:  " + descriptor.idProduct());
                System.out.println("Descriptor: " + descriptor.toString());

            }
        } finally {
            // Ensure the allocated device list is freed
            LibUsb.freeDeviceList(devices, true);
        }
    }
}

```

tormodvolden, 2025-01-07:
Please see if you experience the same with xusb from the examples folder: `xusb -d 04e8:6860`

mcuee, 2025-03-02:
Most likely there is another driver claim the interface.

Try to use sudo to see if that sorted out the issue or not.

mcuee, 2025-03-23:
Close first because of inactivity. Please re-open if there is new info pointing out to a potential libusb problem.

closed 2025-03-23 by mcuee