libusb/libusb #1596: [WINDOWS] Libusb context initialization using UsbDK

lucasTanne, 2025-01-10:
Hello,
I think there is a bug during the context initialization, specially on Windows when I'm trying to use the `UsbDK backend`.
The API documentation tell me:
- `libusb_init()` is deprecated, use `libusb_init_context()` instead
- `libusb_init_context()` take as parameters the context, a list of options and the number of options containing in the list.

But when I'm trying to initialize a context with `libusb_init_context()` with the `LIBUSB_OPTION_USE_USBDK` option, libusb cannot find the driver UsbDK, I have the error `LIBUSB_ERROR_NOT_FOUND [code -5]` :
```c
#include <stdlib.h>
#include <stdio.h>
#include <libusb.h>

int main() {
    libusb_context *ctx = NULL;
    struct libusb_init_option opts[4];
    struct libusb_init_option logOpt = {
        LIBUSB_OPTION_LOG_LEVEL,
        4
    };
    struct libusb_init_option backOpt = {
        LIBUSB_OPTION_USE_USBDK
    };

    opts[0] = logOpt;
    opts[0] = backOpt;

    int res;
    res = libusb_init_context(&ctx, opts, 2);
    if (res == LIBUSB_ERROR_NOT_FOUND) {
        printf("NOT FOUND: %d\n", res);
        exit(1);
    } else if (res > 0) {
        printf("ERR OTHER: %d\n", res);
        exit(1);
    }
    printf("SUCESS\n");

    libusb_device **list;
    int s;
    s = libusb_get_device_list(ctx, &list);
    printf("RECV size: %d\n", s);
}
```

If I use the "deprecated" way (`libusb_init_context() + libusb_set_option()`):
```c
#include <stdlib.h>
#include <stdio.h>
#include <libusb.h>

int main() {
    libusb_context *ctx = NULL;
    struct libusb_init_option opts[4];
    struct libusb_init_option logOpt = {
        LIBUSB_OPTION_LOG_LEVEL,
        4
    };

    opts[0] = logOpt;

    int res;
    res = libusb_init_context(&ctx, opts, 1);
    if (res == LIBUSB_ERROR_NOT_FOUND) {
        printf("NOT FOUND: %d\n", res);
        exit(1);
    } else if (res > 0) {
        printf("ERR OTHER: %d\n", res);
        exit(1);
    }
    printf("SUCESS\n");

    res = libusb_set_option(ctx, LIBUSB_OPTION_USE_USBDK);
    if (res == LIBUSB_ERROR_NOT_FOUND) {
        printf("SET OPTION ERROR NOT FOUND: %d\n", res);
        exit(1);
    } else if (res > 0) {
        printf("SET OPTION ERROR OTHER: %d\n", res);
        exit(1);
    }
    printf("SET OPTION SUCESS\n");

    libusb_device **list;
    int s;
    s = libusb_get_device_list(ctx, &list);
    printf("RECV size: %d\n", s);
}
```
It's working, the backed is switched to use UsbDK backend and I receive the size 1 `RECV size: 1` (result of the last printf)

When i'm reading the documentation, I understand that using `libusb_init_context()` I can send the option directly without using `libusb_set_option()` and my context will be intialized with all require.

But now I don't know if my code is wrong, if it's a bug or just the wanted behavior ?

## My env

My OS: Linux Debian 11 Bullseye
Libusb version: 1.0.27 and compiled from the master branch (I have to compile libusb because the version provided by APT is 1.0.24)
Target OS: Windows 10/11
UsbDK is installed in last version on the Windows machine.

I cross-compile the code from Linux to Windows using mingw:
```bash
x86_64-w64-mingw32-gcc -o test main.c -I/home/libusb/include/libusb-1.0 -L/home/libusb/lib -lusb-1.0
```
And libusb was also cross-compiled using mingw:
```bash
./autogen.sh --host x86_64-w64-mingw32 --build x86_64 --prefix=/home/libusb
make install
PKG_CONFIG_PATH=/home/libusb/lib/pkgconfig
```

Youw, 2025-01-10:
Looks like this is your problem:
```c
opts[0] = backOpt;
```
Should be:
```c
opts[1] = backOpt;
```

Youw, 2025-01-10:
OK, I've checked the code and confirm - there is a bug for this scenario:
We try to set the UsbDK option [here](https://github.com/libusb/libusb/blob/de38189e8014fa393f4d8c1d9d3fdf5e2a95899d/libusb/core.c#L2499), and it fails [here](https://github.com/libusb/libusb/blob/de38189e8014fa393f4d8c1d9d3fdf5e2a95899d/libusb/os/windows_common.c#L615), since UsbDK [initialisation](https://github.com/libusb/libusb/blob/de38189e8014fa393f4d8c1d9d3fdf5e2a95899d/libusb/os/windows_common.c#L536) is done [later](https://github.com/libusb/libusb/blob/de38189e8014fa393f4d8c1d9d3fdf5e2a95899d/libusb/core.c#L2527). 

---
@lucasTanne using the `libusb_set_option` function later _after_ `libusb_init_context` is perfectly good workaround for you until this is fixed.

lucasTanne, 2025-01-10:
That's cool !

Yes, the idea of the workaround is because the initialization using UsbDK doesn't work, just initialize using the default driver WinUSB and then change the backend (found the log `switching context ... to use the UsbDK backend` in the code, that gave me the idea) .

So, I keep my workaround for the moment, thank you !

tormodvolden, 2025-01-10:
Great analysis, @Youw .

My thoughts on this:
An internal _set_option() (or better name) function should be refactored out from the API libusb_set_option() so that this new _set_option() can be used both from libusb_set_option() and from libusb_init_context() but with various differences.

Separate of the present issue, a motivation for this is to allow us to warn or fail if libusb_set_option() is used with options that really belong to context initialization and are not meant to be changed afterwards, or the opposite case if that would apply.

The new _set_option() would need an extra flag to communicate to the various backends .set_option() functions whether they are called from libusb_set_option() or from libusb_init_context().

If we had this in place, we could for the present issue have libusb_init_context -> _set_option -> windows_set_option only set an intention to use usbdk (perhaps via a another windows_context_priv flag member), and then accordingly set priv->backend in windows_init() if usbdk is desired AND available.

And for backwards compatibility (I don't know if this has any other reason to be supported) libusb_set_option -> _set_option -> windows_set_option would set the backend directly.

I hope there is a quicker fix for this, but I believe the libusb_set_option refactoring is needed in the long run anyway.

Youw, 2025-01-13:
> I hope there is a quicker fix for this

Some quick (and a bit dirty) fix for this particualar one (not tested):
```diff
diff --git a/libusb/core.c b/libusb/core.c
index 676c397e..06c59b57 100644
--- a/libusb/core.c
+++ b/libusb/core.c
@@ -2432,6 +2432,9 @@ int API_EXPORTED libusb_init_context(libusb_context **ctx, const struct libusb_i
        struct libusb_context *_ctx;
        int r;

+       int option_use_usbdk = 0;
+       int option_use_usbdk_value = 0;
+
        usbi_mutex_static_lock(&default_context_lock);

        if (!ctx && default_context_refcnt > 0) {
@@ -2493,6 +2496,10 @@ int API_EXPORTED libusb_init_context(libusb_context **ctx, const struct libusb_i

                case LIBUSB_OPTION_LOG_LEVEL:
                case LIBUSB_OPTION_USE_USBDK:
+                       /* postpone until usbi_backend.init (windows_init) is done */
+                       option_use_usbdk = 1;
+                       option_use_usbdk_value = options[i].value.ival;
+                       continue;
                case LIBUSB_OPTION_NO_DEVICE_DISCOVERY:
                case LIBUSB_OPTION_MAX:
                default:
@@ -2529,6 +2536,13 @@ int API_EXPORTED libusb_init_context(libusb_context **ctx, const struct libusb_i
                        goto err_io_exit;
        }

+       if (option_use_usbdk) {
+               /* LIBUSB_OPTION_USE_USBDK can only be set after usbi_backend.init (windows_init) is done */
+               r = libusb_set_option(_ctx, LIBUSB_OPTION_USE_USBDK, option_use_usbdk_value);
+               if (r)
+                       goto err_io_exit;
+       }
+
        /* Initialize hotplug after the initial enumeration is done. */
        usbi_hotplug_init(_ctx);
```

tormodvolden, 2025-02-23:
Yes, this may be much easier than I thought. But I really wanted to have all that refactored :)

It doesn't look to dirty, but we shouldn't set USBDK if only LOG_LEVEL has been set though.