libusb/libusb #1599: Crash when re-enumerating a device with different class GUID under Windows

jbggt, 2025-01-16:
I am debugging a problem with libusb (current version, 1.0.27), which crashes in a corner case situation. To reproduce the problem a "cooperation from outside the app" is required, therefore I do not share directly a code snippet, just steps describing the situation. I wonder if somebody can comment - if more input is required, please let me know.

The problem is related to Windows platform and can be reproduced as follows:

1. The application (more precisely a library) enumerates the devices using libusb_get_device_list(), collects infos about the devices and (important) increases refcount (libusb_ref_device()) of the device(s) potentially interesting for the library users (more info can be queried about those devices later or the devices can be opened.
2. While that ref is taken, another driver was installed for given device (in the particular case where we observed the problem, WinUSB driver was installed for the device while the library was running.
3. Later a new device enumeration round (libusb_get_device_list()) was triggered. During this operation, libusb sees the device enumerated with a different class GUID. This happens inside winusb_get_device_list(), within following check
https://github.com/libusb/libusb/blob/de38189e8014fa393f4d8c1d9d3fdf5e2a95899d/libusb/os/windows_winusb.c#L1895-L1900
As a side note, this code was introduced in
https://github.com/libusb/libusb/commit/bb021598078eeea9efdbee131909f62c169a4d75
while fixing
https://github.com/libusb/libusb/issues/897
4. When seeing different GUID, libusb calls explicitly usbi_disconnect_device() on the device, removing it from the list and allocates the new one. It calls usbi_disconnect_device() explicitly, even if the call is immediately followed by libusb_unref_device() which would call usbi_disconnect_device() anyway if (important) there were no references to it.
5. Later, during cleanup, our library calls finally itself libusb_unref_device() on the "old" device which is no more enumerated. During this call the refcount goes to zero, libusb attempts to call usbi_disconnect_device() again, but because it was disconnected already in step 4 and the list pointers are NULL, the list_del call
https://github.com/libusb/libusb/blob/de38189e8014fa393f4d8c1d9d3fdf5e2a95899d/libusb/core.c#L751
crashes when dereferencing the NULL pointers.

I wonder whether in step 4 usbi_disconnect_device() should be called explicitly or if just the libusb_unref_device() should suffice, letting the disconnect happen only on last reference - I am not familiar enough with libusb internals to be confident about all consequences, thus I prefer to ask.  Or should the list_del() call be guarded to not happen if the list pointers are NULL already?  Or is the above use case somehow "illegal"? But the only thing our lib did in that case was that it took ref to the device which was later rediscovered with different class GUID.

Thanks for any hints. Jan

tormodvolden, 2025-02-23:
Thanks for your report. It sounds from your explanation about (4) that we shouldn't call usbi_disconnect_device() explicitly. It is possible that code was written with the assumption that nothing else had a reference to the device. If we did that change, would your example work fine? I have to wrap my mind around the concept of some code having a reference to the old device while other code may reference the new device. There was another recent report that made me thinking we need to keep "zombie" devices around for our transfer house-keeping.

jbggt, 2025-02-24:
Thanks for looking into our problem.

I double checked again and yes, if I remove the usbi_disconnect_device() call, it does avoid the crash in my test. However, as mentioned in the initial report, I am not familiar enough with libusb internals enough to be competent to claim this is the correct fix (or if it can have any undesirable side effects), thus please take my description more as a question than concrete change proposal.

When you make an opinion on that, please tell me in case some more information or tests is required.

--

As a side note, based on your last sentence related to transfer house-keeping, could you point me to the actual report you had in mind? I do not want to mix it with the current one, but interestingly we had recently similar issue which we in the end considered our fault. The problem was that we did not anticipate that the transfers (bulk in our case) could be handled/reported out of order, in particular when multiple threads handle them at the same time. In turn, we could close the device too soon after seeing the "last" transfer completed, but if in some corner case scenario some of the "previous" transfers were still residing in your in-flight lists, processing that list would lead to a crash. The docu did not seem to be 100% explicit regarding the transfer handling order and we fixed that problem on our side (waiting explicitly for all transfers). Again, I do not want to mix this with the current ticket, I'm just interested in the reference you mentioned for comparison.

tormodvolden, 2025-06-03:
> As a side note, based on your last sentence related to transfer house-keeping, could you point me to the actual report you had in mind?

Maybe it was #1588 ?

mcuee, 2026-03-01:
@jbggt 

Can you try libusb-1.0.31-rc1 which has Windows hotplug feature to see if that fix your original issue or not? Thanks.

jbggt, 2026-03-13:
Just to be sure, I assume the question has a typo and you mean libusb-1.0.30-rc1 (30 vs 31, I did not find any 31-rc1 tag) - correct?

I tested with both 30-rc1 and current master and in both cases it crashes with hotplug disabled and does not crash with hotplug enabled.

sonatique, 2026-03-13:
@jbggt : yes, sorry for the typo.

sonatique, 2026-03-13:
@jbggt : thanks for the feedback, this is very useful. Seem that the hotplug code somehow prevent the crash. It should be possible to understand the issue and port the fix. Let me see.