libusb/libusb #1602: libusb_bulk_transfer returns LIBUSB_ERROR_TIMEOUT on Android 13

GrieVeR-13, 2025-01-30:
When trying to send any command (SCSI Inquiry, SCSI read capacity (10), etc.) to interact with the SSD via the adapter on android 13, the libusb_bulk_transfer function returns the error code LIBUSB_ERROR_TIMEOUT.

At the same time, on android 9 and windows 10, the same SSD with the same adapter works right.

A regular usb flash drive (8gb) on android 13 works.
Increasing the timeout does not solve the problem.
libusb_bulk_transfer writes to outEndpoint successfully.

The problem is also reproduced in other applications using libusb (tested on android app: ZUGate).

The problem is reproduced on the following configuration:
OS: Android 13 MIUI 14.0.7
SSD: 870 PRO Samsung 4 TB
External SSD Enclosures: DEXP 2518s3 with controller ASMT105x

Scenario:
//Request LUN
libusb_control_transfer(libUsbHandle, 161, 254, 0, index, buffer, length, TRANSFER_TIMEOUT)
//Successfully

//Sending a command SCSI Inquiry (bulk OUT transfer).
libusb_bulk_transfer(libUsbHandle, outEndpoint.address, src.array(), src.position(), src.remaining(), TRANSFER_TIMEOUT)
//Successfully.

//Reading the response from the command (bulk IN transfer).
libusb_bulk_transfer(libUsbHandle, inEndpoint.address, dest.array(), dest.position(), dest.remaining(), TRANSFER_TIMEOUT)
//! returns LIBUSB_ERROR_TIMEOUT

mcuee, 2025-01-30:
This probably has nothing to do with libusb but rather with your decice or your Android platform.
1) Please try with a regular Linux machine to see if it works or not. If not, your device is the issue.
2) If the first works, try with another Android device to see if you can reproduce your issue or not.

GrieVeR-13, 2025-02-03:
>     2. If the first works, try with another Android device to see if you can reproduce your issue or not.

On android 9 this SSD with the same adapter works right. In addition, this problem occurs most often with Realtek and Asmedia chips.

Update:
Despite the fact that the ssd does not work via **libusb** on android 13, data from this drive is perfectly readable via the built-in tools of the **android 13 system** (successfully mounted as sdcard in /storage/emulated/xxx)

That's why I assumed the problem was with how libusb communicates in a different way.

VioletGiraffe, 2025-02-19:
Is the device working in USB 3 or USB 2 mode on the faulty system? If it's USB 3, try forcing it to USB 2 by using a USB 2.0 OTG adapter or cable.

mcuee, 2025-03-23:
> At the same time, on android 9 and windows 10, the same SSD with the same adapter works right.

Do you use libusb for those platforms? If you also use libusb, then your Android 13 is the problem.

If you do not use libusb with those platforms, then your libusb code is probably the problem.

mcuee, 2025-03-23:
Close this issue first. Please re-open if you think this is a libusb issue.

closed 2025-03-23 by mcuee