libusb/libusb #1604: [Windows] Getting wrong value from libusb_get_device_speed

fabiensanglard, 2025-02-07:
I am testing with a Pixel device (which are USB3 capable since Pixel 7). 

If the device is connected with a USB3 cables, I get the right value from `libusb_get_device_speed` (`libusb_speed::LIBUSB_SPEED_SUPER`).

However, if I connect the same device with a USB2 cable, the negotiated speed should be `LIBUSB_SPEED_HIGH`. But I still get `LIBUSB_SPEED_SUPER`.

mcuee, 2025-02-08:
Please help to try out the libusb Windows hotplug PR to see if the issue is there or not.
* https://github.com/libusb/libusb/pull/1406

fabiensanglard, 2025-02-10:
I tried #1406 and it solved the problem. Device speed is correctly detected.

However it looks like #1406 effort is dead, there has not been any update on it in six months.

fabiensanglard, 2025-02-10:
Also, testing it further, Windows hotplug leads to crash:

```
[Inline Frame] adb.exe!list_del(list_head * entry) Line 234
at /external/libusb/libusb/libusbi.h(234)
[Inline Frame] adb.exe!remove_from_flying_list(usbi_transfer * itransfer) Line 1465
at /external/libusb/libusb/io.c(1465)
adb.exe!libusb_submit_transfer(libusb_transfer * transfer) Line 1563
at /external/libusb/libusb/io.c(1563)
adb.exe!do_sync_bulk_transfer(libusb_device_handle * dev_handle, unsigned char endpoint, unsigned char * buffer, int length, int * transferred, unsigned int timeout, unsigned char type) Line 195
at /external/libusb/libusb/sync.c(195)
adb.exe!libusb_bulk_transfer(libusb_device_handle * dev_handle, unsigned char endpoint, unsigned char * data, int length, int * transferred, unsigned int timeout) Line 286
at /external/libusb/libusb/sync.c(286)
```

sonatique, 2025-04-27:
> I tried [#1406](https://github.com/libusb/libusb/pull/1406) and it solved the problem. Device speed is correctly detected.
> 
> However it looks like [#1406](https://github.com/libusb/libusb/pull/1406) effort is dead, there has not been any update on it in six months.

As said in the PR directly, the Windows hotplug PR is and was not dead. It was in a sort of zombie state because there was neither a decision to merge it, nor to discard it, nor a todo list for making it mergeable. I hope that my new effort could lead to merging it: https://github.com/libusb/libusb/pull/1633 .

Now:  I still need to understand why one would think that the hotplug feature would help with some issue with getting the right speed, but the crash is obviously something to look at, and I will.

sonatique, 2025-04-28:
@fabiensanglard : could you please share (part of) your application, or a least describe the relevant part?

Do you periodically call libusb_get_device_list() and print details of found devices, including speed?

Do you then start some bulk transfer with it? The crash trace you reported show that the crash implies issues when submitting transfer, which is a part of the code that has not been modified by my windows hotplug PR, so its effect is probably very indirect.

Could you tell me a little bit more? Thank you in advance

mcuee, 2025-04-28:
@sonatique 

Just wondering if you can accomodate the following PR for testing.
* https://github.com/libusb/libusb/pull/1498

sonatique, 2025-04-28:
@mcuee : sorry, what do you mean exactly? Would you like me to test this PR within the context of the new hotplug PR? Or is it related to this issue here with the speeds not being reported right?

mcuee, 2025-04-28:
> [@mcuee](https://github.com/mcuee) : sorry, what do you mean exactly? Would you like me to test this PR within the context of the new hotplug PR? Or is it related to this issue here with the speeds not being reported right?

Sorry I need to be clear. Not really related to this issue. But rather PR #1498 is supposed to be an example of existing application for platforms without hotplug capability. So it might be a good example to adapt to test backwards compatibility of PR #1633.

mcuee, 2025-04-29:
> Edit: anyway I did it and it was easy. I see things working as expected. I guess the idea is to leave hotplug_fallback as is, I mean: not playing with the opt-in capability because it would make the purpose of the "fallback" obsolete.
>
> I don't think that adding it to my PR makes sense because obviously commit shall be squashed before being merge, but you can play with it be getting the code here: https://github.com/sonatique/libusb/tree/hotplug_fallback_on_top_of_windows_opt_in_hotplug

Thanks a lot for the confirmation that it still works. I will try it as well.

mcuee, 2025-04-29:
@fabiensanglard 

Going back to the existing issue, could you post the debug log with current git head? It may also be interesting to post the debug log with PR #1633. Then we can probably identify the root cause of the issue in the current git. Thanks.

mcuee, 2026-03-01:
Windows hotplug PR has been merged. 
https://github.com/libusb/libusb/commit/0110ab5a3cd9eb11b6cd87ea28f12ad505d349c2

I will close this issue now.

@fabiensanglard Please reopen if you still see the issue with 1.0.31-rc1. Thanks.

closed 2026-03-01 by mcuee