libusb/libusb #1605: Assertion `pthread_mutex_lock(mutex) == 0' failed.

0wwafa, 2025-02-14:
I wrote an application that decodes raw data into a raw video...
It just gets data (very fast) from USB.

I have a problem when closing libusb...

The end of my program is:
```
cleanup:
    if (usb_ctx.dev_handle) {
        libusb_release_interface(usb_ctx.dev_handle, INTERFACE_NUM);
    fprintf(stderr, "Closing libusb device.\n");
        libusb_close(usb_ctx.dev_handle);
    fprintf(stderr, "Libusb device closed.\n");
    }
    libusb_exit(usb_ctx.ctx);
    fprintf(stderr, "Program ended.\n");
    return usb_ctx.return_code;
}
```
If I comment the libusb_close call I don't have this error.
My program waits for the end of every thread.
I tried everything I could.
If I comment the libusb close I just get these warnings:
```
[ 2.084130] [0022cfcc] libusb: debug [libusb_exit]  
[ 2.084154] [0022cfcd] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 2.084221] [0022cfcc] libusb: debug [libusb_unref_device] destroy device 2.36
[ 2.084231] [0022cfcc] libusb: debug [libusb_unref_device] destroy device 2.2
[ 2.084236] [0022cfcc] libusb: debug [libusb_unref_device] destroy device 2.1
[ 2.084241] [0022cfcc] libusb: debug [libusb_unref_device] destroy device 1.3
[ 2.084245] [0022cfcc] libusb: debug [libusb_unref_device] destroy device 4.1
[ 2.084249] [0022cfcc] libusb: debug [libusb_unref_device] destroy device 3.2
[ 2.084254] [0022cfcc] libusb: debug [libusb_unref_device] destroy device 3.1
[ 2.084259] [0022cfcc] libusb: debug [usbi_remove_event_source] remove fd 4
[ 2.084268] [0022cfcc] libusb: debug [usbi_remove_event_source] remove fd 3
[ 2.084278] [0022cfcc] libusb: warning [libusb_exit] device 1.127 still referenced
[ 2.084283] [0022cfcc] libusb: warning [libusb_exit] device 1.2 still referenced
[ 2.084288] [0022cfcc] libusb: warning [libusb_exit] device 1.1 still referenced
[ 2.084292] [0022cfcc] libusb: warning [libusb_exit] application left some devices open
Program ended.
```
but if I leave the code as it is I get this:

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.018872] [0022cf1f] libusb: debug [libusb_init] libusb v1.0.25.11696
[ 0.018933] [0022cf1f] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.018953] [0022cf1f] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.018962] [0022cf1f] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.018984] [0022cf1f] libusb: debug [get_kernel_version] reported kernel version is 5.15.0-131-lowlatency
[ 0.019043] [0022cf1f] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.019055] [0022cf1f] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.019069] [0022cf1f] libusb: debug [op_init] sysfs is available
[ 0.019749] [0022cf20] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.023578] [0022cf1f] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.023590] [0022cf1f] libusb: debug [linux_get_device_address] scan usb3
[ 0.023627] [0022cf1f] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.023633] [0022cf1f] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.023638] [0022cf1f] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.023753] [0022cf1f] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.023761] [0022cf1f] libusb: debug [linux_get_device_address] scan 3-3
[ 0.023791] [0022cf1f] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.023796] [0022cf1f] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.023800] [0022cf1f] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.023837] [0022cf1f] libusb: debug [linux_get_parent_info] dev 0x55e64f3efec0 (3-3) has parent 0x55e64f411e40 (usb3) port 3
[ 0.023917] [0022cf1f] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.023923] [0022cf1f] libusb: debug [linux_get_device_address] scan usb4
[ 0.023952] [0022cf1f] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.023958] [0022cf1f] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[... 148 lines cut ...]
[ 9.363057] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f7774001330
[ 9.366449] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f7774001330
[ 9.366596] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f7774001460
[ 9.369933] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f7774001460
[ 9.370028] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f7774001590
[ 9.373276] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f7774001590
[ 9.373385] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f77740016c0
[ 9.375922] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f77740016c0
[ 9.376001] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f77740017f0
[ 9.379315] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f77740017f0
[ 9.379374] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f7774001920
[ 9.382629] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f7774001920
[ 9.382751] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f7774001a50
[ 9.386091] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f7774001a50
[ 9.386181] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f7774001b80
[ 9.388878] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f7774001b80
[ 9.388961] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f7774001cb0
[ 9.391902] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f7774001cb0
[ 9.392141] [0022cf21] libusb: debug [libusb_cancel_transfer] transfer 0x7f7774001de0
[ 9.395489] [0022cf21] libusb: debug [libusb_free_transfer] transfer 0x7f7774001de0
[ 9.395672] [0022cf1f] libusb: debug [libusb_release_interface] interface 0
Closing libusb device.
[ 9.395768] [0022cf1f] libusb: debug [libusb_close]  
fx2img: ../../libusb/os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed.
```

seanm, 2025-02-14:
1.0.25 is old.  Can you try master?

0wwafa, 2025-02-14:
> 1.0.25 is old. Can you try master?

I did! Same thing.

Oh and by the way, if you set libusb_debug and then run lsusb you see the same warnings.

Perhaps they thought as I did that a warning is better than an unhandled error.

0wwafa, 2025-02-14:
if it helps, this is an strace log when I use libusb_close()
```
ioctl(7, USBDEVFS_REAPURBNDELAY, 0x7ffff7c85ce0) = 0
write(2, "[ 0.603557] [0023bdd2] libusb: d"..., 94) = 94
write(2, "[ 0.603629] [0023bdd2] libusb: d"..., 107) = 107
write(2, "[ 0.603702] [0023bdd2] libusb: d"..., 104) = 104
write(2, "[ 0.603787] [0023bdd2] libusb: d"..., 123) = 123
ioctl(7, USBDEVFS_REAPURBNDELAY, 0x7ffff7c85ce0) = -1 EAGAIN (Resource temporarily unavailable)
write(2, "[ 0.603903] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4b69d0) = 0
write(2, "[ 0.606970] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e2724a000, 266240)          = 0
write(2, "[ 0.607170] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4afc80) = 0
write(2, "[ 0.609986] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e27209000, 266240)          = 0
write(2, "[ 0.610262] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4a96b0) = 0
write(2, "[ 0.612951] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e271c8000, 266240)          = 0
write(2, "[ 0.613202] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df499a50) = 0
write(2, "[ 0.615999] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e27187000, 266240)          = 0
write(2, "[ 0.616278] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df49a0b0) = 0
write(2, "[ 0.618946] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e27146000, 266240)          = 0
write(2, "[ 0.619170] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df498ac0) = 0
write(2, "[ 0.621972] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e27105000, 266240)          = 0
write(2, "[ 0.622288] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4ae860) = 0
write(2, "[ 0.624955] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e270c4000, 266240)          = 0
write(2, "[ 0.625216] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4bb230) = 0
write(2, "[ 0.627961] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e27083000, 266240)          = 0
write(2, "[ 0.628218] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4badf0) = 0
write(2, "[ 0.630957] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e27042000, 266240)          = 0
write(2, "[ 0.631262] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4a99f0) = 0
write(2, "[ 0.634002] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e27001000, 266240)          = 0
write(2, "[ 0.634332] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4b39e0) = 0
write(2, "[ 0.636947] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e26fc0000, 266240)          = 0
write(2, "[ 0.637174] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4b2100) = 0
write(2, "[ 0.639976] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e26f7f000, 266240)          = 0
write(2, "[ 0.640255] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4b1940) = 0
write(2, "[ 0.642961] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e26f3e000, 266240)          = 0
write(2, "[ 0.643239] [0023bdd2] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x5559df4b3670) = 0
write(2, "[ 0.646128] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e26efd000, 266240)          = 0
write(2, "[ 0.646402] [0023bdd2] libusb: d"..., 86) = 86
write(2, "[ 0.646510] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e26ebc000, 266240)          = 0
write(2, "[ 0.646715] [0023bdd2] libusb: d"..., 86) = 86
write(2, "[ 0.646867] [0023bdd2] libusb: d"..., 84) = 84
munmap(0x7f4e26e7b000, 266240)          = 0
write(2, "[ 0.647062] [0023bdd2] libusb: d"..., 76) = 76
ioctl(7, USBDEVFS_RELEASEINTERFACE, 0x7ffff7c85fc4) = 0
ioctl(7, USBDEVFS_IOCTL, 0x7ffff7c85f90) = 0
write(2, "Closing libusb device.\n", 23) = 23
write(2, "[ 0.647388] [0023bdd2] libusb: d"..., 54) = 54
write(2, "fx2img: ../../libusb/os/threads_"..., 112) = 112
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f4e27e3e000
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
gettid()                                = 2342354
getpid()                                = 2342354
tgkill(2342354, 2342354, SIGABRT)       = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=2342354, si_uid=0} ---
+++ killed by SIGABRT (core dumped) +++
```

0wwafa, 2025-02-14:
this instead is if I comment the libusb close:
```
ioctl(7, USBDEVFS_REAPURBNDELAY, 0x7ffe72375530) = 0
write(2, "[ 0.571440] [0023be0a] libusb: d"..., 94) = 94
write(2, "[ 0.571531] [0023be0a] libusb: d"..., 107) = 107
write(2, "[ 0.571599] [0023be0a] libusb: d"..., 104) = 104
write(2, "[ 0.571665] [0023be0a] libusb: d"..., 123) = 123
ioctl(7, USBDEVFS_REAPURBNDELAY, 0x7ffe72375530) = -1 EAGAIN (Resource temporarily unavailable)
write(2, "[ 0.571781] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d109d0) = 0
write(2, "[ 0.574679] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f98753de000, 266240)          = 0
write(2, "[ 0.574937] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d09c80) = 0
write(2, "[ 0.577631] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f987539d000, 266240)          = 0
write(2, "[ 0.577805] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d036b0) = 0
write(2, "[ 0.580633] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f987535c000, 266240)          = 0
write(2, "[ 0.580862] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993cf3a50) = 0
write(2, "[ 0.583652] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f987531b000, 266240)          = 0
write(2, "[ 0.583970] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993cf40b0) = 0
write(2, "[ 0.586646] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f98752da000, 266240)          = 0
write(2, "[ 0.586890] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993cf2ac0) = 0
write(2, "[ 0.589645] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f9875299000, 266240)          = 0
write(2, "[ 0.589910] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d08860) = 0
write(2, "[ 0.592640] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f9875258000, 266240)          = 0
write(2, "[ 0.592898] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d15230) = 0
write(2, "[ 0.595619] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f9875217000, 266240)          = 0
write(2, "[ 0.595802] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d14df0) = 0
write(2, "[ 0.598642] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f98751d6000, 266240)          = 0
write(2, "[ 0.598881] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d039f0) = 0
write(2, "[ 0.601641] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f9875195000, 266240)          = 0
write(2, "[ 0.601850] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d0d9e0) = 0
write(2, "[ 0.604658] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f9875154000, 266240)          = 0
write(2, "[ 0.604944] [0023be0a] libusb: d"..., 86) = 86
write(2, "[ 0.605088] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f9875113000, 266240)          = 0
write(2, "[ 0.605317] [0023be0a] libusb: d"..., 86) = 86
write(2, "[ 0.605426] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f98750d2000, 266240)          = 0
write(2, "[ 0.605655] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d0d670) = 0
write(2, "[ 0.608640] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f9875091000, 266240)          = 0
write(2, "[ 0.608885] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d03e20) = 0
write(2, "[ 0.611685] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f9875050000, 266240)          = 0
write(2, "[ 0.611971] [0023be0a] libusb: d"..., 86) = 86
ioctl(7, USBDEVFS_DISCARDURB, 0x562993d08820) = 0
write(2, "[ 0.614635] [0023be0a] libusb: d"..., 84) = 84
munmap(0x7f987500f000, 266240)          = 0
write(2, "[ 0.614913] [0023be0a] libusb: d"..., 76) = 76
ioctl(7, USBDEVFS_RELEASEINTERFACE, 0x7ffe72375814) = 0
ioctl(7, USBDEVFS_IOCTL, 0x7ffe723757e0) = 0
write(2, "[ 0.615174] [0023be0a] libusb: d"..., 53) = 53
write(6, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x7f9875c1f910, FUTEX_WAIT_BITSET|FUTEX_CLOCK_REALTIME, 2342429, NULL, FUTEX_BITSET_MATCH_ANY) = 0
close(6)                                = 0
close(5)                                = 0
write(2, "[ 0.615556] [0023be0a] libusb: d"..., 79) = 79
write(2, "[ 0.615642] [0023be0a] libusb: d"..., 78) = 78
write(2, "[ 0.615718] [0023be0a] libusb: d"..., 78) = 78
write(2, "[ 0.615789] [0023be0a] libusb: d"..., 78) = 78
write(2, "[ 0.615859] [0023be0a] libusb: d"..., 78) = 78
write(2, "[ 0.615930] [0023be0a] libusb: d"..., 78) = 78
write(2, "[ 0.616000] [0023be0a] libusb: d"..., 78) = 78
write(2, "[ 0.616097] [0023be0a] libusb: d"..., 76) = 76
close(4)                                = 0
write(2, "[ 0.616263] [0023be0a] libusb: d"..., 76) = 76
close(3)                                = 0
write(2, "[ 0.616430] [0023be0a] libusb: w"..., 81) = 81
write(2, "[ 0.616562] [0023be0a] libusb: w"..., 81) = 81
write(2, "[ 0.616654] [0023be0a] libusb: w"..., 81) = 81
write(2, "[ 0.616743] [0023be0a] libusb: w"..., 88) = 88
write(2, "Program ended.\n", 15)        = 15
exit_group(0)                           = ?
+++ exited with 0 +++
```

0wwafa, 2025-02-14:
@mcuee this is not a question. 99% is a bug.

the bug shows up only if you start libusb transfers. if you don't, libusb close works normally.
in my program all transfers are completed or cancelled on program exit. I checked the code a hunderd times.

seanm, 2025-02-16:
Have you used Thread Sanitizer before? Could you try with it?

tormodvolden, 2025-02-23:
> Oh and by the way, if you set libusb_debug and then run lsusb you see the same warnings.
> 
> Perhaps they thought as I did that a warning is better than an unhandled error.

They fixed it in usbutils v018, thanks to mcuee.

mcuee, 2025-02-24:
> [@mcuee](https://github.com/mcuee) this is not a question. 99% is a bug.

We will update the label if it is confirmed to be a libusb bug.

sonatique, 2026-05-17:
@user20793: thank you very much for your report. I think your symptom (assertion on the first bulk transfer from a non-opening thread, before any free has happened) almost certainly has a different root cause from @0wwafa's. The cancel-then-free use-after-free that explains the original report does not apply when no transfers have been freed yet.

Could you please open a new issue with:

- a minimal reproducer (open + claim on thread A, single libusb_bulk_transfer() on thread B, nothing else);
- the full debug log (LIBUSB_DEBUG=4);
- a backtrace from the crash (gdb or coredumpctl);
- ideally a ThreadSanitizer run, since this sounds like a cross-thread ordering issue.

Keeping it separate will let it be triaged on its own without conflating two different bugs in this thread.

Thank you very much in advance!