libusb/libusb #1608: [1.0.27] [x86_64 Linux] undefined symbol: libusb_init_context

stefanGaina, 2025-02-20:
Hello! I have downloaded libusb-1.0.27.tar.bz2 and ran the commands: ./configure, make, sudo make install. I now have the library installed in /usr/local/lib:

/usr/local/lib$ ls
gcc           libusb-1.0.la  libusb-1.0.so.0      pkgconfig  python3.10
libusb-1.0.a  libusb-1.0.so  libusb-1.0.so.0.4.0  python2.7  x86_64-linux-gnu

The issue I encounter is libusb_init_context() function is not found.

/example$ cat main.cpp
#include <iostream>
#include <libusb-1.0/libusb.h>

int main()
{
	if (LIBUSB_API_VERSION >= 0x0100010A)
	{
		std::cout << "The API version is at least 1.0.27\n";
	}

#ifdef USE_DEPRECATED_INIT
	(void)libusb_init(nullptr);
#else
	(void)libusb_init_context(nullptr, nullptr, 0);
#endif

	std::cout << "Success!\n";
	return 0;
}
/example$ /usr/bin/g++ main.cpp -lusb-1.0
/example$ ./a.out
./a.out: symbol lookup error: ./a.out: undefined symbol: libusb_init_context
/example$ /usr/bin/g++ main.cpp -DUSE_DEPRECATED_INIT -lusb-1.0
/example$ ./a.out 
The API version is at least 1.0.27
Success!
/example$ /usr/bin/g++ --version
g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0

Could you please help me configure the library correctly so that I have access to the newest functions? Thank you in advance!

Youw, 2025-02-20:
Please share the following:

1) The output of `sudo make install`
2) `ldd a.out`
3) `pkg-config --libs --cflags --debug libusb-1.0`

stefanGaina, 2025-02-20:
Sure!

/libusb-1.0.27$ **sudo make install**
Making install in libusb
make[1]: Entering directory '/path/to/libusb-1.0.27/libusb'
make[2]: Entering directory '/path/to/libusb-1.0.27/libusb'
 /usr/bin/mkdir -p '/usr/local/lib'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   libusb-1.0.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libusb-1.0.so.0.4.0 /usr/local/lib/libusb-1.0.so.0.4.0
libtool: install: (cd /usr/local/lib && { ln -s -f libusb-1.0.so.0.4.0 libusb-1.0.so.0 || { rm -f libusb-1.0.so.0 && ln -s libusb-1.0.so.0.4.0 libusb-1.0.so.0; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libusb-1.0.so.0.4.0 libusb-1.0.so || { rm -f libusb-1.0.so && ln -s libusb-1.0.so.0.4.0 libusb-1.0.so; }; })
libtool: install: /usr/bin/install -c .libs/libusb-1.0.lai /usr/local/lib/libusb-1.0.la
libtool: install: /usr/bin/install -c .libs/libusb-1.0.a /usr/local/lib/libusb-1.0.a
libtool: install: chmod 644 /usr/local/lib/libusb-1.0.a
libtool: install: ranlib /usr/local/lib/libusb-1.0.a
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/sbin" ldconfig -n /usr/local/lib

Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.

 /usr/bin/mkdir -p '/usr/local/include/libusb-1.0'
 /usr/bin/install -c -m 644 libusb.h '/usr/local/include/libusb-1.0'
make[2]: Leaving directory '/path/to/libusb-1.0.27/libusb'
make[1]: Leaving directory '/path/to/libusb-1.0.27/libusb'
make[1]: Entering directory '/path/to/libusb-1.0.27'
make[2]: Entering directory '/path/to/libusb-1.0.27'
make[2]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 libusb-1.0.pc '/usr/local/lib/pkgconfig'
make[2]: Leaving directory '/path/to/libusb-1.0.27'
make[1]: Leaving directory '/path/to/libusb-1.0.27'

/example$ **ldd a.out**
	linux-vdso.so.1 (0x00007ffcc5c82000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x000071b256482000)
	libstdc++.so.6 => /usr/local/lib64/libstdc++.so.6 (0x000071b256200000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x000071b255e00000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x000071b2561d6000)
	/lib64/ld-linux-x86-64.so.2 (0x000071b2564c6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x000071b2560ef000)
	libgcc_s.so.1 => /usr/local/lib64/libgcc_s.so.1 (0x000071b2560c1000)

/example$ **pkg-config --libs --cflags --debug libusb-1.0**
Error printing enabled by default due to use of output options besides --exists, --atleast/exact/max-version or --list-all. Value of --silence-errors: 0
Error printing enabled
Adding virtual 'pkg-config' package to list of known packages
Looking for package 'libusb-1.0'
Looking for package 'libusb-1.0-uninstalled'
Reading 'libusb-1.0' from file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
Parsing package file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>prefix=/usr/local
 Variable declaration, 'prefix' has value '/usr/local'
  line>exec_prefix=${prefix}
 Variable declaration, 'exec_prefix' has value '/usr/local'
  line>libdir=${exec_prefix}/lib
 Variable declaration, 'libdir' has value '/usr/local/lib'
  line>includedir=${prefix}/include
 Variable declaration, 'includedir' has value '/usr/local/include'
  line>
  line>Name: libusb-1.0
  line>Description: C API for USB device access from Linux, Mac OS X, Windows, OpenBSD/NetBSD and Solaris userspace
  line>Version: 1.0.27
  line>Libs: -L${libdir} -lusb-1.0
  line>Libs.private: -ludev 
Unknown keyword 'Libs.private' in '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>Cflags: -I${includedir}/libusb-1.0
Path position of 'libusb-1.0' is 2
Adding 'libusb-1.0' to list of known packages
 post-recurse: libusb-1.0
adding CFLAGS_OTHER string ""
 post-recurse: libusb-1.0
 original: libusb-1.0
   sorted: libusb-1.0
adding CFLAGS_I string "-I/usr/local/include/libusb-1.0 "
 post-recurse: libusb-1.0
 original: libusb-1.0
   sorted: libusb-1.0
adding LIBS_L string "-L/usr/local/lib "
 post-recurse: libusb-1.0
adding LIBS_OTHER | LIBS_l string "-lusb-1.0 "
returning flags string "-I/usr/local/include/libusb-1.0 -L/usr/local/lib -lusb-1.0"
-I/usr/local/include/libusb-1.0 -L/usr/local/lib -lusb-1.0

Youw, 2025-02-20:
> As you can see:

```
/example$ ldd a.out
...
libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x000071b256482000)
...
```

You are linking against a different version of libusb - the one you have installed in your system, probably with `apt` or so.

To avoid this ambiguity use `pkg-config` to add libusb as your dependency, e.g.:
```bash
/usr/bin/g++ main.cpp $(pkg-config --libs --cflags libusb-1.0)
```

or even to make sure you use this specific version of libusb (yes, this is a single bash command line):
```bash
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig /usr/bin/g++ main.cpp $(pkg-config --libs --cflags libusb-1.0)
```

stefanGaina, 2025-02-20:
I previously had it installed via apt, but apparently I did not remove it correctly (sudo apt autoremove fixed the issue). Thank you very much for the support!

closed 2025-02-20 by stefanGaina