libusb/libusb #1609: Where should the files get installed to be found by RTL-SDR?

eabase, 2025-02-23:
Trying to use the libs with RTL-SDR, but RTL-SDR cannot find the libraries. 
The instructions here are saying **how** and **where** this gets installed in Windows.

Where should all the files go? 
What kind of environment variables are needed to tell other packages where to find these files?

tormodvolden, 2025-02-23:
On Windows the simplest is to put the libusb DLLs in the same folder as the application executables. To have them shared among applications, you can also put them in the C:\windows\system32 folder.

eabase, 2025-02-24:
Hi Tormod, @tormodvolden 
Sorry, I wasn't being clear. 

I need to build the RTL-SDR drivers as their libraries (and god know what other parts) are needed for another package, `SoapyRTLSDR`, which is then needed by `SoapySDR`, which is then needed by `LimeSuitNG`. 🤯 

And why do I need to build them again? Probably because they need to have the same compiler and version as the other tools, AFAIK, and seems to be indicated by some of them.

tormodvolden, 2025-02-24:
If you want to build with CMake, please see https://github.com/libusb/libusb-cmake

Youw, 2025-02-24:
In any case, this doesn't seem like a libusb issue or so, closing this ticket.

closed 2025-02-24 by Youw