libusb/libusb #1615: 1.0.27-rc1 test results

tormodvolden, 2025-03-06:
For as many platforms as possible we want to test building and a run of the RC. At least xusb or lsusb -v should be tested. Please use the libusb-1.0.28-rc1.tar.bz2 tarball, and not a git checkout. Use the included configure script, no autogen or bootstrap needed:

tar jxvf libusb-1.0.28-rc1.tar.bz2
cd libusb-1.0.28-rc1
./configure --enable-tests-build --enable-examples-build
make check

Also more application testing is welcome.

seanm, 2025-03-07:
On my arm64 Mac Studio I ran `make check` in a loop 50 times, no failures.

On on old Intel Mac with Mac OS X 10.11 I get 2 test failures:

```
PASS: stress
FAIL: stress_mt
PASS: set_option
PASS: init_context
FAIL: macos
```

I believe `stress_mt` is known to fail intermittently?

seanm, 2025-03-07:
Also tried on FreeBSD 14.2 and OpenBSD 7.5 VMs (on Intel), all 4 tests pass.

ptribble, 2025-03-07:
Builds cleanly on x86 illumos (Tribblix) with gcc10. `make check` is clean.

mcuee, 2025-03-08:
Last update on 8-March-2025

## Test platforms

### Windows 11 64-bit
- [x] build: (MSYS2 mingw64 and MSYS2 clang64) and `make check`
  - [ ] run: xusb, lsusb, listdev, testlibusb

Lots of compiler warnings with MSYS2, same as github action CI build.
https://github.com/libusb/libusb/actions/runs/13709359027/job/38342293707
https://github.com/libusb/libusb/actions/runs/13709359034/job/38342293704

### Windows 11 64-bit
- [ ] build: (MSVC64)
  - [ ] run: xusb, lsusb, listdev, testlibusb

### Linux x86_64
- [ ] build and `make check`: (Ubuntu 20.04 and 24.04)
  - [ ] run: xusb, lsusb, hotplugtest, listdev, testlibusb

### Linux ARM64 (Raspberry Pi OS 64bit on RPi 500, Debian 12 based)
- [ ] build and `make check` (with udev)
    - [ ] run: xusb, lsusb, hotplugtest, listdev, testlibusb
- [ ] build and `make check` (with netlink)
    - [ ] run: xusb, lsusb, hotplugtest, listdev, testlibusb

### macOS 15.3.1 (ARM64, mac Mini M1 2020)
- [ ] build and `make check` using Apple clang
    - [ ] run: xusb, lsusb, hotplugtest, listdev, testlibusb

## Other platforms

### OpenBSD 7.6  x86_64 virtual machine under Proxmox VE
- [ ] build and `make check` using OpenBSD clang
    - [ ] run: xusb, lsusb, listdev, testlibusb

### NetBSD 10.1 x86_64 virtual machine under Proxmox VE
- [ ] build and `make check` using NetBSD gcc
    - [ ] run: xusb, lsusb, listdev, testlibusb

## Applications
- [ ] Super Speed USB bulk transfer IN & OUT streaming with custom device - with Windows 11/Linux/macOS using libusbdotnet_v2 benchmark application

mcuee, 2025-03-08:
Issue reports for the testing.

1) testlibusb example. `testlibusb -v` works but `testlibusb -vv`  or similar will fail.

Windows 11 24H2, MSys2 Clang64.
```
$ ./examples/testlibusb.exe -vvv
Usage C:/work/libusb/libusb-1.0.28-rc1/examples/.libs/testlibusb.exe [-v] [-d </dev/bus/usb/...>]
Note use -d to test libusb_wrap_sys_device()
```

tormodvolden, 2025-03-09:
> I believe `stress_mt` is known to fail intermittently?

I thought it only had been seen on virtual machines, like the CI builders. Can you see why it fails?

tormodvolden, 2025-03-09:
> 1. `testlibusb -v` works but `testlibusb -vv`  or similar will fail if not running under Linux.

testlibusb  -vv fails also on Linux, because it is an unknown option / doesn't allow repeated options. I don't think this is anything new.

mcuee, 2025-03-09:
> > I believe `stress_mt` is known to fail intermittently?
> 
> I thought it only had been seen on virtual machines, like the CI builders. Can you see why it fails?

Indeed I have the same impression that the only failures reported are the CI builders.

mcuee, 2025-03-09:
> > 1. `testlibusb -v` works but `testlibusb -vv`  or similar will fail if not running under Linux.
> 
> testlibusb -vv fails also on Linux, because it is an unknown option / doesn't allow repeated options. I don't think this is anything new.

Indeed nothing new here. Looks like I never tested it this way previously. Not a big issue since it is just an example. 

But PR will still be welcome to fix this issue from the contributors. ;-)

tormodvolden, 2025-03-15:
### Linux x86_64
- [x] build and `make check`: (Arch Linux), gcc 14.2.1
  - [x] run: xusb, lsusb, listdev, testlibusb
- [x] build and `make check` (incl. umockdev): (Debian unstable), gcc 14.2.0
  - [x] run: isochronous streaming with libfreenect from XBox Kinect camera

mcuee, 2025-03-16:
It is all good for NetBSD 10.1.

Other than `make check`, simple tests with `xusb, listdev, testlibusb` are also done with Raspberry Pico 2 W device using Circuit Python default FW.

<details>
<summary>NetBSD 10.1 VM test results</summary>

```
netbsd101vm1$ ./configure --enable-tests-build --enable-examples-build
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking for inline... inline
checking for a BSD-compatible install... /usr/bin/install -c
checking whether sleep supports fractional seconds... yes
checking filesystem timestamp resolution... 2
checking whether build environment is sane... yes
checking for a race-free mkdir -p... mkdir -p
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking xargs -n works... yes
checking dependency style of gcc... gcc3
checking dependency style of g++... gcc3
checking build system type... x86_64-unknown-netbsd10.1
checking host system type... x86_64-unknown-netbsd10.1
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 196608
checking how to convert x86_64-unknown-netbsd10.1 file names to x86_64-unknown-netbsd10.1 format... func_convert_file_noop
checking how to convert x86_64-unknown-netbsd10.1 file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... match_pattern /lib[^/]+(\.so|_pic\.a)$
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ranlib... ranlib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
[... 108 lines cut ...]
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 1.0.28-rc1
============================================================================
# TOTAL: 4
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================

netbsd101vm1$ sudo usbdevs
addr 1: UHCI root hub, NetBSD
addr 0: xHCI root hub, NetBSD
addr 0: xHCI root hub, NetBSD
 addr 1: Pico 2 W, Raspberry Pi

netbsd101vm1$ uname -a
NetBSD netbsd101vm1.myhome.local 10.1 NetBSD 10.1 (GENERIC) #0: Mon Dec 16 13:08:11 UTC 2024  mkrepro@mkrepro.NetBSD.org:/usr/src/sys/arch/amd64/compile/GENERIC amd64
```

</details>

mcuee, 2025-03-16:
`make check` is fine under OpenBSD 7.6 VM.

<details>
<summary>OpenBSD 7.6 VM test results with `make check`</summary>

```
openbsd76vm1$ cd libusb-1.0.28-rc1

openbsd76vm1$ ./configure --enable-tests-build --enable-examples-build
checking for gcc... no
checking for cc... cc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether cc accepts -g... yes
checking for cc option to enable C11 features... none needed
checking whether cc understands -c and -o together... yes
checking for g++... no
checking for c++... c++
checking whether the compiler supports GNU C++... yes
checking whether c++ accepts -g... yes
checking for c++ option to enable C++11 features... none needed
checking for inline... inline
checking for a BSD-compatible install... /usr/bin/install -c
checking whether sleep supports fractional seconds... yes
checking filesystem timestamp resolution... 0.01
checking whether build environment is sane... yes
checking for a race-free mkdir -p... mkdir -p
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking xargs -n works... yes
checking dependency style of cc... gcc3
checking dependency style of c++... gcc3
checking build system type... x86_64-unknown-openbsd7.6
checking host system type... x86_64-unknown-openbsd7.6
checking how to print strings... print -r
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by cc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 393216
checking how to convert x86_64-unknown-openbsd7.6 file names to x86_64-unknown-openbsd7.6 format... func_convert_file_noop
checking how to convert x86_64-unknown-openbsd7.6 file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... match_pattern /lib[^/]+(\.so\.[0-9]+\.[0-9]+|\.so|_pic\.a)$
checking for dlltool... no
checking how to associate runtime and link libraries... print -r --
checking for ranlib... ranlib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking command to parse /usr/bin/nm -B output from cc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
[... 99 lines cut ...]
  CC       stress_mt-stress_mt.o
  CCLD     stress_mt
  CC       set_option.o
  CCLD     set_option
  CC       init_context.o
  CCLD     init_context
make  check-TESTS
PASS: stress
PASS: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 1.0.28-rc1
============================================================================
# TOTAL: 4
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
```

</details>

mcuee, 2025-03-16:
No issues with my new Raspberry Pi 500, tested with both libudev and with `--disable-udev`.

mcuee, 2025-03-16:
Cypress EZ-USB FX2LP Bulkloop example -- High Speed USB Bulk Transfer

Using libusbK.sys, libusb0.sys and WinUSB driver --> all are okay under Windows 11 x64.

```
PS C:\work\libusb\libusbdotnet\libusbdotnet_v2_git\stage\BenchmarkCon\bin\Debug\net6.0> .\BenchmarkCon.exe list notestselect
LibUsbDotNet USB Benchmark v2.2.0.0
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
1. 1C7A:0575 Egis Technology Inc. - EgisTec Touch Fingerprint Sensor
2. 04B4:1004 Cypress Semiconductor Corp. - EZ-USB

Select device (1-2) :2

Benchmark device 04B4:1004 opened..
Loop Test Information
        Vid / Pid       : 04B4h / 1004h
        Interface #     : 00h
        Alt Interface # : 00h
        Priority        : Normal
        Buffer Size     : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Read (Ep86h) max packet size: 512
Bulk Write (Ep02h) max packet size: 512

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 408410.77 Transfers: 97 Bytes/s: 408410.77
Avg. Bytes/s: 404294.04 Transfers: 195 Bytes/s: 400300.25
Avg. Bytes/s: 406147.66 Transfers: 296 Bytes/s: 409774.95
Avg. Bytes/s: 406366.53 Transfers: 396 Bytes/s: 407015.78
Avg. Bytes/s: 406154.68 Transfers: 496 Bytes/s: 405317.90
Avg. Bytes/s: 406641.96 Transfers: 596 Bytes/s: 409076.26
Avg. Bytes/s: 406797.61 Transfers: 697 Bytes/s: 407718.56
Avg. Bytes/s: 407043.41 Transfers: 797 Bytes/s: 408764.93
Avg. Bytes/s: 407386.16 Transfers: 898 Bytes/s: 410111.18
Avg. Bytes/s: 407289.79 Transfers: 998 Bytes/s: 406426.42
Avg. Bytes/s: 407548.77 Transfers: 1099 Bytes/s: 410125.57
Avg. Bytes/s: 407750.68 Transfers: 1201 Bytes/s: 409938.96
Avg. Bytes/s: 407613.64 Transfers: 1300 Bytes/s: 405958.43
Avg. Bytes/s: 407723.43 Transfers: 1401 Bytes/s: 409141.96
Avg. Bytes/s: 407774.36 Transfers: 1501 Bytes/s: 408489.24
Avg. Bytes/s: 407893.18 Transfers: 1602 Bytes/s: 409667.17
Avg. Bytes/s: 408013.07 Transfers: 1703 Bytes/s: 409924.08
Avg. Bytes/s: 408140.63 Transfers: 1803 Bytes/s: 410325.37
Avg. Bytes/s: 408245.94 Transfers: 1904 Bytes/s: 410135.08
Avg. Bytes/s: 408176.86 Transfers: 2005 Bytes/s: 406878.91
Avg. Bytes/s: 407883.65 Transfers: 2104 Bytes/s: 402034.68
Avg. Bytes/s: 407852.68 Transfers: 2203 Bytes/s: 407195.69
Avg. Bytes/s: 407941.87 Transfers: 2304 Bytes/s: 409897.07
Avg. Bytes/s: 408033.34 Transfers: 2404 Bytes/s: 410152.06
Avg. Bytes/s: 407883.02 Transfers: 2505 Bytes/s: 404337.57
Avg. Bytes/s: 407958.56 Transfers: 2606 Bytes/s: 409841.12
Avg. Bytes/s: 407951.94 Transfers: 2706 Bytes/s: 407779.47
Avg. Bytes/s: 407855.92 Transfers: 2806 Bytes/s: 405274.83
qAvg. Bytes/s: 407925.48 Transfers: 2907 Bytes/s: 409867.51
stopped Ep86h thread.
stopped Ep02h thread.
Loop Test Information
        Vid / Pid       : 04B4h / 1004h
[... 82 lines cut ...]
        Alt Interface # : 00h
        Num Endpoints   : 4
        Priority        : 0
        Read Size       : 4096
        Write Size      : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Read (Ep86h) Maximum Packet Size: 512
        Total Bytes     : 5332992
        Total Transfers : 1302
        Avg. Bytes/sec  : 408282.96
        Elapsed Time    : 13.06 seconds

Bulk Write (Ep02h) Maximum Packet Size: 512
        Total Bytes     : 5332992
        Total Transfers : 1302
        Avg. Bytes/sec  : 408282.96
        Elapsed Time    : 13.06 seconds

Press any key to exit..
```

tormodvolden, 2025-03-17:
Found a regression #1619, so holding back the release.

seanm, 2025-03-18:
> > I believe `stress_mt` is known to fail intermittently?
> 
> I thought it only had been seen on virtual machines, like the CI builders. Can you see why it fails?

I've just tried with thread sanitizer and it detects races when running stress_mt. Though TSan sometimes has false positives, considering we see this fail intermittently I'd guess they are true positives. Problem not clear though, I'll need to spend more time...

sl1200mk2, 2025-03-19:
Hi, 
on macOS I've two warnings:
Making all in libusb
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
io.c:2856:27: warning: unused variable 'transfer_to_cancel' [-Wunused-variable]
 2856 |                 struct libusb_transfer *transfer_to_cancel = USBI_TRANSFER_TO_LIBUSB_TRANSFER(to_cancel);
      |                                         ^~~~~~~~~~~~~~~~~~
1 warning generated.
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_posix.lo
  CC       os/threads_posix.lo
  CC       os/darwin_usb.lo
os/darwin_usb.c:2654:15: warning: unused variable 'transfer_types' [-Wunused-variable]
 2654 |   const char *transfer_types[] = {"control", "isoc", "bulk", "interrupt", "bulk-stream", NULL};
      |               ^~~~~~~~~~~~~~
1 warning generated.

best regards,
nicolas

tormodvolden, 2025-03-19:
@sl1200mk2 You may get these warnings if you are building without logging. The configure script included in the RC has logging enabled by default. Do you explicitly disable logging when you build?

sl1200mk2, 2025-03-19:
@tormodvolden yes

tormodvolden, 2025-03-19:
OK. It is normal, and we have the same warnings in our "disable-log" Linux CI builds. Thanks anyway for reporting it!