libusb/libusb #1616: segfault when using adb

Gounlaf, 2025-03-09:
Good evening,

I'm having trouble with a device that seems to disconnect due to a segfault.

Some context :
- it's a phone (Google Pixel 6), connected via USB, debug mode
- I'm using https://github.com/Genymobile/scrcpy to remote control it
- I've updated the Android version few days ago (and the problem starts here)
- according to apt, i'm using the latest stable libusb version (I don't remember when it was updated on my computer)

So, scrcpy use adb to push some library/binaries and then allow me to control my device remotely.
It was working "before" the Android update (march 2025).
It doesn't work anymore: a segfault happens
It still works over tcp.

Except this dmesg log, what can I provide you (or what can I try) to understand if it comes from libusb or something else?

Thanks,
Regards.

```
[118546.226525] usb 4-4: new SuperSpeed Plus Gen 2x1 USB device number 127 using xhci_hcd
[118546.239946] usb 4-4: LPM exit latency is zeroed, disabling LPM.
[118546.241137] usb 4-4: New USB device found, idVendor=18d1, idProduct=4ee7, bcdDevice= 5.10
[118546.241152] usb 4-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[118546.241160] usb 4-4: Product: Pixel 6
[118546.241166] usb 4-4: Manufacturer: Google
[118546.241172] usb 4-4: SerialNumber: XXXXXX
[119287.505081] libusb[3738]: segfault at 5a1edc934d9c ip 0000759b6369fff4 sp 0000759b60bffb68 error 4 in libc.so.6[759b63628000+188000] likely on CPU 0 (core 0, socket 0)
[119287.505169] Code: 00 e8 20 b3 f9 ff 48 8d 0d a9 4d 13 00 ba 5f 02 00 00 48 8d 35 82 c4 12 00 48 8d 3d a6 c4 12 00 e8 01 b3 f9 ff 90 f3 0f 1e fa <8b> 47 10 89 c2 81 e2 7f 01 00 00 90 83 e0 7c 0f 85 a7 00 00 00 55
[119311.735755] usb 4-4: USB disconnect, device number 127
[119312.520096] usb 4-4: new SuperSpeed Plus Gen 2x1 USB device number 2 using xhci_hcd
[119312.533093] usb 4-4: LPM exit latency is zeroed, disabling LPM.
[119312.534472] usb 4-4: New USB device found, idVendor=18d1, idProduct=4ee7, bcdDevice= 5.10
[119312.534503] usb 4-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[119312.534511] usb 4-4: Product: Pixel 6
[119312.534517] usb 4-4: Manufacturer: Google
[119312.534522] usb 4-4: SerialNumber: XXXXXX
```

```
sudo apt info libusb-1.0-0
Package: libusb-1.0-0
Version: 2:1.0.27-1
Priority: standard
Section: libs
Source: libusb-1.0
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Aurelien Jarno <aurel32@debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 140 kB
Depends: libc6 (>= 2.38), libudev1 (>= 183)
Homepage: http://www.libusb.info
Task: cloud-minimal, standard, server-minimal
Download-Size: 54.0 kB
APT-Manual-Installed: no
APT-Sources: http://archive.ubuntu.com/ubuntu noble/main amd64 Packages
Description: userspace USB programming library
 Library for programming USB applications without the knowledge
 of Linux kernel internals.
 .
 This package contains what you need to run programs that use this
 library.
```

tormodvolden, 2025-03-09:
Does other adb operations (other than scrcpy) work fine?

Maybe you can try this to debug it:
Run `adb devices` to make sure the daemon has started, then `ps ax | grep adb` to get the PID of the adb daemon. Then attach to it with gdb (replace with the full path of your adb binary, use `which adb` if in doubt): `sudo gdb /usr/bin/adb PID`
At the (gdb) prompt type `cont` then plug in your phone and see if the crash can be reproduced. If so, type `thread apply all bt` to get a backtrace.

If you use the adb shipped by Google, it will probably have its own libusb statically linked, and your distro libusb is not used. In that case please also try the distro "adb".

Gounlaf, 2025-03-10:
Hi,

> Does other adb operations (other than scrcpy) work fine?

`scrcopy` is not an `adb` command; it's another software that use adb.

> If you use the adb shipped by Google, it will probably have its own libusb statically linked, and your distro libusb is not used. In that case please also try the distro "adb".

Yes I used adb distribued by Google. Tryed the one shipped by Ubuntu : same issue.

Will do the gdb debug once I have physical access to my device.

Interesting things to note: there is no crash on my laptop, with my Pixel 8:
- same Android version
- same libusb version
- same adb version/distributor

I will do the others tests once physically possible (next week).

Thanks for your help.

1oh1, 2025-03-12:
This may be relevant to an issue we're seeing on the latest ADB version. Are you using ADB v35? If you kill ADB server and start it with `export ADB_LIBUSB=0; adb start-server` does that work ?

Gounlaf, 2025-03-12:
Hi @1oh1,

> Are you using ADB v35

Yes

```shell
$ adb --version
Android Debug Bridge version 1.0.41
Version 35.0.2-12147458
```

> If you kill ADB server and start it with export ADB_LIBUSB=0; adb start-server does that work ?

No. Still crash. But the libusb segfault is not displayed anymore in dmesg messages.

tormodvolden, 2025-03-12:
Just noting down that adb uses libusb by default on Linux since adb 35.0.0 (February 2024), but on macOS already since adb 34.0.5 (October 2023). Previous versions, since 25.0.4 (March 16, 2017), could force it with `ADB_LIBUSB=true`.

tormodvolden, 2025-03-16:
Yes, perfect, exactly what I wanted.

I think this means it is crashing in adb code, just in a callback executed by libusb code:
```
#0  ___pthread_mutex_lock (mutex=0x5d82662ae8b1) at ./nptl/pthread_mutex_lock.c:80
#1  0x00005d8795722ec6 in std::__1::mutex::lock() ()
#2  0x00005d8795382e17 in LibusbConnection::write_cb(libusb_transfer*) ()
#3  0x00005d879564dce3 in usbi_handle_transfer_completion ()
```
And if you also see a similar crash with ADB_LIBUSB=0 it seems like the problem is not in libusb, but in adb.

Gounlaf, 2025-03-17:
> Yes, perfect, exactly what I wanted.
>
> I think this means it is crashing in adb code, just in a callback executed by libusb code:
>
> ```
> #0  ___pthread_mutex_lock (mutex=0x5d82662ae8b1) at [...]

```shell
gounlaf@zotac:~$ adb kill-server
gounlaf@zotac:~$ export ADB_LIBUSB=0; adb start-server
* daemon not running; starting now at tcp:5037
* daemon started successfully
gounlaf@zotac:~$ ps ax | grep adb
   7509 ?        Ssl    0:00 adb -L tcp:5037 fork-server server --reply-fd 4
   7518 pts/0    S+     0:00 grep --color=auto adb
gounlaf@zotac:~$ adb devices -l
List of devices attached
18041FDF60000Y         device usb:4-4 product:oriole model:Pixel_6 device:oriole transport_id:1
gounlaf@zotac:~$ sudo gdb /home/gounlaf/platform-tools/adb 7509
[sudo] password for gounlaf: 
GNU gdb (Ubuntu 15.0.50.20240403-0ubuntu1) 15.0.50.20240403-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/gounlaf/platform-tools/adb...
(No debugging symbols found in /home/gounlaf/platform-tools/adb)
Attaching to program: /home/gounlaf/platform-tools/adb, process 7509
[New LWP 7513]
[New LWP 7512]
[New LWP 7511]
[New LWP 7510]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
0x000073dcdd32a042 in epoll_wait (epfd=3, events=0x6033937dad20, maxevents=6, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
warning: 30	../sysdeps/unix/sysv/linux/epoll_wait.c: No such file or directory
(gdb) cont
Continuing.
[New Thread 0x73dcda8006c0 (LWP 7543)]
[New Thread 0x73dcd9e006c0 (LWP 7544)]
[Thread 0x73dcd9e006c0 (LWP 7544) exited]
[Thread 0x73dcda8006c0 (LWP 7543) exited]
[New Thread 0x73dcd9e006c0 (LWP 7546)]
[New Thread 0x73dcda8006c0 (LWP 7547)]
[Thread 0x73dcda8006c0 (LWP 7547) exited]
[Thread 0x73dcd9e006c0 (LWP 7546) exited]
[New Thread 0x73dcda8006c0 (LWP 7548)]
[New Thread 0x73dcd9e006c0 (LWP 7549)]
[Thread 0x73dcd9e006c0 (LWP 7549) exited]
[Thread 0x73dcda8006c0 (LWP 7548) exited]
[New Thread 0x73dcd9e006c0 (LWP 7550)]
[New Thread 0x73dcda8006c0 (LWP 7551)]
[Thread 0x73dcda8006c0 (LWP 7551) exited]
[Thread 0x73dcd9e006c0 (LWP 7550) exited]
[New Thread 0x73dcda8006c0 (LWP 7552)]
[New Thread 0x73dcd9e006c0 (LWP 7553)]
[Thread 0x73dcd9e006c0 (LWP 7553) exited]
[Thread 0x73dcda8006c0 (LWP 7552) exited]
[...] and lot and lot and lot of similar thread things
```

scrcopy still crash/failed because adb is not responding correctly

```
scrcpy        
scrcpy 3.1 <https://github.com/Genymobile/scrcpy>
INFO: ADB device found:
INFO:     -->   (usb)  18041FDF60000Y                  device  Pixel_6
/opt/homebrew/Cellar/scrcpy/3.1/share/scrcpy/scrcpy-server: 1 file pushed, 0 skipped. 25.1 MB/s (90640 bytes in 0.003s)
adb: error: failed to read copy response
ERROR: "adb push" returned with value 1
ERROR: Server connection failed
```

dmesg doesn't show a "refresh" on USB, nor a libusb segfault.