libusb/libusb #1618: libusb: debug [parse_configuration] skipping descriptor 0xb

AmosZJH, 2025-03-17:
libusb: debug [parse_configuration] skipping descriptor 0xb

[image]

when i use realsense camera d435， it occurs this bug

tormodvolden, 2025-03-17:
There is likely not a bug but something strange about the camera's descriptors, and libusb warns about it. Can you please provide a dump of the descriptors? If you are on Linux, you can find the bus/device numbers with `lsusb`, then, say they are 3 and 7, look it up with e.g. `udevadm info -q path  -n /dev/bus/usb/003/007` then prepend /sys and append /descriptors to what you got. Either upload or show a hexdump of the descriptors file, e.g.:
`hd  /sys/devices/pci0000:00/0000:00:14.0/usb3/3-7/descriptors`

mcuee, 2025-03-23:
This is quite common debug log and not a bug. I will close this first.

closed 2025-03-23 by mcuee