libusb/libusb #1619: infinite recursion in libusb_init_context() if LIBUSB_DEBUG=5 or higher (regression)

tormodvolden, 2025-03-17:
```
$ LIBUSB_DEBUG=6 lsusb
Segmentation fault (core dumped)
```

Probably introduced in commit 9d595d4 because reverting it partially (the libusb/core.c hunk) restores expected behavior.

@seanm Can you take a look?

tormodvolden, 2025-03-17:
```
#18064 0x00007ffff7fa667d in usbi_log (ctx=ctx@entry=0x0, level=level@entry=LIBUSB_LOG_LEVEL_WARNING, 
    function=function@entry=0x7ffff7fb7910 <__func__.2> "get_env_debug_level", 
    format=format@entry=0x7ffff7fb55e8 "LIBUSB_DEBUG is invalid or out of range; clamping") at core.c:2858
#18065 0x00007ffff7fa78ab in get_env_debug_level () at core.c:2392
#18066 0x00007ffff7fa7b47 in log_v (ctx=0x0, level=LIBUSB_LOG_LEVEL_WARNING, 
    function=0x7ffff7fb7910 <__func__.2> "get_env_debug_level", 
    format=0x7ffff7fb55e8 "LIBUSB_DEBUG is invalid or out of range; clamping", args=args@entry=0x7fffffffdcb8)
    at core.c:2775
#18067 0x00007ffff7fa667d in usbi_log (ctx=ctx@entry=0x0, level=level@entry=LIBUSB_LOG_LEVEL_WARNING, 
    function=function@entry=0x7ffff7fb7910 <__func__.2> "get_env_debug_level", 
    format=format@entry=0x7ffff7fb55e8 "LIBUSB_DEBUG is invalid or out of range; clamping") at core.c:2858
#18068 0x00007ffff7fa78ab in get_env_debug_level () at core.c:2392
#18069 0x00007ffff7fa83f6 in libusb_init_context (ctx=0x7fffffffde50, options=0x0, num_options=0) at core.c:2461
#18070 0x0000555555565796 in ?? ()
#18071 0x00007ffff7d6fca8 in __libc_start_call_main (main=main@entry=0x5555555654e0, argc=argc@entry=1, 
    argv=argv@entry=0x7fffffffdfa8) at ../sysdeps/nptl/libc_start_call_main.h:58
```

seanm, 2025-03-17:
I repro... looking now...

seanm, 2025-03-17:
Oh, I see. `get_env_debug_level` tries to log if the debug level is out of range, but the function it tries to log with can call `get_env_debug_level` too...

One solution would be for `get_env_debug_level` to just not log... It didn't used it, it just silently clamped. We can go back to that.

tormodvolden, 2025-03-18:
Yes, we cannot use any log functions inside get_env_debug_level(). If we want to warn about this, it has to be through some warn_once functionality or be propagated to the caller first. I'd say for a debug variable it is fine to just silently ignore invalid values in order to reduce code complexity.

commit 14963242 (2025-03-18): core: Avoid infinite recursion on invalid LIBUSB_DEBUG value

Silently ignore invalid values, since we cannot use logging inside
the get_env_debug_level() function.

Fixes #1619
closed 2025-03-18 by tormodvolden