libusb/libusb #1623: Misc: To release 1.0.29

tormodvolden, 2025-03-19:
Pending review, may be postponed if not enough reviews:

(Windows)
- #1512 (new API)
- #1181
- #749
- #1227
- #1018 (libusb0.sys)
- #1127

(General)
- #1067
- #890

Neustradamus, 2025-03-19:
To follow this ticket :)

mcuee, 2025-03-20:
* https://github.com/libusb/libusb/pull/1406

Looks Windows hotplug may not make it to libusb 1.0.29 either.

mcuee, 2025-03-20:
Some smaller PRs which may be worth looking at for 1.0.29.
* https://github.com/libusb/libusb/pull/1269

seanm, 2025-03-20:
Also please:

- https://github.com/libusb/libusb/pull/1542
- https://github.com/libusb/libusb/pull/1540
- https://github.com/libusb/libusb/pull/1461

mcuee, 2025-04-29:
> * [windows: hotplug implementation #1406](https://github.com/libusb/libusb/pull/1406)
> 
> Looks Windows hotplug may not make it to libusb 1.0.29 either.

Let's see if PR #1633 will fair better than PR #1406.

For users who care about Windows hotplug, please give it a try. Thanks.

mcuee, 2025-05-13:
Let's welcome @sonatique to the admin team.

@sonatique is one of the few persons who know about libusb Windows. I believe he will be a very good addition to the libusb team.

tormodvolden, 2025-05-15:
Since I forgot to bump the LIBUSB_API_VERSION in 1.0.28 #1642 , and we also have a regression fix for macOS #1637 we may need to make a quick "brown paper bag"-release.

mcuee, 2025-05-21:
> Since I forgot to bump the LIBUSB_API_VERSION in 1.0.28 [#1642](https://github.com/libusb/libusb/issues/1642) , and we also have a regression fix for macOS [#1637](https://github.com/libusb/libusb/issues/1637) we may need to make a quick "brown paper bag"-release.

That will be a good idea. Thanks.

sonatique, 2025-05-21:
I second this idea, thank you.

tormodvolden, 2025-05-22:
I plan to make the 1.0.29 release this weekend, with what is in master right now. Since the changes are so few, I don't plan for an official RC. I just hope the macOS change commit 9cef804b has had sufficient testing. Can those with macs please test and confirm?

Neustradamus, 2025-05-22:
@tormodvolden: Do you report some tickets for 1.0.30 version?

Neustradamus, 2025-05-22:
@tormodvolden: Do you report some tickets for 1.0.30 version?

Note: Can you add you in public here:
- https://github.com/orgs/libusb/people

mcuee, 2025-05-25:
@Neustradamus 

We welcome contributions to libusb project. 
In order to be added to libusb organization, we need the user to have proven track record in the community. You can start with reviewing codes and fixing known issues. Thanks.

mcuee, 2025-05-25:
@tormodvolden 

Unfortunately I cannot reproduce the issues of #1637 using the xusb flash drive test mentioned in #1610.

But interestingly I got good results with libusb git and bad results with libusb 1.0.28 release. I am not so sure if it is due to[ commit https://github.com/libusb/libusb/commit/9cef804b2454a2226f5fa5db79a7e9aa8a45d4d4 which fixes Issue #1637.

My test results mentioned in #1635.
* https://github.com/libusb/libusb/issues/1635

Neustradamus, 2025-05-25:
@mcuee: I do not understand your answer, @tormodvolden is not in the team?
He can not change private to public here?
- https://github.com/orgs/libusb/people

tormodvolden, 2025-05-25:
Please do not use this release ticket for all kind of requests and questions, use "Discussions" if needed.

To remedy the API version hickup a bit, I am thinking it would help to add the missing "since API version" documentation of the affected functions, see #1644. Maybe good to have this in the released source, so I'll await the release until this is in.

mcuee, 2025-05-25:
@Neustradamus 
I have deleted the comments not related to this issue. In any case, Tormod is in the team and he is the main maintainer currently.

Neustradamus, 2025-06-01:
@tormodvolden: Thanks!