libusb/libusb #1624: "Modules" link on api.libusb.info broken

tormodvolden, 2025-03-19:
When updating the web site for 1.0.28 I used `make -C docs docs-upload` . It deleted these files:
deleting api-1.0/modules.html
deleting api-1.0/folderopen.png
deleting api-1.0/folderclosed.png
deleting api-1.0/doc.png
deleting api-1.0/dir_6560186179ae86ab3fbed805e7cb86e4.html
deleting api-1.0/bdwn.png

(I may for previous releases have done a manual rsync without --delete, I am not sure)

Now the "Modules" button still appear in the top menu but the target modules.html is obviously not there and returns 404.

mcuee, 2025-03-19:
> Now the "Modules" button still appear in the top menu but the target modules.html is obviously not there and returns 404.

Hmm, I do not see the "Modules" button.
https://libusb.sourceforge.io/api-1.0/index.html

[image]

tormodvolden, 2025-03-19:
Yes, it has been renamed to "Topics". Must be a intermittent caching issue.

closed 2025-03-19 by tormodvolden

tormodvolden, 2025-03-19:
Depending on which doxygen version it will be Topics or Modules.