libusb/libusb #1627: libusb crashed if tries to 'remove from the list of in-flight transfers'

kryvosheiaivan, 2025-04-03:
If while doing libusb_close() there are in-flight transfers,
then programming flow goes into for_each_transfer_safe() loop in function do_close().
Then it fail in attempt to lock mutex:
`usbi_mutex_lock(&itransfer->lock);`
Every time when programming flows are there because of  in-flight transfers - bug is reproduced.

In real life, I founded bug when I use CYW920829M2EVK-02 connected via USB-HUB and along with openOCD.
Since, bug is reproduced only when there is in-flight transfers, it needs 10-30 launches.

`#!/bin/bash
OPENOCD=C:/openocd

cd "C:/openocd"

LOG_FILE="super_log.txt"
SEARCH_TERM="Segmentation"

echo "Monitoring $LOG_FILE for '$SEARCH_TERM'..."
n=0

while true; do
    echo "Launnch $n..."
    ./bin/openocd -s scripts -f interface/cmsis-dap.cfg -c "set ENABLE_ACQUIRE 0" -c "set SMIF_BANKS { 0 {addr 0x60000000 size 0x800000 psize 0x100 esize 0x40000} }" -f target/infineon/cyw20829.cfg -c "init; reset init;  shutdown" &> super_log.txt
    ((n++))
    # Check if the term exists in the log file
    if grep -q "$SEARCH_TERM" "$LOG_FILE"; then
        echo "Found '$SEARCH_TERM' in $LOG_FILE!"
        
        # Optional: Get and display the matching line(s)
        matches=$(grep "$SEARCH_TERM" "$LOG_FILE")
        echo "Matching line(s):"
        echo "$matches"
        
        break  # Exit the loop
    fi
    
    # Wait before checking again (adjust sleep time as needed)
    sleep 2
done

echo "Monitoring complete."`

[image]

openOCD is taken here:
https://github.com/Infineon/openocd/releases/tag/release-v5.7.0

Bug is not reproduced when logs on libusb set ON so I replaced dll in openOCD
and put there dll from local build of libusb. I put logs in do_close() and founded
reason of crash.

Can anybody explain why dedicated code does not work and crashes instead?

mcuee, 2025-04-04:
Can you reproduce the issue of official OpenOCD and not the fork from Infineon?

Basically I tend to think this is an application bug and not libusb issue. You should not close the device when there are still  in-flight transfers.
https://github.com/libusb/libusb/wiki/FAQ#what-are-the-differences-between-synchronous-and-asynchronous-device-io

mcuee, 2025-06-27:
Looks like this is an OpenOCD bug which has been fixed.
* https://github.com/openocd-org/openocd/commit/8046f2a38f8af0048c0d2d0842b8a708b79c0c57

closed 2025-06-27 by mcuee