libusb/libusb #1628: `make install` failure on Windows using Git Bash

recursivenomad, 2025-04-04:
My first installation of libusb on Windows happened to be 1.0.26, which conveniently offered the choice between `libusb-1.0.26.7z` and the pre-built `libusb-1.0.26-binaries.7z`.  I was able to extract the `-binaries.7z`, and it worked as expected.  Unfortunately, that choice is no longer the case.

Reading the directions [here](https://github.com/libusb/libusb/issues/1468#issuecomment-1972878898), I have tried running `./configure && make install` on the 1.0.28 release directory, but it fails.  Since I have *Git for Windows* installed, I use *Git Bash* as my shell on Windows.  Running `make install` from *Git Bash* returns `/usr/bin/sh: line 2: C:/Program: No such file or directory`.  I am assuming this is due to spaces in the default install path of *Git Bash* (`C:/Program Files/Git/git-bash.exe`), which is also where it installs sh (`C:/Program Files/Git/usr/bin/sh`).

This is where my investigation ends so far, and I'm inquiring here before digging further.  Is there any way the configuration can allow for spaces in file paths, or is there a way to obtain a current release of the `-binaries.7z`, like those found in 1.0.26?

Thank you for maintaining this project!

mcuee, 2025-04-05:
Please do not use git bash but rather MSYS2 mingw64.
https://www.msys2.org/

mcuee, 2025-04-05:
libusb-1.0.28.7z is here.
https://github.com/libusb/libusb/releases/tag/v1.0.28

closed 2025-04-05 by mcuee

recursivenomad, 2025-04-05:
Alright; FWIW adding MSYS2 is inconvenient for my toolchain, as I already have MinGW64 set up otherwise - Is it possible to bring back the precompiled binaries as an option for future Windows releases?

And if it is known that MinGW64 + Git Bash causes issues, it might be worth documenting that in [`INSTALL_WIN.txt`](https://github.com/libusb/libusb/blob/master/INSTALL_WIN.txt), as MSYS2 is not standard for everyone's Windows dev environment.

mcuee, 2025-04-05:
What is the problem you have with the official Windows binary libusb-1.0.28.7z?

mcuee, 2025-04-05:
You can also manual copy the files as well to your MinGW64 installation, if you can not get `make install` to work.

recursivenomad, 2025-04-05:
What is the process for manually copying the files to the MinGW64 installation, when files like the pkgconfig are not distributed in the releases?

The problem with [`libusb-1.0.28.7z`](https://github.com/libusb/libusb/releases/download/v1.0.28/libusb-1.0.28.7z) is that it has a different directory structure to [`libusb-1.0.26-binaries.7z`](https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26-binaries.7z).  The project I am compiling apparently expects the directory structure of `libusb-1.0.26-binaries.7z`.

Inside each respective MinGW32 folder, for example, this is what is different:

| `libusb-1.0.26-binaries.7z` | `lib-usb-1.0.28.7z` |
|--|--|
| - bin/ | - dll/ |
| - - msys-usb-1.0.dll | - - libusb-1.0.dll |
| - include/ | |
| - - libusb-1.0/ | |
| - - - libusb.h | |
| - lib/ | - static/ |
| - - libusb-1.0.a | - - libusb-1.0.a |
| - - libusb-1.0.dll.a | - - libusb-1.0.dll.a |
| - - libusb-1.0.la | |
| - - pkgconfig/ | |
| - - - libusb-1.0.pc | |