libusb/libusb #1632: Configure ends up with error even if libudev is installed.

Mrjagx, 2025-04-15:
Greetings,
I run configure script with the following parameters _CFLAGS="-m32 -mstackrealign -fPIC" --prefix="$PWD" --disable-shared_ and it terminates with :
...
checking for libudev.h... no
configure: error: udev support requested but libudev header not installed
....

PS: if I run it only with ./configure --prefix=/usr --disable-static it runs like a charm.

mcuee, 2025-04-16:
Just wonder what is the compiler and Linux distro used? What is the motivation to use `--disable-shared`? Is this an embedded Linux distro?

Mrjagx, 2025-04-16:
I am using OS: Pop!_OS 22.04 LTS x86_64 Kernel: Linux 6.12.10-76061203-generic. 
there is a windows application that i want to run using wine, and there is wrapper around libusb to access a programmer, this wrapper is a windows dll compiled with winegcc and it requires a specific build for libusb with i386 instruction + some stack alignment to work correctly as far as i understood.
link: https://github.com/radiomanV/TL866/tree/master/wine

mcuee, 2025-04-16:
I will just say this is not a supported use case. But maybe @tormodvolden can chime in to see he has other suggestions.

tormodvolden, 2025-05-01:
Can you compile anything (like `int main (void) {}` ) with -m32 ? Maybe you are missing some 32-bit development support packages? Did you install libudev1:i386 as described in your link? What about libudev-dev:i386?

mcuee, 2025-06-27:
As this does not seem to be a libusb issue, I will convert this into a discussion.