libusb/libusb #1634: smart card readers data not set on windows

gwenchailleu, 2025-04-25:
Hello,
I comme accross this issue using libusb on Windows : some data from smart card readers records are not set.
Please see attached files ( the txt file contains the hole story, the others are screen shots)
Thanks!
Gwen

[issue_reading_smart_card_reader_data.txt](https://github.com/user-attachments/files/19904499/issue_reading_smart_card_reader_data.txt)

[image]
[image]
[image]
[image]

mcuee, 2025-04-26:
You may want to summarize the issue -- what exactly is the issue?

gwenchailleu, 2025-04-29:
Hello,

The first lines of the attached file are :

My goal : retreive id product, id vendor, manufacturer descriptor and product descriptor of all USB SmartCard readers.
My problem : libUSB seems unable to retreive the manufacturer and product descriptors of my device, when USBView is able. Everything is fine on macOS (linux not tested yet).
My device : USB SmartCard Reader Gemalto Gemplus (08E6:3437) on Windows 11 Pro 24H2

I  should have copied them.
I used an attached file because of the ticket body size limit..

Regards
Gwen

mcuee, 2025-05-02:
Are you able to get it work with WinUSB driver after using WinUSB driver to replace the original driver?

gwenchailleu, 2025-05-05:
Hello,
Yes ( see attached file), but our problem is that we cannot ask 400 000 health professionnals to change their usb driver, so we'll have to code our own bypass outside libusb which is a bit disapointing. 
Especially since macOs output is Ok. 
As far as Linux is concerned, we should  know more before the end of the year.
Regards
Gwen

tormodvolden, 2025-05-05:
If the challenge is to retrieve the manufacturer string descriptor and product string descriptor, please try out PR #1532 which will get those from the OS instead of requesting the device directly.

tormodvolden, 2025-05-05:
> If I'm not wrong, shouldn't the WUDFRd driver work when using the WinUSB API ? The Wiki says that it should be able to work with WinUSB, as a subapi in this case.

Can you please point to where that is written, so that we can review it?

(WUDFRd is [UMDF](https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/detailed-view-of-the-umdf-architecture) reflector driver)