libusb/libusb #1635: libusb_attach_kernel_driver on MacOS segfaults

Mr-Bossman, 2025-04-27:
When running `libusb_attach_kernel_driver` on MacOS, it either stalls at exit or
segfaults in `libusb/hotplug.c:usbi_hotplug_exit` at `for_each_device_safe`.
for dereferencing `0xfff...ffe0`, which is `-32`, the offset of `list` in `libusb_device`.

Do note that even if the kernel driver is active or inactive, it still has this behavior. 
I will probably attempt to fix this soon, just reporting the bug.

My system:
```
$ uname -a
Darwin johnnys-MacBook-Pro.local 22.6.0 Darwin Kernel Version 22.6.0: Mon Apr 22 20:54:28 PDT 2024; root:xnu-8796.141.3.705.2~1/RELEASE_X86_64 x86_64
```

Here is a demo program. 
```c
#include <stdio.h>

#include "libusb.h"
 
static void print_devs(libusb_device **devs)
{
        struct libusb_device_handle *devh = NULL;
        libusb_device *dev;
        int rc, i = 0, j = 0;
        uint8_t path[8];

        while ((dev = devs[i++]) != NULL) {
                struct libusb_device_descriptor desc;
                int r = libusb_get_device_descriptor(dev, &desc);
                if (r < 0) {
                        fprintf(stderr, "failed to get device descriptor");
                        return;
                }

                rc = libusb_open(dev, &devh);
                if (rc < 0) {
                        fprintf(stderr, "Error opening USB device: %s\n", libusb_error_name(rc));
                        return;
                }

                rc = libusb_attach_kernel_driver(devh, 0);
                if (rc < 0)
                        fprintf(stderr, "Error attaching kernel driver: %s\n", libusb_error_name(rc));

                libusb_close(devh);

                printf("%04x:%04x (bus %d, device %d)",
                        desc.idVendor, desc.idProduct,
                        libusb_get_bus_number(dev), libusb_get_device_address(dev));

                r = libusb_get_port_numbers(dev, path, sizeof(path));
                if (r > 0) {
                        printf(" path: %d", path[0]);
                        for (j = 1; j < r; j++)
                                printf(".%d", path[j]);
                }
                printf("\n");
        }
}

int main(void)
{
        libusb_device **devs;
        int r;
        ssize_t cnt;

        r = libusb_init_context(/*ctx=*/NULL, /*options=*/NULL, /*num_options=*/0);
        if (r < 0)
                return r;

        cnt = libusb_get_device_list(NULL, &devs);
        if (cnt < 0){
                libusb_exit(NULL);
                return (int) cnt;
        }

        print_devs(devs);
        libusb_free_device_list(devs, 1);

        libusb_exit(NULL);
        return 0;
}
```

tormodvolden, 2025-05-15:
Is this is a new bug or is it seen with older versions of libusb too?

mcuee, 2025-05-15:
To be honest, I actually do not know that we really supported `libusb_attach_kernel_driver`, but I just checked PR #911 and indeed it has this function.
* https://github.com/libusb/libusb/pull/911/

And in practice, this is very difficult to be used for most of the application due to the limitations from Apple.
* https://github.com/libusb/libusb/issues/1014

mcuee, 2025-05-15:
Looking at the example, it is a bit strange though.

If there is no kernel driver for the device, then I am not so sure how libusb_attach_kernel_driver will do.

If there is already a kernel driver for th device, then you have to detach the driver first.

Anyway, I guess the current behavior is not good if confirmed (I will try to reproduce the issue over the weekend).

Mr-Bossman, 2025-05-16:
I can check how long this behavior has been in libusb. Thought it might have just changed due to macOS changing. From what I can tell the detach function does work on macOS, and so does attaching. The problem is in usbi_hotplug_exit from what looks like a messed-up list.

mcuee, 2025-05-25:
Seems to have no issue on my system (Mac Mini M1 2020 model, macOS 15.4.1)

Using libusb-1.0.28 release.

```
mcuee@Macmini libusb/libusb_test  > uname -a
Darwin Macmini 24.4.0 Darwin Kernel Version 24.4.0: Fri Apr 11 18:32:43 PDT 2025; root:xnu-11417.101.15~117/RELEASE_ARM64_T8103 arm64

mcuee@Macmini libusb/libusb_test  > sw_vers 
ProductName:		macOS
ProductVersion:		15.4.1
BuildVersion:		24E263
mcuee@Macmini libusb/libusb_test  > cat test_issue1635.c
#include <stdio.h>

#include "libusb.h"
 
static void print_devs(libusb_device **devs)
{
        struct libusb_device_handle *devh = NULL;
        libusb_device *dev;
        int rc, i = 0, j = 0;
        uint8_t path[8];

        while ((dev = devs[i++]) != NULL) {
                struct libusb_device_descriptor desc;
                int r = libusb_get_device_descriptor(dev, &desc);
                if (r < 0) {
                        fprintf(stderr, "failed to get device descriptor");
                        return;
                }

                rc = libusb_open(dev, &devh);
                if (rc < 0) {
                        fprintf(stderr, "Error opening USB device: %s\n", libusb_error_name(rc));
                        return;
                }

                rc = libusb_attach_kernel_driver(devh, 0);
                if (rc < 0)
                        fprintf(stderr, "Error attaching kernel driver: %s\n", libusb_error_name(rc));

                libusb_close(devh);

                printf("%04x:%04x (bus %d, device %d)",
                        desc.idVendor, desc.idProduct,
                        libusb_get_bus_number(dev), libusb_get_device_address(dev));

                r = libusb_get_port_numbers(dev, path, sizeof(path));
                if (r > 0) {
                        printf(" path: %d", path[0]);
                        for (j = 1; j < r; j++)
                                printf(".%d", path[j]);
                }
                printf("\n");
        }
}

int main(void)
{
        libusb_device **devs;
        int r;
        ssize_t cnt;

        r = libusb_init_context(/*ctx=*/NULL, /*options=*/NULL, /*num_options=*/0);
        if (r < 0)
                return r;

        cnt = libusb_get_device_list(NULL, &devs);
        if (cnt < 0){
                libusb_exit(NULL);
                return (int) cnt;
        }

        print_devs(devs);
        libusb_free_device_list(devs, 1);

        libusb_exit(NULL);
        return 0;
}

mcuee@Macmini libusb/libusb_test  > gcc -I /opt/homebrew/include/libusb-1.0  test_issue1635.c -o test_issue1635 -L /opt/homebrew/lib -lusb-1.0

mcuee@Macmini libusb/libusb_test  > sudo ./test_issue1635
05e3:0749 (bus 2, device 22) path: 6.1.2
1915:1025 (bus 2, device 21) path: 2.4.7
046d:c52b (bus 2, device 20) path: 2.4.3
0781:5581 (bus 2, device 19) path: 2.4.5.1
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
1a40:0201 (bus 2, device 14) path: 2.4.5
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
1a40:0201 (bus 2, device 13) path: 2.4
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:5411 (bus 2, device 12) path: 2.1
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:5411 (bus 2, device 11) path: 2
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:0411 (bus 2, device 2) path: 6.1
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:0411 (bus 2, device 1) path: 6
```

mcuee, 2025-05-25:
Hmm, I can reproduce the issue if run a few times. I will capture the debug log tomorrow.

```
mcuee@Macmini libusb/libusb_test  > sudo ./test_issue1635
Error attaching kernel driver: LIBUSB_ERROR_TIMEOUT
05e3:0749 (bus 2, device 5) path: 6.1.2
Error attaching kernel driver: LIBUSB_ERROR_TIMEOUT
1915:1025 (bus 2, device 8) path: 2.4.7
Error attaching kernel driver: LIBUSB_ERROR_TIMEOUT
046d:c52b (bus 2, device 10) path: 2.4.3
Error attaching kernel driver: LIBUSB_ERROR_TIMEOUT
0781:5581 (bus 2, device 4) path: 2.4.5.1
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
1a40:0201 (bus 2, device 14) path: 2.4.5
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
1a40:0201 (bus 2, device 13) path: 2.4
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:5411 (bus 2, device 12) path: 2.1
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:5411 (bus 2, device 11) path: 2
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:0411 (bus 2, device 2) path: 6.1
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:0411 (bus 2, device 1) path: 6
^C
```

mcuee, 2025-05-25:
@Mr-Bossman 

Please test with latest git head to see if the issue is still present on your system or not. Thanks.

Mr-Bossman, 2025-05-25:
> [@Mr-Bossman](https://github.com/Mr-Bossman)
> 
> Please test with latest git head to see if the issue is still present on your system or not. Thanks.

Yes my head was at libusb-1.0.28-rc1, commit https://github.com/libusb/libusb/commit/9cef804b2454a2226f5fa5db79a7e9aa8a45d4d4 does fix it.

closed 2025-05-25 by Mr-Bossman

mcuee, 2025-05-25:
> > [@Mr-Bossman](https://github.com/Mr-Bossman)
> > Please test with latest git head to see if the issue is still present on your system or not. Thanks.
> 
> Yes my head was at libusb-1.0.28-rc1, commit [9cef804](https://github.com/libusb/libusb/commit/9cef804b2454a2226f5fa5db79a7e9aa8a45d4d4) does fix it.

@Mr-Bossman 
Thanks a lot for the confirmation.

@tormodvolden 
FYI as well.

tormodvolden, 2025-05-26:
Maybe you can try to dump the device list before (and inside) for_each_device_safe to see if there is something strange to it?

@mcuee can you also reproduce the segfault with this timeout value?

Mr-Bossman, 2025-05-26:
> Maybe you can try to dump the device list before (and inside) for_each_device_safe to see if there is something strange to it?

Not quite sure what you mean, are you talking about in usbi_hotplug_exit? Also, do you have a Mac?

> 
> [@mcuee](https://github.com/mcuee) can you also reproduce the segfault with this timeout value?

Also here are the debug logs if anyone wanted.

```
(gdb) r
Starting program: /Users/jesse/libusb_stuff/libusb/listdevs 
[New Thread 0x1903 of process 15400]
[New Thread 0x1503 of process 15400]
[Thread 0x1903 of process 15400 exited]
warning: unhandled dyld version (17)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000032] [009bee83] libusb: debug [libusb_init_context] created default context
[ 0.000060] [009bee83] libusb: debug [libusb_init_context] libusb v1.0.28.11951
[ 0.000078] [009bee83] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000225] [009bee85] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000518] [009bee85] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000685] [009bee83] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.000693] [009bee83] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.000715] [009bee83] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x0
[ 0.003708] [009bee83] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003720] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003723] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0100
[ 0.003725] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003726] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0xff
[ 0.003728] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.003730] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.003732] [009bee83] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.003733] [009bee83] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8007
[ 0.003735] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0198
[ 0.003737] [009bee83] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.003812] [009bee83] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.003821] [009bee83] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003824] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003936] [009bee83] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.003943] [009bee83] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.003962] [009bee83] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x14000000 against cached device with sessionID/locationID 0x0/0x1000000
[ 0.003965] [009bee83] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.004015] [009bee83] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1b15e88a8
[ 0.004049] [009bee83] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x1b15e88a8/0x14100000 against cached device with sessionID/locationID 0x0/0x1000000
[ 0.004053] [009bee83] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1b15e88a8
[ 0.005475] [009bee83] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.005771] [009bee83] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.006077] [009bee83] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.006332] [009bee83] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.006557] [009bee83] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.006592] [009bee83] libusb: debug [darwin_device_from_service] could not set up plugin for service: out of resources
[ 0.006680] [009bee83] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x515572f502
[ 0.006731] [009bee83] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x515572f502/0x1100000 against cached device with sessionID/locationID 0x0/0x1000000
[ 0.006739] [009bee83] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x515572f502
[ 0.007235] [009bee83] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.007242] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.007245] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.007247] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.007248] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.007250] [009bee83] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[... cut ...]
[ 0.289217] [009bee83] libusb: debug [libusb_unref_device] destroy device 1.59
[New Thread 0x1807 of process 15400]

Thread 2 received signal SIGSEGV, Segmentation fault.
0x000000010000a788 in usbi_hotplug_exit (ctx=0x100104470) at hotplug.c:219
219             for_each_device_safe(ctx, dev, next_dev) {
(gdb) bt
#0  0x000000010000a788 in usbi_hotplug_exit (ctx=0x100104470) at hotplug.c:219
#1  0x00000001000067dc in libusb_exit (ctx=0x0) at core.c:2609
#2  0x0000000100003046 in main () at examples/listdevs.c:65
(gdb) 
```

mcuee, 2025-05-26:
> [@mcuee](https://github.com/mcuee) can you also reproduce the segfault with this timeout value?

Not yet with built-in xusb example.

```
mcuee@Macmini libusb/libusb  - (master)> git diff
diff --git a/libusb/os/darwin_usb.c b/libusb/os/darwin_usb.c
index a06e1d72..17db02d0 100644
--- a/libusb/os/darwin_usb.c
+++ b/libusb/os/darwin_usb.c
@@ -40,6 +40,7 @@
 
 /* Default timeout to 10s for reenumerate. This is needed because USBDeviceReEnumerate
  * does not return error status on macOS. */
+//#define DARWIN_REENUMERATE_TIMEOUT_US (10ULL * USEC_PER_SEC)
 #define DARWIN_REENUMERATE_TIMEOUT_US (10ULL * USEC_PER_SEC)
 
 #include <AvailabilityMacros.h>

mcuee@Macmini libusb/libusb  - (master)> sudo ./examples/xusb 0781:5581
Using libusb v1.0.28.11951

Opening device 0781:5581...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0781:5581
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: 2 caps
    USB 2.0 extension:
      attributes             : 02
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x01): " USB"
   String (0x02): " SanDisk 3.2Gen1"
   String (0x03): "0401c03d40d3c02d1cdaba0d0877b9be0b3419bffa8485b645f273d486397af05c3600000000000000000000e45ccdfdff926c18815581075dae9a76"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Reading Max LUN:
   Max LUN = 0

Sending Inquiry:
   sent 6 CDB bytes
   received 36 bytes
   VID:PID:REV "     USB":" SanDisk":"1.00"
   Mass Storage Status: 00 (Success)

Reading Capacity:
   sent 10 CDB bytes
   received 8 bytes
   Max LBA: 03956FFF, Block Size: 00000200 (28.67 GB)
   Mass Storage Status: 00 (Success)

Attempting to read 512 bytes:
   sent 10 CDB bytes
   READ: received 512 bytes
   Mass Storage Status: 00 (Success)

  00000000  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000010  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000020  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000030  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000040  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000050  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000060  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000070  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000080  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000090  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  000000a0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  000000b0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  000000c0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  000000d0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  000000e0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  000000f0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000100  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000110  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000120  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000130  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000140  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000150  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000160  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000170  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000180  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000190  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  000001a0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  000001b0  00 00 00 00 00 00 00 00 f9 24 0c 47 00 00 80 00  .........$.G....
  000001c0  01 01 0c 3f a0 00 00 08 00 00 00 00 10 00 00 00  ...?............
  000001d0  81 01 83 3f e0 ff 00 08 10 00 df 67 85 03 00 00  ...?.......g....
  000001e0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  000001f0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 aa  ..............U.

Releasing interface 0...
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
libusb: info [darwin_capture_release_interface] on attempt to reattach the kernel driver got ret=-5
Closing device...
mcuee@Macmini libusb/libusb  - (master)> 
```

mcuee, 2025-05-27:
I cannot reproduce the issue with the test prorgam either -- no crash after many runs. Maybe the timeout value is too long for my devices.

```
mcuee@Macmini libusb/libusb_test  > gcc -I /Users/mcuee/bin/include/libusb-1.0 test_issue1635.c -o test_issue1635_mod -L /Users/mcuee/bin/lib -lusb-1.0

mcuee@Macmini libusb/libusb_test  > sudo su

sh-3.2# whoami
root
sh-3.2# DYLD_LIBRARY_PATH=/Users/mcuee/bin/lib ./test_issue1635_mod
0781:5581 (bus 2, device 13) path: 2.4.5.1
046d:c52b (bus 2, device 9) path: 2.4.3
1915:1025 (bus 2, device 30) path: 2.4.7
05e3:0749 (bus 2, device 31) path: 6.1.2
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
1a40:0201 (bus 2, device 7) path: 2.4.5
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
1a40:0201 (bus 2, device 5) path: 2.4
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:5411 (bus 2, device 4) path: 2.1
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:5411 (bus 2, device 3) path: 2
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:0411 (bus 2, device 2) path: 6.1
Error attaching kernel driver: LIBUSB_ERROR_NO_DEVICE
0bda:0411 (bus 2, device 1) path: 6
```

Mr-Bossman, 2025-05-27:
I did something like this

```
    long delta_sec = now.tv_sec - start.tv_sec;
    long delta_nsec = now.tv_nsec - start.tv_nsec;
    unsigned long long elapsed_us = (unsigned long long)delta_sec * USEC_PER_SEC +
                                    (unsigned long long)delta_nsec / 1000ULL;

    TIMESPEC_SUB(&now, &start, &delta);
    unsigned long long elapsed_us_c = (unsigned long long)delta.tv_sec * USEC_PER_SEC +
                                    (unsigned long long)delta.tv_nsec / 1000ULL;

    if (elapsed_us >= DARWIN_REENUMERATE_TIMEOUT_US) {
      printf("darwin/reenumerate_device: timeout waiting for reenumerate %llu", elapsed_us_c);
      usbi_err (ctx, "darwin/reenumerate_device: timeout waiting for reenumerate");
    }
```

Sorry im on my phone otherwise I would share a commit.

mcuee, 2025-05-27:
> Sorry im on my phone otherwise I would share a commit. Have you been able to recreate the issue on 1.0.28 so without [9cef804](https://github.com/libusb/libusb/commit/9cef804b2454a2226f5fa5db79a7e9aa8a45d4d4) ? You need to run it a few times

Please share your commit when you have access to your Mac. Thanks.

I will carry out more tests this evening when I have access to my Mac at home (GMT+8 here in Singapore).

tormodvolden, 2025-05-27:
> Interestingly last time we had a similar crashing issue as well.
> 
>     * [Segmentation fault with the kernel detaching function under macOS 14.2 using sudo #1386](https://github.com/libusb/libusb/issues/1386)
> 
> 
> Fixed in commit [9401e6c](https://github.com/libusb/libusb/commit/9401e6c7d9b45234f14075ccfcaaeccc48fadbe4).

Yes, that is probably the same underlying issue, in the same context of detach/reattach kernel drivers. It was acknowledged in the bug fix that there still could be something missing, but at least it wouldn't trigger so often.

mcuee, 2025-05-27:
@hjelmn and @osy 

Just wondering if you can take a look. Thanks.

sbfkcel, 2025-09-16:
I also found some anomalies under macos26, but I am not sure if it is caused by the same problem.