libusb/libusb #1637: [macOS] Changes to remove compiler warnings breaks macOS re-enumeration timeout.

pjpetree, 2025-05-01:
libusb-1.0.28 introduced some changes to fix some compiler warnings. With these changes, part of the calculation of the elapsed µs time (the nsec portion) was cast as an `unsigned long long`. This cast incorrectly promotes small negative values to huge positive values cause the enumeration timeout to expire too soon. This, ultimately, causes some slower devices to fail to enumeration consistently.

For example, a delta between 1.9 s and 2.1 s means the `delta_sec` value will be 1 s and the `delta_nsec` value should be -0.8 s. Adding these two values would (correctly) make the delta 0.2 s. 

The code, unfortunately, casts the `delta_nsec` to an unsigned value which makes the value very large. This means adding the two values creates an invalidly large elapsed time, which is larger than the timeout, and the device is not re-enumerated.

seanm, 2025-05-01:
I'll take a look...

tormodvolden, 2025-05-01:
Thanks for reporting and the analysis. Can we say something about how likely this bug will kick in? Will most devices be so fast that it won't go wrong? Then some safe and unsafe intervals for the slower ones.

pjpetree, 2025-05-01:
Seems like it would be proportional to how long the device takes to re-enumerate.

E.g., if the device takes 250 ms to re-enumerate, the issue will only happen if the starting `nsec` value is ≥ 750 ms.  Which, if I'm not mistaken, would mean a probability to hit the issue of 0.25.  A device that takes 650 ms would have a probability of 0.65.

closed 2025-05-03 by tormodvolden

tormodvolden, 2025-05-03:
Just wondering if this warrants an accelerated new release. The issue was sitting in git master for 10 months before getting detected and reported (in PR #1610)  though.

sonatique, 2025-05-03:
I vote yes for the accelerated release.

commit 9cef804b (2025-05-03): darwin: Fix regression causing re-enumeration timeout

Fix time calculation for negative nanosecond deltas to resolve
re-enumerate timeout. Fixup of commit a07ecfe02.

Fixes #1637
References #1610