libusb/libusb #1642: LIBUSB_API_VERSION not updated in v1.0.28 despite new SuperSpeed+ functions

FilippoGuggino, 2025-05-12:
In libusb v1.0.28, several new functions were introduced to support SuperSpeed+ devices (#1428). However, it appears that the LIBUSB_API_VERSION macro has not been updated to reflect these additions.

Could you clarify whether this is an oversight or intentional?
If intentional, is there an alternative macro that reliably indicates the version of the installed libusb at compile time, including the availability of the new API functions?

Thank you in advance!

tormodvolden, 2025-05-15:
It was a terrible oversight! I have updated the release checklist to make sure this is not forgotten again.

Since we also have a regression fix for macOS, we might make a 1.0.29 soon anyway.

tormodvolden, 2025-05-15:
If your application build uses autoconf you can use AC_CHECK_FUNCS to probe for functions, for other build tools there may be similar ways.

commit 94e0d8c2 (2025-05-15): Bump LIBUSB_API_VERSION

1.0.28 shipped with new SuperSpeed+ functions, but the
LIBUSB_API_VERSION macro had unfortunately not been changed.

Fixes #1642
closed 2025-05-15 by tormodvolden