libusb/libusb #1647: Misc: To Release 1.0.30

mcuee, 2025-06-02:
libusb-1.0.29 has been released.
* https://github.com/libusb/libusb/issues/1623

Pending review, may be postponed if not enough reviews:

(Windows)

Let's see if Windows hotplug can make it into libusb 1.0.30 release.
https://github.com/libusb/libusb/pull/1406

https://github.com/libusb/libusb/pull/1512 (new API)
https://github.com/libusb/libusb/pull/1181
https://github.com/libusb/libusb/pull/749
https://github.com/libusb/libusb/pull/1227
https://github.com/libusb/libusb/pull/1018 (libusb0.sys)
https://github.com/libusb/libusb/pull/1127
https://github.com/libusb/libusb/pull/1269
https://github.com/libusb/libusb/pull/1353
https://github.com/libusb/libusb/pull/1323

(macOS)
https://github.com/libusb/libusb/pull/1610
https://github.com/libusb/libusb/issues/1635

(General)
https://github.com/libusb/libusb/pull/1558
https://github.com/libusb/libusb/pull/1550
https://github.com/libusb/libusb/pull/1067
https://github.com/libusb/libusb/issues/890
https://github.com/libusb/libusb/pull/1542
https://github.com/libusb/libusb/pull/1540
https://github.com/libusb/libusb/pull/1461
...

mcuee, 2025-08-26:
Now we have a new admin @sonatique who is actively working on the project. Hopefully we can get some of the PRs merged and rtelease 1.0.30.

mcuee, 2025-08-26:
> Thoughts on adding [@seanm](https://github.com/seanm) as an admin too? Continuing discussion here rather than a random pr. [#1508](https://github.com/libusb/libusb/pull/1508)

@seanm 

I just sent you the invitation to join libusb project as an admin. Thanks a lot!

sonatique, 2025-08-28:
> Looks like [#1512](https://github.com/libusb/libusb/pull/1512) will be merged soon.

It's done.
 
> I am wondering if we should target the 1.0.30 release in about two months time, say by the end of October.

I think it would be nice indeed. Does this imply a master freeze starting from now?

I think we should include https://github.com/libusb/libusb/pull/1532 as well, we're close to completion.

seanm, 2025-08-28:
> Does this imply a master freeze starting from now?

I hope not. We currently have some momentum going, and there are several PRs I'd like to see merged...

mcuee, 2025-09-07:
@sonatique @tormodvolden and @seanm 

The other long standing PR is for Android. Not so sure if we can do anything about it.
* https://github.com/libusb/libusb/pull/1164

It was the conrinuation of PR #874 
* https://github.com/libusb/libusb/pull/874

Again libusb is quite broken under Android (non rooted device). If we can get #1164 merged (after the rebase and more testing), it is at least not that broken.
https://github.com/libusb/libusb/issues?q=is%3Aissue%20state%3Aopen%20label%3Aandroid

mcuee, 2025-09-18:
New macOS 26 related issue:

* #1661, to be fixed by PR #1700 
* #1701

mcuee, 2025-09-18:
@sonatique, @tormodvolden and @seanm 

My suggestion is that we still release based on git, if we encounter issues for libusb-1.0.30 release, we can always release 1.0.31 as a bug fixe release.

mcuee, 2025-09-19:
> Would be nice to get these 3 in:
> 
> [#1696](https://github.com/libusb/libusb/pull/1696) [#1686](https://github.com/libusb/libusb/pull/1686) [#1694](https://github.com/libusb/libusb/pull/1694)

I believe #1696 and #1694 should be safe to merge. 

@sonatique Please review the above three issues and merge them if you are fine with the changes. 

After the above three, probably you can tag the 1.0.30-rc1 release.

mcuee, 2025-09-19:
> [@seanm](https://github.com/seanm) [@mcuee](https://github.com/mcuee) I merged [#1696](https://github.com/libusb/libusb/pull/1696) and [#1694](https://github.com/libusb/libusb/pull/1694) . I am a bit reluctant with [#1686](https://github.com/libusb/libusb/pull/1686) because I am the only approver and [@tormodvolden](https://github.com/tormodvolden) has an unresolved discussion.
> 
> I never did a release with libusb, are there some explanations somewhere? Thanks.

@sonatique 

I agree it is a good idea for @tormodvolden to comment before merging #1686.

As for RC/Release work, please refer to here.
https://github.com/libusb/libusb/wiki/Maintainers'-corner#rcrelease-work

If you have some doubts, feel free to ask and @tormodvolden should be able to help as he has done a few releases.

mvp, 2026-01-20:
@seanm, I totally understand. Just was wondering if anything is happening for the release, and perhaps how I can help.

mcuee, 2026-01-21:
> [@seanm](https://github.com/seanm), I totally understand. Just was wondering if anything is happening for the release, and perhaps how I can help.

One potential way is to join libusb project as we are short of resources.

seanm, 2026-01-21:
> Again, I can do the release process 

Please do so then.

>We need as much as possible of testing of real applications on real hardware. This is something that anyone can help out with.

People are constantly testing master.

>Preferably we make a freeze so that I can make a Release Candidate for testing.

We've been frozen for weeks. Only tiny things have been merged. Only 3 commits since September.

>Are there any issues that must be fixed before release (that would be for instance a notable regression from previous release, or something wrong in a new API)?

Not that I know of.

mcuee, 2026-02-28:
> Just was wondering if anything is happening for the release, and perhaps how I can help.

@mvp 

Just wondering if you can carry out some testing on libusb-1.0.31rc1 if you have some time. Thanks.

mcuee, 2026-03-15:
@tormodvolden, @sonatique and @seanm 

My take on the pending PRs for 1.0.30 release.

* https://github.com/libusb/libusb/pull/1680 (good to have for macOS)
* https://github.com/libusb/libusb/pull/1695 (good to have for macOS), @hjelmn has reviwed it.
* https://github.com/libusb/libusb/pull/1774 (delay to 1.0.31)
* https://github.com/libusb/libusb/pull/1775 (good to have for WebAssembly)
* https://github.com/libusb/libusb/pull/1776 (delay to 1.0.31)

But I am okay with any decisions -- you three can decide.

tormodvolden, 2026-05-03:
The RC2 has been released and please use #1824 to report any relevant issues.