libusb/libusb #1648: libusb github action build failed

mcuee, 2025-06-04:
@tormodvolden 

Somehow github action for Linux failed because of the following step (test failed)
```
- name: emscripten
        run: emconfigure .private/ci-build.sh --build-dir build-emscripten -- --host=wasm32-unknown-emscripten
```

Reference:
https://github.com/libusb/libusb/actions/runs/15379068007/job/43267492190
```
FAIL: stress_mt
===============

Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
emconfigure: error: '.private/ci-build.sh --build-dir build-emscripten -- --host=wasm32-unknown-emscripten' failed (returned 1)
Thread 0 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 1 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 2 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 3 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 4 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 5 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 6 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 7 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
All threads discovered 0 devices (0 not opened)
All done, 8 errors
FAIL stress_mt (exit status: 1)
```

mcuee, 2025-06-06:
I do not think the failure is due to libusb changes, probably due to github side. 

Maybe we should disable the test for WebAssembly.

mcuee, 2025-06-08:
Trying to reproduce the issue under Ubuntu 24.04, but I am not so sure how to run `make check`.

Preparation of emsdk and buid of libusb using emsdk

Reference:
https://web.dev/articles/porting-libusb-to-webusb

```
cd ~/build
cd libusb
git clone https://github.com/emscripten-core/emsdk.git
cd emsdk/
./emsdk install latest
./emsdk activate latest
source "/home/mcuee/build/libusb/emsdk/emsdk_env.sh"
echo 'source "/home/mcuee/build/libusb/emsdk/emsdk_env.sh"' >> $HOME/.bash_profile
emcc --version
cd ..
git clone https://github.com/libusb/libusb.git libusb_webusb
cd libusb_webusb/
emconfigure ./configure -host=wasm32-emscripten --prefix=/home/mcuee/bin --enable-examples-build --enable-tests-build
emmake make
emmake make install
emmake make check
```

Error log.
```
mcuee@PCRyzen5600G:~/build/libusb/libusb_webusb$ emcc --version
emcc (Emscripten gcc/clang-like replacement + linker emulating GNU ld) 4.0.10 (b7dc6e5747465580df5984e723b9d1f10d8e804b)
Copyright (C) 2025 the Emscripten authors (see AUTHORS.txt)
This is free and open source software under the MIT license.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mcuee@PCRyzen5600G:~/build/libusb/libusb_webusb$ emmake make check
make: make check
Making check in libusb
make[1]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/libusb'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/libusb'
Making check in examples
make[1]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/examples'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/examples'
Making check in tests
make[1]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make  check-TESTS
make[2]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[3]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/tests'
FAIL: stress
FAIL: stress_mt
FAIL: set_option
FAIL: init_context
============================================================================
Testsuite summary for libusb-1.0 1.0.29
============================================================================
# TOTAL: 4
# PASS:  0
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log
Please report to libusb-devel@lists.sourceforge.net
============================================================================
make[3]: *** [Makefile:800: test-suite.log] Error 1
make[3]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[2]: *** [Makefile:908: check-TESTS] Error 2
make[2]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[1]: *** [Makefile:1015: check-am] Error 2
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make: *** [Makefile:491: check-recursive] Error 1
emmake: error: 'make check' failed (returned 2)
mcuee@PCRyzen5600G:~/build/libusb/libusb_webusb$ cat tests/test-suite.log 
=============================================
   libusb-1.0 1.0.29: tests/test-suite.log
=============================================

# TOTAL: 4
# PASS:  0
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: stress
============

../test-driver: line 112: ./stress: Permission denied
FAIL stress (exit status: 126)

FAIL: stress_mt
===============

../test-driver: line 112: ./stress_mt: Permission denied
FAIL stress_mt (exit status: 126)

FAIL: set_option
================

../test-driver: line 112: ./set_option: Permission denied
FAIL set_option (exit status: 126)

FAIL: init_context
==================

../test-driver: line 112: ./init_context: Permission denied
FAIL init_context (exit status: 126)
```

mcuee, 2025-06-08:
Full build log for reference.

<details>
<summary>Full build log for reference</summary>

```
mcuee@PCRyzen5600G:~/build/libusb/libusb_webusb$ emconfigure ./configure -host=wasm32-emscripten --prefix=/home/mcuee/bin --enable-examples-build --enable-tests-build
configure: ./configure -host=wasm32-emscripten --prefix=/home/mcuee/bin --enable-examples-build --enable-tests-build
checking for wasm32-emscripten-gcc... /home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether /home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc accepts -g... yes
checking for /home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc option to enable C11 features... none needed
checking whether /home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc understands -c and -o together... yes
checking whether the compiler supports GNU C++... yes
checking whether /home/mcuee/build/libusb/emsdk/upstream/emscripten/em++ accepts -g... yes
checking for /home/mcuee/build/libusb/emsdk/upstream/emscripten/em++ option to enable C++11 features... none needed
checking for inline... inline
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking dependency style of /home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc... gcc3
checking dependency style of /home/mcuee/build/libusb/emsdk/upstream/emscripten/em++... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... wasm32-unknown-emscripten
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by /home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc... /home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc
checking if the linker (/home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /home/mcuee/build/libusb/emsdk/upstream/bin/llvm-nm
checking the name lister (/home/mcuee/build/libusb/emsdk/upstream/bin/llvm-nm) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to wasm32-unknown-emscripten format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc option to reload object files... -r
checking for wasm32-emscripten-file... no
checking for file... file
checking for wasm32-emscripten-objdump... no
checking for objdump... objdump
checking how to recognize dependent libraries... unknown
checking for wasm32-emscripten-dlltool... no
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for wasm32-emscripten-ar... /home/mcuee/build/libusb/emsdk/upstream/emscripten/emar
checking for archiver @FILE support... @
checking for wasm32-emscripten-strip... no
checking for strip... strip
checking for wasm32-emscripten-ranlib... /home/mcuee/build/libusb/emsdk/upstream/emscripten/emranlib
checking command to parse /home/mcuee/build/libusb/emsdk/upstream/bin/llvm-nm output from /home/mcuee/build/libusb/emsdk/upstream/emscripten/emcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for wasm32-emscripten-mt... no
checking for mt... mt
checking if mt is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
[... 155 lines cut ...]
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/libusb'
Making install in examples
make[1]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/examples'
make[2]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/examples'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/examples'
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/examples'
Making install in tests
make[1]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[2]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[1]: Entering directory '/home/mcuee/build/libusb/libusb_webusb'
make[2]: Entering directory '/home/mcuee/build/libusb/libusb_webusb'
make[2]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/home/mcuee/bin/lib/pkgconfig'
 /usr/bin/install -c -m 644 libusb-1.0.pc '/home/mcuee/bin/lib/pkgconfig'
make[2]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb'
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb'
```

</details>

mcuee, 2025-06-08:
@RReverser 
Please help here when you get a chance to look at it. Thanks.

RReverser, 2025-06-08:
Looks like the successful run was with Emscripten 4.0.8, the new one with Emscripten 4.0.9. If you're sure it's not a regression on libusb side, or at least want to narrow it down, you can configure setup-sdk (https://github.com/libusb/libusb/blob/15a7ebb4d426c5ce196684347d2b7cafad862626/.github/workflows/linux.yml#L41) with `version: 4.0.8` as per https://github.com/mymindstorm/setup-emsdk.

RReverser, 2025-06-08:
The build seems successful either way, so it's a runtime regression _somewhere_.

RReverser, 2025-06-08:
I'll try to repro locally as well. Disabling the test is certainly not ideal as it would only hide the issue, and real projects depend on this backend.

RReverser, 2025-06-08:
> emconfigure ./configure -host=wasm32-emscripten

I think you had some issues because of this typo, should be `--host` not `-host`. Looks like blog rendering has gone awry and it converted all `--` into an em-dash instead.

The other thing is that, somewhat counter-intuitively, everything - including `make check` - needs to be run under `emconfigure` not `emmake`, as Emscripten adds shebang only if use use `emconfigure` (it should probably change that).

This works for me locally:

```sh
$ emconfigure ./configure --host=wasm32-emscripten --enable-examples-build --enable-tests-build
[...]
$ emconfigure make check -j8
configure: make check -j8
Making check in libusb
make[1]: Entering directory '/home/rreverser/libusb/libusb'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/home/rreverser/libusb/libusb'
Making check in examples
make[1]: Entering directory '/home/rreverser/libusb/examples'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/home/rreverser/libusb/examples'
Making check in tests
make[1]: Entering directory '/home/rreverser/libusb/tests'
make  check-TESTS
make[2]: Entering directory '/home/rreverser/libusb/tests'
make[3]: Entering directory '/home/rreverser/libusb/tests'
PASS: set_option
PASS: init_context
PASS: stress
FAIL: stress_mt
============================================================================
Testsuite summary for libusb-1.0 1.0.29
============================================================================
# TOTAL: 4
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log
Please report to libusb-devel@lists.sourceforge.net
============================================================================
=============================================
   libusb-1.0 1.0.29: tests/test-suite.log
=============================================

# TOTAL: 4
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: stress_mt
===============

Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 0 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 1 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 2 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 3 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 4 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 5 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 6 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 7 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
All threads discovered 0 devices (0 not opened)
All done, 8 errors
FAIL stress_mt (exit status: 1)
```

So it looks like something got broken for the multithreaded tests, at least others still work as expected.

tormodvolden, 2025-06-08:
The init/exit without enumeration works fine. With enumeration, and with 0 devices detected, I assume the multiple threads running simultaneous enumeration is causing the problem. Can you see if it is libusb_get_device_list() that fails with LIBUSB_ERROR_NOT_SUPPORTED?

mcuee, 2025-06-08:
I can reproduce the same issue now.
```
mcuee@debian12j4125:~/build/libusb/libusb_webusb$  ./bootstrap.sh 
...
mcuee@debian12j4125:~/build/libusb/libusb_webusb$  emconfigure ./configure --host=wasm32-emscripten --prefix=/home/mcuee/bin --enable-examples-build --enable-tests-build
...

mcuee@debian12j4125:~/build/libusb/libusb_webusb$ emconfigure make check -j4
configure: make check -j4
Making check in libusb
make[1]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/libusb'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/libusb'
Making check in examples
make[1]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/examples'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/examples'
Making check in tests
make[1]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make  check-TESTS
make[2]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[3]: Entering directory '/home/mcuee/build/libusb/libusb_webusb/tests'
PASS: set_option
PASS: init_context
PASS: stress
FAIL: stress_mt
============================================================================
Testsuite summary for libusb-1.0 1.0.29
============================================================================
# TOTAL: 4
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log for debugging.
Some test(s) failed.  Please report this to libusb-devel@lists.sourceforge.net,
together with the test-suite.log file (gzipped) and your system
information.  Thanks.
============================================================================
make[3]: *** [Makefile:803: test-suite.log] Error 1
make[3]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[2]: *** [Makefile:938: check-TESTS] Error 2
make[2]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make[1]: *** [Makefile:1045: check-am] Error 2
make[1]: Leaving directory '/home/mcuee/build/libusb/libusb_webusb/tests'
make: *** [Makefile:496: check-recursive] Error 1
emconfigure: error: 'make check -j4' failed (returned 2)

mcuee@debian12j4125:~/build/libusb/libusb_webusb$ cat tests/test-suite.log
=============================================
   libusb-1.0 1.0.29: tests/test-suite.log
=============================================

# TOTAL: 4
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

System information (uname -a): Linux 6.1.0-37-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.140-1 (2025-05-22) x86_64
Distribution information (/etc/os-release):
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"

.. contents:: :depth: 2

FAIL: stress_mt
===============

Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 0 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 1 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 2 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 3 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 4 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 5 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 6 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
Thread 7 failed (iteration 0): LIBUSB_ERROR_NOT_SUPPORTED
All threads discovered 0 devices (0 not opened)
All done, 8 errors
FAIL stress_mt (exit status: 1)

```

RReverser, 2025-06-08:
> There is one in getDeviceList() for emscripten but I suppose that look-up must be thread-safe?

I'd expect so too, especially since it worked before and Emscripten backend hasn't changed in some time.

RReverser, 2025-06-08:
Ok found it. Looks like Node.js 21 added an experimental global `navigator` object so our shim that assigns the `navigator` (to simulate browser env) no longer works as expected: https://nodejs.org/api/globals.html#navigator_1

Should be an easy fix...

RReverser, 2025-06-08:
Please see the attached PR for the fix.

Btw, I wonder if I could/should be added to the maintainer list? I probably won't be able to spend much time on rest of libusb (not to mention lack of expertise for other OS), but having an access might make maintenance of this backend a bit quicker.

mcuee, 2025-06-09:
> Oh wait, I missed an important bit for local debugging:
> 
> [libusb/.private/ci-build.sh](https://github.com/libusb/libusb/blob/15a7ebb4d426c5ce196684347d2b7cafad862626/.private/ci-build.sh#L81)
> 
> Line 81 in [15a7ebb](/libusb/libusb/commit/15a7ebb4d426c5ce196684347d2b7cafad862626)
> 
>  export NODE_OPTIONS="--require ${scriptdir}/../tests/webusb-test-shim/" 
> You need to set the same env variable to point to the WebUSB shim if testing via Node.js:
> 
> ```
> rreverser@P14:~/libusb/tests/webusb-test-shim$ npm ci
> 
> added 4 packages, and audited 5 packages in 2s
> 
> found 0 vulnerabilities
> rreverser@P14:~/libusb/tests/webusb-test-shim$ cd ../..
> rreverser@P14:~/libusb$ export NODE_OPTIONS="--require $(pwd)/tests/webusb-test-shim/"
> rreverser@P14:~/libusb$ emconfigure make check -j8
> ...
> ```

Thanks. Looks like we need to document this one.

mcuee, 2025-06-09:
> Ok found it. Looks like Node.js 21 added an experimental global `navigator` object so our shim that assigns the `navigator` (to simulate browser env) no longer works as expected: https://nodejs.org/api/globals.html#navigator_1
> 
> Should be an easy fix, although what's concerning is that only `stress_mt` has caught this even though right now `libusb_get_device_list` returns "not supported" unconditionally, so all tests should've failed.

Great. PR #1656 fixed this issue. Thanks.

mcuee, 2025-06-09:
> Please see the attached PR for the fix.
> 
> Btw, I wonder if I could/should be added to the maintainer list? I probably won't be able to spend much time on rest of libusb (not to mention lack of expertise for other OS), but having an access might make maintenance of this backend a bit quicker.

Let me check with the other maintainers. Thanks.

tormodvolden, 2025-06-09:
I am glad it was not a threading issue :)

commit 4b9cf1fb (2025-06-09): tests: Fix Node.js shim for Node 21+ (webusb/emscripten)

Fixes #1648
Closes #1656