libusb/libusb #1661: libusb_get_port_numbers returns nothing on some macOS versions

eswierk, 2025-06-15:
The name of the IORegistry port property is not consistent across macOS versions: it's `port` on some, `usb-port-number` on others. (Run `ioreg -c AppleUSB30HubPort` or `ioreg -c AppleUSB20HubPort` to show the properties on your Mac.)

`get_device_port` in os/darwin_usb.c should pass `kUSBHostPortPropertyPortNumber` rather than `"port"` to `get_ioregistry_value_data`. This makes `libusb_get_port_numbers` return port numbers on all recent versions of macOS.

mvp, 2025-06-15:
@eswierk, unfortunately this is not very good approach and arguably is macOS 26 bug.
Scenario: I have an executable (e.g. uhubctl) that links to libusb and built using MacOSX15.5 sdk or earlier, and suppose I even patched libusb to use macro `kUSBHostPortPropertyPortNumber` as you suggest.

However, such executable will not work correctly in macOS 26 because this macro would have `"port"` linked into it, but macOS 26 would require `"usb-port-number"` to work correctly.

Better way to fix this is to use function `get_running_version()` defined in `os/darwin_usb.c` and pass `port` or `usb-port-number` depending on currently running macOS version. Or, better yet, have Apple rollback macOS 26 USB stack to use `"port"` like it always did before it goes out of beta.

eswierk, 2025-06-15:
Good point. My bias is showing — on Linux, there's no such thing as building against one version of the OS and running on a different one.

seanm, 2025-06-16:
Apple usually takes binary-compatibility bugs pretty seriously, at least they used to.  I would definitely file a bug. Could you post the FB number here once you do?  And even the text of your bug report if you don't mind...

mvp, 2025-06-18:
Filed to Apple feedback https://feedbackassistant.apple.com/feedback/18170043. Full text is below:

**macOS 26 breaks backwards USB compatibility with libusb**

**Please provide a descriptive title for your feedback:**
macOS 26 breaks backwards USB compatibility with libusb

**Which area are you seeing an issue with?**
USB/Thunderbolt

**What type of issue are you reporting?**
Incorrect/Unexpected Behavior

**Description**
Please describe the issue and what steps we can take to reproduce it:

On macOS 15.5 and any previous versions, IORegistry IOUSB plane port property was always "port". However, on macOS 26 Tahoe "port" does not work, but "usb-port-number" does.

I am a developer for open source software uhubctl https://github.com/mvp/uhubctl.
This utility depends on open source library libusb which among other platforms supports MacOS - all versions from 10.x to 15.5.

macOS Tahoe 26 beta testers reported that my software does not work on macOS 26 
https://github.com/mvp/uhubctl/issues/623.

Upon closer inspection it turned out that in macOS 26 libusb is not able to read USB port properties https://github.com/libusb/libusb/issues/1661.

Steps to reproduce:

Run command line `ioreg -c AppleUSB20HubPort -l`.

Expected result:

You should see that port property have name "port" e.g.
    "port" = <01000000>
    "port" = <02000000>
This is what observed on macOS 15.5 and earlier.

Actual result:

On macOS 26, you will see this instead:
    "usb-port-number" = <01000000>...
Also, any utility that depends on libusb will not work correctly or at all.

Solution that was tested to work on macOS 26 was to patch libusb to refer to port property as "usb-port-number" instead of "port" https://github.com/libusb/libusb/blob/3dbfa16/libusb/os/darwin_usb.c#L1263.

libusb developers consider this breaking backwards binary compatibility by macOS https://github.com/libusb/libusb/issues/1661#issuecomment-2977347018.

Please revert back to using "port" property before macOS 26 goes out of beta, so this issue does any software dependent on libusb.

seanm, 2025-06-20:
I submitted similar as FB18234941.  I encourage others to do so too; duplicate bug reports increases priority.

parisbutterfield, 2025-07-22:
Beta 4 `25A5316i` of macOS 26 is still using `usb-port-number`. I opened a bug with Apple a few weeks ago, but haven't heard anything. How can we amplify this issue?

```   
"usb-port-number" = <02000000>
    | |   | |       "locationID" = 35651584
    | |   | |       "kUSBWakePortCurrentLimit" = 3000
    | |   | |       "port-status" = 5
    | |   | |       "UsbCPortNumber" = 3
    | |   | |       "usb-c-port-number" = <03000000>

```

seanm, 2025-07-23:
>I opened a bug with Apple a few weeks ago, but haven't heard anything.

Which is typical. Apple's bug reporting system is widely known as a black hole: information enters, but never exits. :(

>How can we amplify this issue?

More reports increases priority. Otherwise pray?

paulzaptec, 2025-09-15:
Unfortunately this bug is still present in the release version of macOS 26.0

paulzaptec, 2025-09-16:
Amazing! Yes, can confirm this patch has the desired effect on macOS 26.0.

Before patch:
```
0fd9:0098 (bus 1, device 10)
08bb:2704 (bus 1, device 11)
0bda:5411 (bus 1, device 9)
0634:5602 (bus 1, device 6)
0bda:5411 (bus 1, device 8)
0bda:8153 (bus 1, device 7)
1a40:0801 (bus 1, device 5)
1908:0226 (bus 1, device 4)
2109:2817 (bus 1, device 3)
2109:0102 (bus 1, device 2)
046d:0843 (bus 0, device 3)
0bda:5411 (bus 0, device 2)
0bda:0411 (bus 0, device 1)
1a40:0801 (bus 1, device 1)
```

After patch:

```
0fd9:0098 (bus 1, device 10) path: 1.2.4.4
08bb:2704 (bus 1, device 11) path: 1.2.4.1.1
0bda:5411 (bus 1, device 9) path: 1.2.4.1
0634:5602 (bus 1, device 6) path: 1.2.2
0bda:5411 (bus 1, device 8) path: 1.2.4
0bda:8153 (bus 1, device 7) path: 1.2.3
1a40:0801 (bus 1, device 5) path: 1.3
1908:0226 (bus 1, device 4) path: 1.4
2109:2817 (bus 1, device 3) path: 1.2
2109:0102 (bus 1, device 2) path: 1.1
046d:0843 (bus 0, device 3) path: 1.1
0bda:5411 (bus 0, device 2) path: 1
0bda:0411 (bus 0, device 1) path: 2
1a40:0801 (bus 1, device 1) path: 1
```

mvp, 2025-09-17:
Fix pr https://github.com/libusb/libusb/pull/1700 is ready to merge now.

This fix has been confirmed to work with uhubctl https://github.com/mvp/uhubctl/issues/623#issuecomment-3300325458
Also, I am getting users requests to fix uhubctl on macOS 26 https://github.com/mvp/uhubctl/issues/633.
 
@seanm, @mcuee, @Youw: can you guys release new libusb version which contains this fix?
Thank you!

Youw, 2025-09-17:
> can you guys release new libusb version which contains this fix?

I can only vote that a patch release for such a fix is a good idea, but that is for someone else to actually perform.

mcuee, 2025-09-18:
I am in the process of upgrading to macOS 26. I will test PR #1700 before approving.

Then @seanm or @sonatique can merge the PR.

mvp, 2025-09-18:
Waiting few months is less than ideal. But at least we can have
```
brew install libusb --head
```
until then. Because without this, macOS 26 users are in huge pain.

mcuee, 2025-09-18:
As for the timing of the release, that will be discussed among the more active maintainers.

@tormodvolden @sonatique and @seanm are the more active developers now.

And for some more complicated macOS related issues, it will still be good to hear from @hjelmn if he got some time.

sonatique, 2025-09-18:
I merged PR #1700 . @mcuee can we close this issue as well?

@mcuee  @tormodvolden : I am not yet familiar with libusb release process, but we recently released 1.0.29 to fix a bug in 1.0.28 soon after 1.0.28 without RC if I remember correctly.

Since then, we merged a lot of "heavy" things in master so we may not want to rush for releasing everything, but what about doing a new "maintenance" (it's not "bug fix" this time) release with the content of 1.0.29 + the macOS fix of #1700, maybe even just releasing the macOS binaries. If we do so, are we bound to use 1.0.30 for this maintenance release of can we use 1.0.29.1 or something similar?

mcuee, 2025-09-18:
@sonatique 

I will prefer we release from the git master branch.

We have never done otherwise.

Usually we will one RC release and then wait for the testing results in two weeks or one month time for the official release.

Personally I think we can wait a bit for more macOS related issues to be reported as macOS 26 seems to be a big release which may have more issues. Some of the issues will be fixed by Apple in macOS 26.1. Some of the other issues may require fixes from libusb side.

sonatique, 2025-09-18:
@seanm OK, maybe "heavy" depends on personal appreciation : I was thinking of the API addition we introduced, which, considering how libusb project is cautious, this might be seen as heavy ;-)

Anyway, in general I am in favor of shorter release cycle, and in this particular case I don't see the benefit of delaying an RC: we merged 33 commits since last 1.0.29 release with long awaited new features. We merged all priority PR for Windows and the 2 listed for MacOS doesn't seem really important anymore...

We can always do more and try to merge the listed PR still in review, but we can also set them for 1.0.31.

mcuee, 2025-09-19:
> [@seanm](https://github.com/seanm) OK, maybe "heavy" depends on personal appreciation : I was thinking of the API addition we introduced, which, considering how libusb project is cautious, this might be seen as heavy ;-)

That cautiousness is actually due to lack of active developers ...

> Anyway, in general I am in favor of shorter release cycle, and in this particular case I don't see the benefit of delaying an RC: we merged 33 commits since last 1.0.29 release with long awaited new [...]

Yes, I agree.

> We can always do more and try to merge the listed PR still in review, but we can also set them for 1.0.31.

Yes, that should be the way.

We can continue the discussion in the main 1.0.30 release issue. 
* https://github.com/libusb/libusb/issues/1647

paulzaptec, 2025-09-22:
Thank you for the very quick response on this issue! 

I can confirm that 
```
brew uninstall libusb --ignore-dependencies
brew install libusb --head
```
resolves the issue for me on macOS Tahoe 26.0 (until this makes it into a future release).