libusb/libusb #1669: CMSIS-DAP detection failure after repeated open/close cycles due to vanishing string descriptors (0d28:0204)​

zh522130, 2025-07-18:
### Environment:​​

- libusb version: ​1.0.29​
- Host software: ([hpmicro/riscv-openocd)](https://github.com/hpmicro/riscv-openocd)
- Test tool: ​rusb​ (rust libusb binding) [list_devices.rs example](https://github.com/a1ien/rusb/blob/master/examples/list_devices.rs)
- Device: Custom ​CMSIS-DAP debugger​ (VID/PID: ​0d28:0204)

### Normal State (initial OpenOCD run):​

list_devices result:

```
Bus 001 Device 002 ID 0d28:0204  480 Mbps
Device Descriptor:
  bLength               18
  bDescriptorType        1
  bcdUSB              2.10
  bDeviceClass        0xef
  bDeviceSubClass     0x02
  bDeviceProtocol     0x01
  bMaxPacketSize0       64
  idVendor          0x0d28 NXP
  idProduct         0x0204 ARM mbed
  bcdDevice           1.00
  iManufacturer          1 CherryUSB
  iProduct               2 CherryUSB CMSIS-DAP
  iSerialNumber          3 012345ABCDEF
  bNumConfigurations     1
  Config Descriptor:
    bLength                9
    bDescriptorType        2
    wTotalLength      0x0062
    bNumInterfaces         3
    bConfigurationValue    1
    iConfiguration         0
    bmAttributes:
      Self Powered     false
      Remote Wakeup    false
    bMaxPower            500mW
    no extra data
    Interface Descriptor:
      bLength                9
      bDescriptorType        4
      bInterfaceNumber       0
      bAlternateSetting      0
      bNumEndpoints          2
      bInterfaceClass     0xff
      bInterfaceSubClass  0x00
      bInterfaceProtocol  0x00
      iInterface             2 CherryUSB CMSIS-DAP
    []
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x01 EP 1 Out
        bmAttributes:
          Transfer Type          Bulk
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0200
        bInterval              0
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x81 EP 1 In
        bmAttributes:
          Transfer Type          Bulk
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0200
        bInterval              0
    Interface Descriptor:
      bLength                9
      bDescriptorType        4
      bInterfaceNumber       1
      bAlternateSetting      0
      bNumEndpoints          1
      bInterfaceClass     0x02
      bInterfaceSubClass  0x02
      bInterfaceProtocol  0x01
      iInterface             0
    no extra data
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x83 EP 3 In
        bmAttributes:
          Transfer Type          Interrupt
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0008
        bInterval             10
    Interface Descriptor:
      bLength                9
      bDescriptorType        4
      bInterfaceNumber       2
      bAlternateSetting      0
      bNumEndpoints          2
      bInterfaceClass     0x0a
      bInterfaceSubClass  0x00
      bInterfaceProtocol  0x00
      iInterface             0
    []
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x02 EP 2 Out
        bmAttributes:
          Transfer Type          Bulk
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0200
        bInterval              0
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x82 EP 2 In
        bmAttributes:
          Transfer Type          Bulk
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0200
        bInterval              0
```

### Failed State (after multiple open/close cycles):​

openocd reports:

```
Error: unable to find a matching CMSIS-DAP device
```

list_devices result:

```
Bus 001 Device 002 ID 0d28:0204  480 Mbps
Device Descriptor:
  bLength               18
  bDescriptorType        1
  bcdUSB              2.10
  bDeviceClass        0xef
  bDeviceSubClass     0x02
  bDeviceProtocol     0x01
  bMaxPacketSize0       64
  idVendor          0x0d28 NXP
  idProduct         0x0204 ARM mbed
  bcdDevice           1.00
  iManufacturer          1
  iProduct               2
  iSerialNumber          3
  bNumConfigurations     1
  Config Descriptor:
    bLength                9
    bDescriptorType        2
    wTotalLength      0x0062
    bNumInterfaces         3
    bConfigurationValue    1
    iConfiguration         0
    bmAttributes:
      Self Powered     false
      Remote Wakeup    false
    bMaxPower            500mW
    no extra data
    Interface Descriptor:
      bLength                9
      bDescriptorType        4
      bInterfaceNumber       0
      bAlternateSetting      0
      bNumEndpoints          2
      bInterfaceClass     0xff
      bInterfaceSubClass  0x00
      bInterfaceProtocol  0x00
      iInterface             2
    []
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x01 EP 1 Out
        bmAttributes:
          Transfer Type          Bulk
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0200
        bInterval              0
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x81 EP 1 In
        bmAttributes:
          Transfer Type          Bulk
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0200
        bInterval              0
    Interface Descriptor:
      bLength                9
      bDescriptorType        4
      bInterfaceNumber       1
      bAlternateSetting      0
      bNumEndpoints          1
      bInterfaceClass     0x02
      bInterfaceSubClass  0x02
      bInterfaceProtocol  0x01
      iInterface             0
    no extra data
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x83 EP 3 In
        bmAttributes:
          Transfer Type          Interrupt
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0008
        bInterval             10
    Interface Descriptor:
      bLength                9
      bDescriptorType        4
      bInterfaceNumber       2
      bAlternateSetting      0
      bNumEndpoints          2
      bInterfaceClass     0x0a
      bInterfaceSubClass  0x00
      bInterfaceProtocol  0x00
      iInterface             0
    []
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x02 EP 2 Out
        bmAttributes:
          Transfer Type          Bulk
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0200
        bInterval              0
      Endpoint Descriptor:
        bLength                7
        bDescriptorType        5
        bEndpointAddress    0x82 EP 2 In
        bmAttributes:
          Transfer Type          Bulk
          Synch Type             NoSync
          Usage Type             Data
        wMaxPacketSize    0x0200
        bInterval              0
```

iManufacturer, iProduct, iSerialNumber, iInterface become empty, so openocd reports error.

**use nusb(another rust crate) consistently returns correct strings​ while libusb-based rusb fails**

zh522130, 2025-07-18:
While rusb(libusb) shows empty strings, ​sysfs confirms strings are intact​ (device path 1-1):
```
# Manufacturer string (iManufacturer)
$ cat /sys/bus/usb/devices/1-1/manufacturer
CherryUSB

# Product string (iProduct) - contains detection key
$ cat /sys/bus/usb/devices/1-1/product
CherryUSB CMSIS-DAP  # ← Critical "CMSIS-DAP" identifier present

# Serial number (iSerialNumber)
$ cat /sys/bus/usb/devices/1-1/serial
012345ABCDEF
```

zh522130, 2025-07-18:
I compiled testlibusb, and the execution results are as follows:​​

```sh
./testlibusb
Dev (bus 2, device 1): 1D6B - 0001 speed: 12M
  Manufacturer:              Linux 5.10.110 ohci_hcd
  Product:                   Generic Platform OHCI controller
Dev (bus 1, device 3): 0D28 - 0204 speed: 480M
Dev (bus 1, device 1): 1D6B - 0002 speed: 480M
  Manufacturer:              Linux 5.10.110 ehci_hcd
  Product:                   EHCI Host Controller
```

mcuee, 2025-07-18:
I tend to believe this is a device FW issue and not a libusb issue.

zh522130, 2025-07-18:
> I tend to believe this is a device FW issue and not a libusb issue.我倾向于相信这是一个设备 FW 问题，而不是 libusb 问题。

When `testlibusb` attempted to read the device descriptors, it encountered a `LIBUSB_ERROR_OVERFLOW` error. I suspect this might be caused by a potential memory overflow in the CMSIS-DAP firmware, leading to descriptor corruption? Meanwhile, nusb and sysfs might be using cached data? I will investigate this further.