libusb/libusb #1701: There are many incompatibilities after updating macOS 26

sbfkcel, 2025-09-17:
Please proceed with caution when upgrading to macOS 26. The specific impacts are still under investigation.
I'm backing up my data and rolling back to macOS 15.7.

mcuee, 2025-09-18:
@sbfkcel 

Please help to provide more details about the issue you encountered in macOS 26.

Some of the issues may be fixed in the next minor release 26.1 or 26.2. Let's see.

sbfkcel, 2025-09-22:
After further verification and testing, I believe the error I encountered may not be a libusb issue. Further confirmation may be required.

Rather, it's due to a limitation in macOS 26 for specific interfaces.

**Specifically, the testing steps are as follows:**
- Discover the device using libusb;
- Open the QuickTime interface and switch to it;
- Attempt to release the interface;
- Claim the interface again.

---
Attempting to claim interfaces other than QuickTime successfully results in a specific limitation in macOS 26.

---

```c
/*
 * C implementation of main.js libusb functionality
 * Find USB device by serial number, get device info, control QuickTime configuration
 */

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>
#include <time.h>
#include "libusb/libusb/libusb.h"

// Target device serial number
#define TARGET_SERIAL_NUMBER "1403412cb13fdecb02c96623819f928fd5f684dc"

// Device information structure
typedef struct {
    char serial_number[256];
    char product[256];
    int mux_config_index;
    int qt_config_index;
    libusb_device *device;
    libusb_device_handle *handle;
} usb_device_info_t;

// Function declarations
static int find_device_by_serial(const char *serial_number, libusb_device **device);
static int get_device_info(libusb_device *device, usb_device_info_t *info);
static int find_interface_for_subclass(struct libusb_config_descriptor *config, uint8_t subclass, int *interface_number);
static int find_configurations(libusb_device *device, int *mux_config_index, int *qt_config_index);
static int control_qt_config(libusb_device_handle *handle, const char *flag);
static int switch_qt_config(libusb_device *device, usb_device_info_t *info);
static int find_quicktime_interface(libusb_device_handle *handle, int *interface_number);
static int release_interface(libusb_device_handle *handle, int interface_number);
static int claim_interface(libusb_device_handle *handle, int interface_number);
static void print_error(const char *message, int error_code);

int main(int argc, char *argv[]) {
    (void)argc;  // Avoid unused parameter warning
    (void)argv;
    int result = 0;
    libusb_context *ctx = NULL;
    libusb_device *device = NULL;
    usb_device_info_t info = {0};
    int qt_interface_number = -1;
    
    printf("Starting USB device operations...\n");
    
    // Initialize libusb
    result = libusb_init_context(&ctx, NULL, 0);
    if (result < 0) {
        print_error("Failed to initialize libusb", result);
        return 1;
    }
[... 307 lines cut ...]
    
    result = find_interface_for_subclass(config, 0x2A, interface_number);
    libusb_free_config_descriptor(config);
    
    return result;
}

// Release interface
static int release_interface(libusb_device_handle *handle, int interface_number) {
    return libusb_release_interface(handle, interface_number);
}

// Claim interface
static int claim_interface(libusb_device_handle *handle, int interface_number) {
    return libusb_claim_interface(handle, interface_number);
}

// Print error information
static void print_error(const char *message, int error_code) {
    const char *error_name = libusb_error_name(error_code);
    fprintf(stderr, "Error: %s (Error code: %d, Error name: %s)\n", 
            message, error_code, error_name ? error_name : "Unknown error");
}

```

Mitch9192, 2025-10-06:
Basic stuff looks mostly right with the exception of:
"libusb_get_port_numbers"

Tested against:
- Tahoe - 26.0.1
- Sequoia - 15.6.1

Output:
- [simple_libusb_test_tahoe_26_0_1.txt](https://github.com/user-attachments/files/22731100/simple_libusb_test_tahoe_26_0_1.txt)
- [simple_libusb_test_sequoia_15.6.1.txt](https://github.com/user-attachments/files/22731103/simple_libusb_test_sequoia_15.6.1.txt)

Build Files: 
Note: will need to remove ".txt" from the makefile.
- [simple_usb_list.c](https://github.com/user-attachments/files/22731066/simple_usb_list.c)
- [Makefile.txt](https://github.com/user-attachments/files/22731071/Makefile.txt)
- Source code should live at the same level as release 1.0.29 (From here: https://github.com/libusb/libusb/archive/refs/tags/v1.0.29.zip)

There might be more complex connection stability issues, but keeping it simple for now.

seanm, 2025-10-06:
So are you saying https://github.com/libusb/libusb/issues/1661 is the only issue you've seen?

Mitch9192, 2025-10-06:
Sorry, didn't scrub/search the issues. This seemed like a catch all for Tahoe "incompatibilities"

At this time this is the only reproducible issue I have. Will chime in if I find others.

mcuee, 2025-10-08:
I will close this issue for now. Please open new issue if something comes up. Thanks.