libusb/libusb #1703: When a UVC device, under the libusbk(3.0.7.0) driver, is unplugged during the image output process, the computer BSODs.

Timi0310, 2025-09-18:
Hello, I'm developing an application using the libuvc library and libusb (1.0.29). I'm working with a UVC device that outputs images in bulk mode. I used Zadig 2.4 to install the libusbK (3.0.7.0) driver for the UVC device. The developed program can normally acquire images from the camera module.

However, when I disconnect the USB cable between the device and the computer, the computer immediately blue screens (BSOD). When I use Zadig 2.4 to install the libusb-win32 (v1.2.6.0) or WinUSB (v6.1.7600.16385) drivers, no blue screen occurs when the UVC device is unplugged during image output.

Using the same program and procedure with an isochronous (isoc) mode UVC device under the libusbK driver does not result in a blue screen.

mcuee, 2025-09-18:
If the problem only occurs with libusbk driver but not libusb-win32 driver and WinUSB driver, then this is not a libusb issue.

You can create an issue for libusbK project.

But in reality there will be unlikely new release of libusbK driver release due to kernel signing certificate issue. The recommendation is to switch to WinUSB driver.

mcuee, 2025-09-18:
Most likely the issue is the same as the existing libusbK issue.

* https://github.com/mcuee/libusbk/issues/54

closed 2025-09-18 by mcuee